wills106/homeassistant-solax-modbus #1120: [FR]: SolaX - Add support for X3-MIC with Serial Starting MU902T

CVKBaca, 2024-11-02:
### Description

Add suport inverter type Solax X3-MIC (model: X3-9.0-T-D) SN: MU902T

### Suggested Solution

Hello It is posible add suport Solax X3-MIC sn MU902T******** mayby protocol is similar to other solax gen1.
Thenks

CVKBaca, 2024-11-04:
With ARM firmware version 1.38 and after editing the files plugin_solax.py and plugin_solax.cpython-312.pyc, I got Modbus communication up and running.
I added the following piece of code to plugin_solax.py:

```"
elif seriesnumber.startswith('MU902T'):
    invertertype = MIC | GEN | X3 # MIC X3
    self.inverter_model = "X3-MIC""
``` 
I added the following to the plugin_solax.cpython-312.pyc file, although I'm not sure if it was necessary. :)
[image]

The result is as follows. The only sensor that didn’t load and isn’t working yet is the RTC sensor.
[image]
[image]

wills106, 2024-11-16:
Do you perhaps need to first sync the RTC for it to return values?

closed 2025-09-03 by wills106