wills106/homeassistant-solax-modbus #1126: [Bug]: Export Control User Limit 10x less then real value

jansidlo, 2024-11-06:
### Describe the bug

Found out that SolaX Export Control User Limit is 10x less then the value in Solax cloud
[image]
[image]

### Integration Version

2024.11.1

### Homeassistant core version

2024.10.4

### Inverter brand

Solax X3-Hybrid Gen4

### Plugin used

plugin_solax.py

### Serial prefix

H34B10H

### Inverter firmware versions

DSP v1.42 ARM v1.41

### Connection Method

USB modbus convertor

### Dongle firmware

10.16

Codiwr, 2024-11-08:
10x more on mine still! Reads 60000W!

wills106, 2024-11-10:
If you add
`('H34B10H', 1),`
to this block:

https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L458-L468

Does it read correctly?

jansidlo, 2024-11-10:
it didn't work for me.

I added **('H34B10H', 1),** to end of block, restarted HA. Then added string to begining of block and nothing has changed.

jansidlo, 2024-11-12:
I tried to change scale for H34 and I got the same result.

BretonLud, 2024-11-12:
> I tried to change scale for H34 and I got the same result.

Try this ('H34B10H', 10),

BretonLud, 2024-11-12:
> 10x more on mine still! Reads 60000W!

Can you tell us yours serial number ?

Codiwr, 2024-11-12:
H4752AJ3415002.

Thanks!

BretonLud, 2024-11-12:
> H4752AJ3415002. Thanks!
> […](#)

So add this ('H4752A',1) on [homeassistant-solax-modbus/custom_components/solax_modbus/plugin_solax.py](https://github.com/wills106/homeassistant-solax-modbus/blob/1243dc8b050fa5af885a9a8b80207c05d1b24d8f/custom_components/solax_modbus/plugin_solax.py#L458-L468)

it's work for me

Codiwr, 2024-11-12:
Sorted, you absolute genius!

styriaman, 2024-11-15:
This problem occurs also on an X3 Ultra 
in HA 3000W -> on the inverter 30000W

BretonLud, 2024-11-15:
> This problem occurs also on a X3 Ultra in HA 3000W -> on the inverter 30000W BTW: The serial of my X3 Ultra starts with "H3BC....." and not with "H3BD"

then add this line ('H3BC', 10) on 

```
EXPORT_LIMIT_SCALE_EXCEPTIONS = [ 
     ('H4', 10 ),     # assuming all Gen4s 
     ('H34', 10),     # assuming all Gen4s 
     ('H3UE', 10),  # Issue #339, 922 
     ('H4372A', 1),   # Issue #857 
     ('H4502A', 1),   # Issue #857 
     ('H4502T', 1),   # Issue #418 
     ('H4602A', 1),   # Issue #882 
     ('H3BD', 10),    # X3-Ultra
     ('H4752A', 1),
     ('H3BC',10),
      ('H1E', 10 ), # more specific entry comes last and wins 
 ]
 ```

BretonLud, 2024-11-15:
@wills106 

The new variable EXPORT_LIMIT_SCALE_EXCEPTIONS : 

```
EXPORT_LIMIT_SCALE_EXCEPTIONS = [ 
     ('H4', 10 ),     # assuming all Gen4s 
     ('H34', 10),     # assuming all Gen4s 
     ('H3UE', 10),  # Issue #339, 922 
     ('H4372A', 1),   # Issue #857 
     ('H4502A', 1),   # Issue #857 
     ('H4502T', 1),   # Issue #418 
     ('H4602A', 1),   # Issue #882 
     ('H3BD', 10),    # X3-Ultra
     ('H4752A', 1),
     ('H3BC',10),
     ('H34B10H', 10) # need return @jansidlo ,
      # ('H1E', 10 ), # more specific entry comes last and wins 
 ]
 ```

wills106, 2024-11-16:
Just released [2024.11.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.11.2) with these changes.

Codiwr, 2024-11-16:
Thanks for trying.

The update by BretomLud worked.

This update had broken the integration - pulling the dongle and replacing
doesn't restore entities as is usually the case - Had to restore HA backup.

BretonLud, 2024-11-16:
> Thanks for trying. The update by BretomLud worked. This update had broken the integration - pulling the dongle and replacing doesn't restore entities as is usually the case - Had to restore HA backup.
> […](#)

The update work for me and dont broken ^^

jansidlo, 2024-11-17:
> > I tried to change scale for H34 and I got the same result.
> 
> Try this ('H34B10H', 10),

no change after entering this line. I expected to see a difference. 
but I found **SolaX Config Export Control Limit Readscale** in diagnostic menu. I changed it to 10 and Export Control Limit value is correct now

BretonLud, 2024-11-17:
> no change after entering this line. I expected to see a difference. but I found **SolaX Config Export Control Limit Readscale** in diagnostic menu. I changed it to 10 and Export Control Limit value [...]

Can you show me where you find this ?

wills106, 2024-11-17:
It's disabled by default, it's under entities not shown.
[image]

jansidlo, 2024-11-17:
Go to settings -> devices and services -> solax inverter modbus -> choose
your solax

closed 2025-09-17 by github-actions[bot]