wills106/homeassistant-solax-modbus #1127: [Bug]: Platform solax_modbus does not generate unique IDs

prehor, 2024-11-08:
### Describe the bug

When I enable the EPS Frequency sensor, this entry appears in the HASS log:

```
ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID Solax_eps_frequency is already used by sensor.solax_eps_frequency - ignoring sensor.Solax_eps_frequency
```

### Integration Version

2024.11.1

### Homeassistant core version

2024.11.0

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

DSP v1.46 ARM v1.44

### Connection Method

SolaX Pocket WiFi+LAN 3.0, LAN Connection

### Dongle firmware

n/a

### Detailed Error Log

```
ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID Solax_eps_frequency is already used by sensor.solax_eps_frequency - ignoring sensor.Solax_eps_frequency
```

prehor, 2024-12-06:
I discovered duplicate sensor `eps_frequency` for the following registers:  
- [Register 0x4F](https://github.com/wills106/homeassistant-solax-modbus/blob/77075cb6f0cb3518ecc7a85aaf910b8e5e089b59/custom_components/solax_modbus/plugin_solax.py#L4246)  
- [Register 0xB9](https://github.com/wills106/homeassistant-solax-modbus/blob/77075cb6f0cb3518ecc7a85aaf910b8e5e089b59/custom_components/solax_modbus/plugin_solax.py#L2814)  

According to version 3.29 of the X1/X3-Hybrid-G4 Modbus Registers PDF, registers 0x4F to 0x7C are marked as "Reserved" (i.e., not used). On the other hand, register 0xB9 is designated as the Off-Grid Frequency.

HA-Supi, 2025-04-02:
Will there be a fix? 

[image]

closed 2025-09-17 by github-actions[bot]