wills106/homeassistant-solax-modbus #1133: [FR]: Missing House Load Today

jannovotnyvdf, 2024-11-11:
### Description

There is no any overview how much was consumed energy in house in a day (House Load Today)-
I am using Solax X3-ULTRA

### Suggested Solution

Could you please create new Entity which will saves the daily house consumption?

Kahless656, 2024-11-13:
For me on Growatt same. After last update House Load not working, no L1,..,Daily

wills106, 2024-11-16:
Not sure if there is one on the Inverter? The energy dashboard calculates it though.

stopcoder, 2024-11-23:
You can let it be calculated based on the existing sensor `sensor.solax_house_load` with the help of the following two integrations:

- [Integral](https://www.home-assistant.io/integrations/integration/): calculates the integral of the current load (W) into the energy consumption (kWh)
- [Utility Meter](https://www.home-assistant.io/integrations/utility_meter/): calculates the sum for a sensor for a certain time period (for example, every hour, every day ...)

closed 2025-09-17 by github-actions[bot]