wills106/homeassistant-solax-modbus #1134: [Bug]: all entities unavailable

motclem, 2024-11-12:
### Describe the bug

All of my Solax entities have become unavailable they appeared to be installed but are showing as not available. I If I try to reinstall the integration it cannot find the inverter at the end IP address I give.

### Integration Version

2024.11.1

### Homeassistant core version

2024.11.1

### Inverter brand

solax power

### Plugin used

plugin_solax.py

### Serial prefix

HCC502

### Inverter firmware versions

are: DSP v3.24 ARM v3.21

### Connection Method

ethernet

### Dongle firmware

unknown

### Detailed Error Log

2024-11-12 18:39:14.127 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-11-12 18:39:19.132 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:39:19.133 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-11-12 18:39:24.138 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:39:24.140 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-11-12 18:39:24.140 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-11-12 18:39:24.140 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-11-12 18:39:39.146 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:39:39.146 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-11-12 18:39:44.151 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:39:44.152 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-11-12 18:39:49.158 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:39:49.159 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-11-12 18:39:49.159 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-11-12 18:39:49.159 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-11-12 18:40:04.166 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:40:04.167 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-11-12 18:40:09.171 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
[... 56 lines cut ...]
2024-11-12 18:41:49.245 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-11-12 18:41:54.251 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:41:54.252 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-11-12 18:41:54.252 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-11-12 18:41:54.253 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-11-12 18:42:09.258 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:42:09.259 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-11-12 18:42:14.263 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:42:14.264 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-11-12 18:42:19.271 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
2024-11-12 18:42:19.272 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2024-11-12 18:42:19.272 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2024-11-12 18:42:19.273 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

### Additional context

I'm not entirely sure when this happened but it was either as a result of an integration upgrade or Home Assistant upgrade. I have not been able to find an old backup where it is working I can ping the inverter and I'm pretty confident it is not a network issue. If I try to reinstall the integration it cannot find the inverter at the end IP address I give.

petr-slavik, 2024-11-14:
I have the same issue:

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed
SolaX: reading serial number from address 0x1a10 failed; other address may succeed

Logger: homeassistant.config_entries
Zdroj: config_entries.py:635
První výskyt: 11:37:52 (1 výskyty)
Naposledy logováno: 11:37:52

Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 188, in async_setup_entry
    await hub.async_init()
  File "/config/custom_components/solax_modbus/__init__.py", line 323, in async_init
    self._invertertype = await self.plugin.async_determineInverterType(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 7463, in async_determineInverterType
    seriesnumber                       = await async_read_serialnr(hub, 0x0)
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 561, in async_read_holding_registers
    await self._check_connection()
  File "/config/custom_components/solax_modbus/__init__.py", line 529, in _check_connection
    return await self.async_connect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 542, in async_connect
    result = await self._client.connect()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/tcp.py", line 88, in connect
    return await self.base_connect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 114, in base_connect
    return await super().connect()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 243, in connect
    self.transport, _protocol = await asyncio.wait_for(
                                ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1104, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1007, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 641, in sock_connect
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

petr-slavik, 2024-11-14:
For me it was caused by a network outage on the wifi dongle. I had to reconfigure the network. After that I reloaded the module in HA and everything works again.

wills106, 2024-11-16:
@motclem are you using a PocketWiFi?

motclem, 2024-11-16:
> @motclem are you using a PocketWiFi?
no ethernet

wills106, 2024-11-16:
If you use DHCP have you made sure your IP address hasn't changed?
Worth also power cycling the Inverter.

motclem, 2024-11-16:
> If you use DHCP have you made sure your IP address hasn't changed? Worth also power cycling the Inverter.

yes using dhcp will try power recycle tat the only thing I have not tried I will confirm if successful thank you :)

wills106, 2024-11-17:
Did a power cycle work?
Don't forget you need either PV production or energy stored in the batteries when reconnecting. If the Inverter is a sleep communications can't be initiated.

motclem, 2024-11-17:
> Did a power cycle work?
> 
> Don't forget you need either PV production or energy stored in the batteries when reconnecting. If the Inverter is a sleep communications can't be initiated.

Unfortunately power recycle didn't work

motclem, 2024-11-17:
> Did a power cycle work? Don't forget you need either PV production or energy stored in the batteries when reconnecting. If the Inverter is a sleep communications can't be initiated. apart from at night it there should always be PV production as far s I'm aware it is it asleep

Xellzul, 2024-11-18:
I had the same problem, today for second time. I did *fix* it by removing the Modbus component, restarting HA, restarting PocketWifi, recreating modbus component.

Getting the modbus data to PC did seem to work.

Using:
Solax X3-Hybrid-G4
Pocket WIFI V3.006.04
Inverter was in Idle-Mode

Error:
`2024-11-18 14:24:43.661 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-11-18 14:24:43.661 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.1.45:502
2024-11-18 14:24:43.735 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.1.45:502
2024-11-18 14:24:43.735 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2024-11-18 14:24:53.736 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-11-18 14:24:53.737 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-11-18 14:24:53.737 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-11-18 14:24:58.741 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-11-18 14:24:58.742 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-11-18 14:24:58.743 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-11-18 14:25:07.264 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 188, in async_setup_entry
    await hub.async_init()
  File "/config/custom_components/solax_modbus/__init__.py", line 323, in async_init
    self._invertertype = await self.plugin.async_determineInverterType(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 7477, in async_determineInverterType
    seriesnumber = await async_read_serialnr(hub, 0x1A10)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError`

motclem, 2024-11-18:
noticed this error this evening. after removing then trying to re add the inverter 
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:7479
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 00:02:14 (2716 occurrences)
Last logged: 18:52:21

SolaX: cannot find any serial number(s)

motclem, 2024-11-18:
thee are the first 6 digits of my s/n HCC5B2

motclem, 2024-12-15:
I can ping the inverter and solar cloud is getting data

motclem, 2024-12-15:
@petr-slavik  did you fix your issue?

greenertux, 2025-01-07:
I have a similar issue - any time I restart Homeassistant the integration is no longer working. My solution is to go to Solax Modbus configuration, delete my inverters and recreate them with the same name. This way old data will be preserved and it will start to get data again almost immediately. 
It is a lot of work to do this on any restart though...

closed 2025-09-17 by github-actions[bot]

infradom, 2025-10-21:
> I have the same problem. Nearly every restart of ha, Solax entities wont be available. Workaround is to replug the ethernet usb adapter on the inverter, then restart ha.

Why do you believe it is the same issue, you do not even provide your model name or connection method or any other details. Ethernet-usb adapter is too vague, the ethernet dongle does not even provide a modbus interface on the models I know ...

coaxx77, 2025-10-23:
I will provide more details:

Inverter:
X3-Hybrid-12kW
by SolaX Power
Firmware: DSP v1.41 ARM v1.40
Hardware: Gen4 

Lan Dongle (connected via Ethernet, no use of Wifi):
Solax Pocket V3.0 WiFi & LAN Dongle

SolaX Inverter Modbus
Version 2025.10.1