wills106/homeassistant-solax-modbus #1142: Solis Firmware 4B00

fboundy, 2024-11-20:
### Describe the bug

Solis firmware 4B00 introduces a completely new set of registers for time based control. It also adds a target SOC (and a voltage though it’s unclear what this is used for) to each slot.

The old registers are no longer read so the registers used need to depend on the FW version.

I’ve polled the API and Modbus and come up with this list: https://github.com/fboundy/pv_opt/issues/305#issuecomment-2489570654

### Integration Version

2024.11.2

### Homeassistant core version

2024.11.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

NA

### Connection Method

Waveshare

alienatedsec, 2024-11-21:
@wills106 Is it possible to introduce a switch that uses a new set of registers if, for example, another register represents a specific value? 

The input register hex value would have part of the firmware version.

The below `33002` is the HMI version - `4A` in my case (currently unaffected) 
[image]

This is usually followed with `00`, and combines with the register `33001` the DSP version - `4C` in my case - and `33069` both unrelated at this moment - or could be if DSP has any specific function defined.
[image]

FW version `33002` + `00` + `33001` + HMI subversion `33069`
[image]

So, the final logic should be that if the register `33002` is `0x004b` or higher, than the new set of registers should be used.

@fboundy @infradom @jmccrohan FYI

fboundy, 2024-11-21:
@alienatedsec - I think it might be easier just to use a separate plugin then all the logic will be very similar but a separate set of registers etc

fboundy, 2024-11-21:
Should be addressed by #1143

alienatedsec, 2024-11-26:
@fboundy 
I got the new firmware files from Solis, but I am delaying to load them. I also believe it's a newer version than 4B00, so I guess I will have to decide at some point.

fboundy, 2024-11-28:
I was told yesterday that 4B00 is still the latest

wills106, 2024-11-28:
If the Firmware was read at the same time as the Serial Numer then it should be possible to use the unused GEN block to split it up?

If lower than y = GEN if above x = GEN2?
Depends how the version numbering works as a whole really on Solis.

fboundy, 2024-12-03:
The differences between the f/w setups are so significant that I think it is simpler to simply use a different plugin as has been done for other brands

alienatedsec, 2024-12-04:
The firmware needs to be read in either plugin as it would be the condition for correct functionality. It's no longer just a Serial Number dependency.

MattSpeke, 2024-12-08:
> I was told yesterday that 4B00 is still the latest
> […](#)

I have version 4C0048 for inverter S5-EH15PK-L

MattSpeke, 2024-12-15:
Hi, is there any progess on this issue? I have the latest firmware and would like to control my Solis inverter via this integration. I have firmware 4C0050-650000 and serial number 6031 S5-EH1P5K-L inverter.

I have copied the code from fboundy and made a file solis_fb00_py and added the code to this file. I am able to select the file in HA config but no data available. is there something else that I need to do to enable this to work?

fboundy, 2024-12-15:
Best thing is to install the whole folder and then delete and reinstall the Solax integration

MattSpeke, 2024-12-15:
How do I install the whole folder? I only see the one file solis_fb00.py

I removed the integration, then tried to re-add, but the old names were still in the files, so I deleted the integration from HACS, but that has also deleted the solis_fb00.py file. I'm back to where I started.

fboundy, 2024-12-15:
Now you can replace the solax_modbus folder. Then you should be able to add the integration.

fboundy, 2024-12-23:
Closing this as it's now supported. Will add a new issue re firmware detection and merging the two plugins.

closed 2024-12-23 by fboundy