wills106/homeassistant-solax-modbus #1146: Minor Bugs in Growatt plugin

rmk-sys, 2024-11-24:
### Describe the bug

My System (GEN4) is 99,5% working well, including Time Slots (when not overlapping. a warning could be helpful, if possible), Inverter Switch is great, like the rest of the plugin :-)
But there are some minor issues left: 
1. System time (rtc) is 'unknown'. ('sync rtc' is working well)
2. System state (might be the same as 'run mode') returns all the times 'waiting'.
3. Firmware Version dispays AL1.0 instead of AL1.0/ALBA090101. 

### Integration Version

2024.11.2

### Homeassistant core version

2024.11.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

AL1.0ZABA

### Connection Method

R485 USB-Serial

rmk-sys, 2024-11-24:
4. In my humble opinion 'Export limit' cannot be negative, like the slider does.

PSMGoossens, 2025-01-12:
First issue is addressed in my PR will look into the other issues as well #1210

rmk-sys, 2025-01-13:
thx for the info and your work ;-))

wills106, 2025-01-14:
Just released [2025.01.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.7) with the first of the changes.

PSMGoossens, 2025-01-16:
@rmk-sys Could you check if issue 2 is also resolved for you? It seems to work in my case. Only the inverter state for me is still at waiting, but that is correct since I don't have any battery connected

[image]

rmk-sys, 2025-01-16:
it remains ‚waiting‘, although my battery is connected and running. (2025.1.8)

rmk-sys, 2025-02-04:
just discovered: systemtime (rtc) is flooding the log (every 15 sec), if activated.

run mode is working as exspected, but not system state:

[image]

rmk-sys, 2025-02-25:
hi, changing state class to 'TOTAL' in total_yield and today's_yield is solving the problem!

2025-02-25 09:35:10.512 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.gmod_today_s_yield from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 1.6 (1.7) with last_updated set to 2025-02-25T08:34:08.092082+00:00. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2025-02-25 09:35:10.513 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.gmod_total_yield from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 1657.1 (1657.2) with last_updated set to 2025-02-25T08:34:08.097059+00:00. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

flopp999, 2025-04-02:
I am working on growatt entities and maybe I can fix some other problems.
Are there still any errors from this Issue?

rmk-sys, 2025-04-03:
hello again, just tested and, yes the issues are still pending. and no, i don't have discord.
are you interestet in Growatt-Inverter-Modbus-RTU-Protocol-II-V1-24-English-new.pdf
and if you have a newer version, please link it.

rmk-sys, 2025-04-03:
[image]

flopp999, 2025-04-03:
1. so systemtime is OK
2. system state is what it should be. I don't know why state says Waiting. I would expect Bat online. When I check register input 3000 I get 1, which means Waiting, Normal. You cannot us web to check this register, you need direct access to your inverter with modbus to check reg input 3000
3. Firmware. Still problem?

[image]

flopp999, 2025-04-03:
> [@rmk-sys](https://github.com/rmk-sys) Could you check if issue 2 is also resolved for you? It seems to work in my case. Only the inverter state for me is still at waiting, but that is correct since I don't have any battery connected
> 
> [image]

I have MID 15 TL3-XH, two BMS connected and it still gives "Waiting", so I think this register we cannot not trust, at least not for state. Run seems to be OK.

flopp999, 2025-04-03:
@rmk-sys I am also checking 1.24 version, thanks anyway.
I tried to set the RTC to DATE but it still storing history data.
Did you try to change it to TOTAL and it worked?

rmk-sys, 2025-04-03:
no, i'm just testing today. the issues with the 'yield' probs i'd solved this way.

flopp999, 2025-04-03:
3. I am testing this.
2 new sensor, "Build Version" and "Communication Version" and I think I will combine the new sensors to the existing "Firmware Version" so it will be DN1.0/
I has also added Mode, I have no idea what it is but I found it in the PDF so why not add it :)

[image]

flopp999, 2025-04-04:
I have updated my code for this and seems to work.
3 Sensors, Build, Firmware and Communication
Firmware will show what Version are in the picture I uploaded, DNBA015101
@rmk-sys Close this Issue?

rmk-sys, 2025-04-04:
mode seems to be a special preset for the hardware and country, but I'm not sure about it.

yes, close it, if you like.

flopp999, 2025-04-04:
> mode seems to be a special preset for the hardware and country, but I'm not sure about it.
> 
> yes, close it, if you like.

I cannot close it, don't have permission. You can close issues that you created.

closed 2025-04-04 by rmk-sys