wills106/homeassistant-solax-modbus #1148: [Bug]: Incorrect sum of phases for "sensor.solax_inverter_current"

George-OPA, 2024-11-24:
### Describe the bug

The entity "sensor.solax_inverter_current" always has the same value as "sensor.solax_inverter_current_l1"? In my opinion, it should contain the sum of "sensor.solax_inverter_current_l1 + L2 + L3". Please let me know if this is an error. Thank you.
[image]

### Integration Version

2024.11.2

### Homeassistant core version

2024.11.2

### Inverter brand

X3-Hybrid

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

DSP v1.47 ARM v1.44

### Connection Method

SolaX WiFi + LAN

### Dongle firmware

1.003.11

simowNgithub, 2025-01-02:
same issue for the inverter power of all phases...

mbeetle, 2025-05-25:
This error is still present in the current version

[image]

mbeetle, 2025-07-27:
Still the same for me in actual Version

[image]