wills106/homeassistant-solax-modbus #1151: [Bug]: Solax X3-PRO-15K-G2.1: Incorrect values

mrichtarsky, 2024-11-29:
### Describe the bug

Some of the values seem to be incorrect/missing, please see the screenshot. At the time of the screenshot there was around ~400W grid import as checked at the meter.

[image]

From the screenshot:
- Grid Import is always 0, perhaps it should be removed?
- Grid Export resembles the generation power reported by the inverter and is fine
- Measured Power always is the same as Grid Export. It does not become negative.
- Measured Power L1-L3 seems to be the most puzzling. They somehow correspond to energy consumption, and can become negative. However, L1 is always 0. E.g. when EV charging with 11kW, L2 and L3 are at ~-3600W, L1 stays zero (and Measured Power, although it seems to refer to the same entity, also keeps reporting PV generation) 

Looking at the sources, the "Measured Power L1" seems to be read here:

https://github.com/wills106/homeassistant-solax-modbus/blob/a4453600cce0d79c654de25de25dbc4342368717/custom_components/solax_modbus/plugin_solax.py#L6407

However, when reading these values via pymodbus from the inverter, I always get 0 for all of L1, L2, L3. Only 0x40E returns a non-zero value. So I wonder how the values of L2 and L3 are obtained here, perhaps there is some fallback in place?

### Integration Version

2024.11.2

### Homeassistant core version

2024.11.3

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

MPT15T

### Inverter firmware versions

DSP v1.122 ARM v1.120

### Connection Method

Waveshare adapter

### Dongle firmware

V1. 486

mrichtarsky, 2024-11-29:
P.S.: Basically I'm looking for a method to determine the feed-in power of the inverter, so I can start charging once there is excess power. The inverter itself knows and displays the excess power, likely from looking at the data it gets from the connected Chint smart meter. Looking at the official Modbus documentation of the inverter, there does not seem to be a register that exposes this. But the values of "Measured Power L1-L3" definitely do correlate, so I wonder whether I could use them, but then I'm missing L1.

Interestingly the JSON API of the Pocket Wifi 3.0 dongle provides the feed-in power and I'm currently using that, but would prefer to get rid of the dongle.

mrichtarsky, 2024-11-30:
> The inverter itself knows and displays the excess power, likely from looking at the data it gets from the connected Chint smart meter. Looking at the official Modbus documentation of the inverter, there does not seem to be a register that exposes this.

I have found the "non-public" documentation (version MIC-G2/Pro-G2 V2.7) and this one includes Feed-in Power at 0x43B. Seems to be this one:

```
    SolaXModbusSensorEntityDescription(
        name = "Inverter Power",
        key = "inverter_power",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        ignore_readerror = True,
        newblock = True, # Do not remove, required for FW <1.38
        register = 0x43B,
        register_type = REG_INPUT,
        unit = REGISTER_S32,
        entity_registry_enabled_default = False,
        allowedtypes = MIC | GEN2,
        icon = "mdi:solar-power-variant",
    ),
```` 

In HA, this sensor is disabled for me. Should it be enabled?

Kars-de-Jong, 2024-12-06:
I have no idea how those Lx powers can be negative by the way, so maybe they do represent "measured power" on G2 inverters. On my G1 X3-MIC, when I add up their values they correspond to the total AC power of the inverter.

Kars-de-Jong, 2024-12-06:
I think I found the problem. The "Inverter Power Lx" registers are not enabled for Gen2. I will update my PR.

closed 2025-08-29 by wills106