wills106/homeassistant-solax-modbus #1155: [FR]: Add support for Senergy/MidNite The One MN15 Inverter

goldserve, 2024-12-04:
### Description

I have the modbus register spec and verified the inverter responds correctly and matches the registers. Should be fairly simple to create a definition for this inverter and add to this plugin.

### Suggested Solution

Add support for The One inverter

wills106, 2024-12-04:
If you want to share the docs I can have a look at adding it in, unless you want to make a start of creating a plugin for this Inverter?

goldserve, 2024-12-04:
It would take me quite a bit longer to feel my way around the code but you can tell me if the modbus spec is clear enough to code blind.

Cheers,
Kevin

wills106, 2024-12-04:
Did you send the document in an email reply?

As attachments including images don't work with email reply's. You need to add them through the website.

loxK, 2025-05-14:
https://www.midnitesolar.com/pdfs/classic_register_map_Rev-C5-December-8-2013.pdf