wills106/homeassistant-solax-modbus #1161: [Bug]: Battery 2 Temperature reads 0

gt008, 2024-12-07:
### Describe the bug

Hi with reference to the recent update 2024.11.1 and the fix SolaX Hybrid Gen5: Corrected "Battery 2 Temperature" 1050/1106  this is not working for me (although previous was working!)

My temperature on battery 2 reads 0 degC within the integration although on SolaxCloud.com it reads correctly for both batteries. I have recently updated the integration and I am now running 2024.11.2

[image]

What is strange is that previously I displaying the correct temperatures within the integration up until the 2nd Nov 24 when it stopped.

[image]

The only difference that I can see between my setup and rgrae81 is that I my batteries are H5.8 and his are HV-S3.6 - two different battery models although connected in the same manner to the same inverter model.

Also looking at the plugin there now seems to be two places where register 0x0132 is read for GEN5 inverters.

[image]

[image]

The reason I am interested in the battery temperature is that I have been running an automation that cycles the batteries when the temperature falls below 5 degC as outside temperature has been below freezing recently. 

Finally I have spoken to Solax and both the inverter and all the batteries are on the latest firmware (as of 5th Dec).

Thx. Giles

### Integration Version

2024.11.2

### Homeassistant core version

2024.12.1

### Inverter brand

Solax X3-Ultra-25kw Gen 5

### Plugin used

plugin_solax.py

### Serial prefix

H3BC 25k

### Inverter firmware versions

dsp 5 arm 18.2

### Connection Method

WIFI+LAN using LAN

### Dongle firmware

1.003.11

gt008, 2024-12-07:
Looking at the Modbus document that rgrae81 uploaded I realise that there are different register "groups" and I obviously do not really understand the data structure - so sorry if my last comment about two rgisters 0x0132 is misleading!! However I did notice that the version history of the document shows battery 2 temperature register to be 0x0131 in ...?? read registers group 4 ??

rgrae81, 2024-12-24:
OK, It must have something to do with Firmware-Version.

The latest Document I have is 3.37
At Changelog 2023.04.14 Bat2 Temp was on 0x0131 but changed to 0x0132
Also wEoutPowerTotal changed from 0x012F~0x0130 to 0x0130~0x0131

Maybe you could do a Firmware-Upgrade?
With DSP 7.1, ARM 9.7 from Feb 2024 it's working.

gt008, 2025-01-18:
hi @rgrae81 @wills106  just wondered if anyone able to get anywhere in finding the location for the battery 2 temperature?

closed 2025-09-17 by github-actions[bot]

wills106, 2025-10-01:
Just relooking into this as I'm trying to add in SolaX Aelio Inverters.
It's stating that Temp 2 Should be on 0x131

Looking at this Issue and #1050 I'm not sure if the current 0x132 is correct or if we should be using 0x131?