wills106/homeassistant-solax-modbus #1166: Growatt MID KTL3-XH

flopp999, 2024-12-10:
I just installed a second battery to my inverter.
As soon as I do that the Solax integration will stop fetching data.
Is this something you knew?
Maybe I can help you with something so you can implement a second battery

PSMGoossens, 2025-01-12:
@flopp999 Could you please provide the home assistant logging with debug log enabled for solax?

DMuenster, 2025-01-20:
Hello everyone,
I have another question about the same inverter. I have the MID30KTL-XH and I have 3 PV strings connected to it (3x MPPT). Unfortunately, only PV1 and PV2 are read out here. Is there a solution?

PSMGoossens, 2025-01-20:
@DMuenster What serial number do you see at the device? 
You could fine it in settings -> integration -> solax -> your device -> top left side
[image]

DMuenster, 2025-01-20:
@PSMGoossens 

Here is the corresponding photo

[image]

flopp999, 2025-03-04:
my mistake, today I tried again and now everything is fine. The problem was my VR and not this repo.

Now I see the SoC as a summary or maybe an average for my two batteries. Should I see SoC for battery 1 and SoC for battery 2 in different entites?