wills106/homeassistant-solax-modbus #1168: Stop working after HA updates

mc4wrk, 2024-12-11:
System stop working after HA updates. All sensors are gone. Retryed to reinstall 

Hardware
Solax inverter X1-Hybrid -G4 with USB PEN

Solax Modbus version 2024.11.2

HA
  Core 2024.12.2
  Supervisor 2024.11.4
  Operating System 14.0

Removed old Hub and recreate a new one does not work any more.

Any advice will be very welcomed.
Thank you

JohnS111, 2024-12-20:
I have exactly the same Solax equipment and HA configuration and the same problem. I too have removed and re-installed the Solax Modbus integration a few times. The Solax Modbus integration will no longer install and it gets stuck saying "initializing".

guy-BW, 2024-12-21:
Seeing this in the logs..

2024-12-21 10:22:02.659 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
        req, timeout=self.comm_params.timeout_connect
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-12-21 10:22:02.675 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-12-21 10:22:18.075 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-12-21 10:22:18.076 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 69, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 562, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
        req, timeout=self.comm_params.timeout_connect
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^

wills106, 2024-12-23:
Have you tried [unplugging the PocletWiFi?](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)
Also check if there is a newer firmware version for the PocketWiFi.

guy-BW, 2024-12-23:
Yes unplugging seems to have resolved this. Thanks