wills106/homeassistant-solax-modbus #1172: HA 2025 pymodbus 3.7.4 breaking change

PatrikTrestik, 2024-12-19:
### Description

In dev version of HA there is pymodbus upgreaded to version 3.7.4 (might be gigher).

There is breaking change referencing AsciiFramer.

Fix should be prepared and tested in advance before HA 2025 release.

### Suggested Solution

Upgrade pymodbus to 3.7.4 in this integration.

wills106, 2024-12-23:
I was aware of the change from #1113 but wasn't sure why that install kept pulling down 3.7.4 instead of 3.6.9 which HA is meant to be pinned at.
Will work on supporting 3.7.4 soon.

wills106, 2024-12-24:
Just released [2025.01.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.0b1) for anyone trying out HA 2025.1.0b

wills106, 2024-12-26:
Just released [2025.01.0b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.0b2) with the fixes from [2024.12.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.12.2)

wills106, 2025-01-05:
Closing as released [2025.01.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.2)

closed 2025-01-05 by wills106

Alfred99-HA, 2025-01-05:
Perfect fix in no time during weekend!
Thanks a lot