wills106/homeassistant-solax-modbus #1175: [Bug]: SolaX X3-MIC Gen1 broken after upgrade

Kars-de-Jong, 2024-12-24:
### Describe the bug

After upgrading from 2024.11.2 to 2024.12.1 the integration is completely broken for me.
All entities are unavailable.

### Integration Version

2024.12.1

### Homeassistant core version

2024.12.5

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

MU802T

### Inverter firmware versions

ARM 1.38-1.02 DSP 1.33

### Connection Method

USR-N510

### Dongle firmware

N/A

### Detailed Error Log

```
2024-12-24 08:18:18.496 INFO (ImportExecutor_0) [custom_components.solax_modbus.config_flow] detected HA core version 2024 12
2024-12-24 08:18:18.496 INFO (ImportExecutor_0) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
32m2024-12-24 08:18:18.691 INFO (SyncWorker_1) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2024-12-24 08:18:33.881 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-12-24 08:18:33.936 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at x.x.x.x:502
2024-12-24 08:18:33.936 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX X3-MIC: trying to determine inverter type
2024-12-24 08:18:34.107 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX X3-MIC: reading serial number from address 0x0 failed; other address may succeed
2024-12-24 08:18:34.107 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX X3-MIC 0x0 serial number: None
2024-12-24 08:18:34.282 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX X3-MIC 0x300 serial number: MU802T
2024-12-24 08:18:36.329 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None
2024-12-24 08:18:36.331 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x318 rtc
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x400 pv_voltage_1
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x401 pv_voltage_2
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x402 pv_current_1
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x403 pv_current_2
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x404 inverter_voltage_l1
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x405 inverter_voltage_l2
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x406 inverter_voltage_l3
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x407 inverter_frequency_l1
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x408 inverter_frequency_l2
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x409 inverter_frequency_l3
2024-12-24 08:18:36.332 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40a inverter_current_l1
2024-12-24 08:18:36.333 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40b inverter_current_l2
2024-12-24 08:18:36.333 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40c inverter_current_l3
2024-12-24 08:18:36.333 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40d inverter_temperature
2024-12-24 08:18:36.333 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40e measured_power
2024-12-24 08:18:36.333 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40f run_mode
2024-12-24 08:18:36.334 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x410 measured_power_l1
2024-12-24 08:18:36.334 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x411 measured_power_l2
2024-12-24 08:18:36.334 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x412 measured_power_l3
[... 180 lines cut ...]
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 373, in decode_16bit_uint
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes
2024-12-24 08:41:06.852 WARNING (MainThread) [custom_components.solax_modbus] SolaX X3-MIC: read failed at 0x425: today_s_yield 
2024-12-24 08:41:06.853 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 670, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 834, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "input")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 784, in async_read_modbus_block
    val = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 373, in decode_16bit_uint
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes
...
```

### Additional context

The inverter was in "Waiting" mode, which means some registers are unavailable. This worked fine with version 2024.11.2.

Haydn23, 2024-12-25:
Hi, I have same issue, all showing Unknown

wills106, 2024-12-25:
Looks like it's trying to setup two instances. If you delete the Inverter from the Integration and add it back in again does it load correctly?

Haydn23, 2024-12-25:
have deleted mine and reinstalled, still same issue here. all items showing unknown. Only happened since last update.

Haydn23, 2024-12-25:
2024-12-24 20:07:49.583 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x318 rtc
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x39a lock_state
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x320 mppt_scan_mode_pv1
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x35c q-curve
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x381 active_power_limit
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x390 firmware_arm
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x394 firmware_dsp
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x3a6 mppt_scan_mode_pv2
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x400 inverter_voltage
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x403 inverter_current
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x406 inverter_frequency
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x408 ct_power
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x409 measured_power
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40a pv_voltage_1
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40b pv_voltage_2
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40d pv_current_1
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40e pv_current_2
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x410 pv_power_1
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x411 pv_power_2
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x412 pv_power_3
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x413 inverter_temperature
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x414 control_board_temperature
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x415 run_mode
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x42b total_yield
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x42f total_grid_export
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x431 total_grid_import
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x437 today_s_yield
2024-12-24 20:07:52.182 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_rtc is already used by sensor.solax_rtc - ignoring sensor.SolaX_rtc
2024-12-24 20:07:52.182 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_inverter_voltage is already used by sensor.solax_inverter_voltage_2 - ignoring sensor.SolaX_inverter_voltage
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_inverter_current is already used by sensor.solax_inverter_current_2 - ignoring sensor.SolaX_inverter_current
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_inverter_frequency is already used by sensor.solax_inverter_frequency_2 - ignoring sensor.SolaX_inverter_frequency
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_ct_power is already used by sensor.solax_ct_power - ignoring sensor.SolaX_ct_power
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_measured_power is already used by sensor.solax_measured_power - ignoring sensor.SolaX_measured_power
[... cut ...]
  File "/config/custom_components/solax_modbus/__init__.py", line 834, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "input")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 784, in async_read_modbus_block
    val = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 373, in decode_16bit_uint
    return unpack(fstring, handle)[0]

CVKBaca, 2024-12-25:
Same problem for me. I did a rollback to version 2024.11.2

wills106, 2024-12-26:
Can you all try [2024.12.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.12.2)

Kars-de-Jong, 2024-12-26:
@wills106 Works for me now, thanks!

wills106, 2024-12-26:
Closing down as resolved.

closed 2024-12-26 by wills106