wills106/homeassistant-solax-modbus #1176: [Bug]: Non unique IDs

Thrasher2020, 2024-12-24:
### Describe the bug

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:773
integration: Sensor (documentation, issues)
First occurred: 20:38:38 (6 occurrences)
Last logged: 20:39:00

Platform solax_modbus does not generate unique IDs. ID SolaX_house_load is already used by sensor.solax_house_load - ignoring sensor.SolaX_house_load
Platform solax_modbus does not generate unique IDs. ID SolaX_pv_power_total is already used by sensor.solax_pv_power_total - ignoring sensor.SolaX_pv_power_total
Platform solax_modbus does not generate unique IDs. ID SolaX_remotecontrol_autorepeat_remaining is already used by sensor.solax_remotecontrol_autorepeat_remaining - ignoring sensor.SolaX_remotecontrol_autorepeat_remaining

### Integration Version

2024.12.2

### Homeassistant core version

2024.1.b1

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H4502A

### Inverter firmware versions

DSP v1.33 ARM v1.32

### Connection Method

Pocket wifi 3

### Dongle firmware

Am away from home and can't find it easily. 

### Detailed Error Log

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:773
integration: Sensor (documentation, issues)
First occurred: 20:38:38 (6 occurrences)
Last logged: 20:39:00

Platform solax_modbus does not generate unique IDs. ID SolaX_house_load is already used by sensor.solax_house_load - ignoring sensor.SolaX_house_load
Platform solax_modbus does not generate unique IDs. ID SolaX_pv_power_total is already used by sensor.solax_pv_power_total - ignoring sensor.SolaX_pv_power_total
Platform solax_modbus does not generate unique IDs. ID SolaX_remotecontrol_autorepeat_remaining is already used by sensor.solax_remotecontrol_autorepeat_remaining - ignoring sensor.SolaX_remotecontrol_autorepeat_remaining

Bl00d-B0b, 2024-12-25:
I have the same issue:
`2024-12-25 10:53:46.946 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_export_power_limit is already used by sensor.x3_8k_g2_export_power_limit - ignoring sensor.X3-8K-G2_export_power_limit
2024-12-25 10:53:46.946 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_voltage_1 is already used by sensor.x3_8k_g2_pv_voltage_1 - ignoring sensor.X3-8K-G2_pv_voltage_1
2024-12-25 10:53:46.946 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_voltage_2 is already used by sensor.x3_8k_g2_pv_voltage_2 - ignoring sensor.X3-8K-G2_pv_voltage_2
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_current_1 is already used by sensor.x3_8k_g2_pv_current_1 - ignoring sensor.X3-8K-G2_pv_current_1
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_current_2 is already used by sensor.x3_8k_g2_pv_current_2 - ignoring sensor.X3-8K-G2_pv_current_2
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_voltage_l1 is already used by sensor.x3_8k_g2_inverter_voltage_l1 - ignoring sensor.X3-8K-G2_inverter_voltage_l1
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_voltage_l2 is already used by sensor.x3_8k_g2_inverter_voltage_l2 - ignoring sensor.X3-8K-G2_inverter_voltage_l2
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_voltage_l3 is already used by sensor.x3_8k_g2_inverter_voltage_l3 - ignoring sensor.X3-8K-G2_inverter_voltage_l3
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_frequency_l1 is already used by sensor.x3_8k_g2_inverter_frequency_l1 - ignoring sensor.X3-8K-G2_inverter_frequency_l1
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_frequency_l2 is already used by sensor.x3_8k_g2_inverter_frequency_l2 - ignoring sensor.X3-8K-G2_inverter_frequency_l2
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_frequency_l3 is already used by sensor.x3_8k_g2_inverter_frequency_l3 - ignoring sensor.X3-8K-G2_inverter_frequency_l3
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_current_l1 is already used by sensor.x3_8k_g2_inverter_current_l1 - ignoring sensor.X3-8K-G2_inverter_current_l1
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_current_l2 is already used by sensor.x3_8k_g2_inverter_current_l2 - ignoring sensor.X3-8K-G2_inverter_current_l2
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_current_l3 is already used by sensor.x3_8k_g2_inverter_current_l3 - ignoring sensor.X3-8K-G2_inverter_current_l3
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_inverter_temperature is already used by sensor.x3_8k_g2_inverter_temperature - ignoring sensor.X3-8K-G2_inverter_temperature
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_measured_power is already used by sensor.x3_8k_g2_measured_power - ignoring sensor.X3-8K-G2_measured_power
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_run_mode is already used by sensor.x3_8k_g2_run_mode - ignoring sensor.X3-8K-G2_run_mode
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_power_1 is already used by sensor.x3_8k_g2_pv_power_1 - ignoring sensor.X3-8K-G2_pv_power_1
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_power_2 is already used by sensor.x3_8k_g2_pv_power_2 - ignoring sensor.X3-8K-G2_pv_power_2
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_pv_power_total is already used by sensor.x3_8k_g2_pv_power_total - ignoring sensor.X3-8K-G2_pv_power_total
2024-12-25 10:53:46.947 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID X3-8K-G2_total_yield is already used by sensor.x3_8k_g2_total_yield - ignoring sensor.X3-8K-G2_total_yield
[... cut ...]
  File "/config/custom_components/solax_modbus/__init__.py", line 834, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "input")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 806, in async_read_modbus_block
    if firstdescr.ignore_readerror != False:  # ignore block read errors and return static data
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'ignore_readerror'`

Haydn23, 2024-12-25:
Same here 

2024-12-24 20:07:49.583 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x318 rtc
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x39a lock_state
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x320 mppt_scan_mode_pv1
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x35c q-curve
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x381 active_power_limit
2024-12-24 20:07:52.170 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x390 firmware_arm
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x394 firmware_dsp
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x3a6 mppt_scan_mode_pv2
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x400 inverter_voltage
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x403 inverter_current
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x406 inverter_frequency
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x408 ct_power
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x409 measured_power
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40a pv_voltage_1
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40b pv_voltage_2
2024-12-24 20:07:52.171 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40d pv_current_1
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x40e pv_current_2
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x410 pv_power_1
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x411 pv_power_2
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x412 pv_power_3
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x413 inverter_temperature
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x414 control_board_temperature
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x415 run_mode
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x42b total_yield
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x42f total_grid_export
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x431 total_grid_import
2024-12-24 20:07:52.172 WARNING (MainThread) [custom_components.solax_modbus.sensor] input register already declared: 0x437 today_s_yield
2024-12-24 20:07:52.182 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_rtc is already used by sensor.solax_rtc - ignoring sensor.SolaX_rtc
2024-12-24 20:07:52.182 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_inverter_voltage is already used by sensor.solax_inverter_voltage_2 - ignoring sensor.SolaX_inverter_voltage
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_inverter_current is already used by sensor.solax_inverter_current_2 - ignoring sensor.SolaX_inverter_current
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_inverter_frequency is already used by sensor.solax_inverter_frequency_2 - ignoring sensor.SolaX_inverter_frequency
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_ct_power is already used by sensor.solax_ct_power - ignoring sensor.SolaX_ct_power
2024-12-24 20:07:52.183 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_measured_power is already used by sensor.solax_measured_power - ignoring sensor.SolaX_measured_power
[... cut ...]
  File "/config/custom_components/solax_modbus/__init__.py", line 834, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "input")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 784, in async_read_modbus_block
    val = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 373, in decode_16bit_uint
    return unpack(fstring, handle)[0]

wills106, 2024-12-26:
Can you all try [2024.12.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.12.2)

Haydn23, 2024-12-26:
Thanks for your quick work on a fix. have downloaded 2024.12.2 and now showing all the data. Thank you very much

wills106, 2024-12-26:
Closing down as resolved.

closed 2024-12-26 by wills106