wills106/homeassistant-solax-modbus #1177: SolaX x1 bost 3,6kw g4

LxMoura, 2024-12-26:
### Describe the bug

Com a última actualização deixou de apresentar dados - todos os sensores “desconhecido”
Qual a solução ???

### Integration Version

 2024.12.1

### Homeassistant core version

2024.12.5

### Inverter brand

SolaX power

### Plugin used

plugin_solax.py

### Serial prefix

XB4A36JC019009

### Inverter firmware versions

034-000

### Connection Method

Wi-Fi 

### Dongle firmware

Wi-Fi

ertho22, 2024-12-26:
Same problem.
I‘m using the LAN/WLAN-Dongle.

wills106, 2024-12-26:
Can you all try [2024.12.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.12.2)

ertho22, 2024-12-26:
> Can you all try [2024.12.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.12.2)

Works perfect for me, thank you.

wills106, 2024-12-26:
Closing down as resolved.

closed 2024-12-26 by wills106