wills106/homeassistant-solax-modbus #1179: [Bug]: 'Growatt' - '10kTL3-XH'  'NoneType' object has no attribute 'split'

Mwirenfeldt, 2024-12-26:
### Describe the bug

Hi Guys

i am trying to make my inverter charge from the grid with a automation from Home Assistant
but when i press the "Update Time Slot" button i get an error.

Log is attached. 

### Integration Version

2025.01.0b2

### Homeassistant core version

2025.1.0b2

### Inverter brand

Growatt 10kTL3-XH

### Plugin used

plugin_growatt.py

### Serial prefix

065101

### Connection Method

Waveshare POE Adapter

### Detailed Error Log

2024-12-26 23:20:33.873 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140167545874048] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1006, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1078, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 132, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 88, in async_press
    res = self.button_info.value_function(0, self.button_info, self._hub.data )
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 122, in value_function_time_slot_1
    time_2_begin = time_to_int(time_2_begin)
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 101, in time_to_int
    hours, minutes = map(int, time_str.split(':'))
                              ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'split'

Mwirenfeldt, 2024-12-27:
Found the error. 

Time_2_Begin, Time_2_End, Time_3_Begin and Time_2_End was not definded. after i had definded the hours and minuts i was able to update the time slots.

TonyMathiesen, 2025-04-07:
The error message if some of the time 2 or 3 entities is deactivated, will be fixed with #1321

wills106, 2025-04-26:
Just released [2025.04.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.2) with this change.

closed 2025-09-17 by github-actions[bot]