wills106/homeassistant-solax-modbus #1180: [Bug]: Integration not working with any version after 2024.09.5 / all entities unknown

ggliniak, 2024-12-27:
### Describe the bug

Hi,
after upgrading to 2024.10.1 this integration stopped working in my installation. Absolutely no changes to my modbus or inverter configuration.
I waited some weeks and did a try to each update, also to the newest 2024.12.2 yesterday, same behavior.
(did all the known steps like removing integration completely, restarting HA, doing it when inverter is well PV powered, and so on)

Modbus communiation is working rock-solid (Schneider Electric PAS600 as modbus gateway).
Each time after downgrading to 2024.09.5 integration is starting to work immediately again (although in night time when inverter is asleep).

Noticed that newer versions (>2024.09.5) can read my serial via modbus but are NOT reading the full firmware informations, see attached screenshots.

Is this a known behavior and I was not able to find the solution it here in the documentation? If yes, please give me a hint! ;-)
Or is this a new situation?

Best regards, Georg

### Integration Version

2024.12.2

### Homeassistant core version

2024.12.5

### Inverter brand

Solax X3-MIC Gen2

### Plugin used

plugin_solax.py

### Serial prefix

MC206T

### Inverter firmware versions

ARM v1.111 DSP v1.115

### Connection Method

Modbus via external gateway (Schneider Electric PAS6000)

### Dongle firmware

Schneider Electric PAS6000 Panel Server (modbus gateway)

### Detailed Error Log

[image]
[image]

Reinhard-M, 2024-12-28:
The same applies to me. The hardware and serial number are still recognized, but the FW versions and all settings or values ​​are missing.

Reinhard-M, 2025-01-07:
Update:
It is possible to change registers or switch the system on and off via "Solax Modbus". Tested, for example, with the PGRIDBIAS register. However, it is not possible to read any data, not even the registers that were changed for test purposes.

dirk79, 2025-01-12:
@wills106
Thank you for the amazing work with this integration!
Unfortunately I can't use the September release since 2025 HA Core is installed.
Do you have any idea how we could resolve this?

ggliniak, 2025-01-12:
Hi @wills106
I would like to join in the thanks! :)
Seems something is broken with GEN2 in plugin_solax.py

I searched a couple of time now with a diff between _2024.09.5/plugin_solax.py_ and _2025.01.6/plugin_solax.py_ but I am not able to find the affecting position in code. :(

Just did a small test and changed  "**GEN2**" to "**GEN**" at this position in plugin_solax.py:

```
        elif seriesnumber.startswith('MC206T'):
            invertertype = MIC | GEN | X3 # MIC X3
            self.inverter_model = "X3-MIC"
```

Most of my entities came back working again after restart.
But I can not really understand why this works, because in _2024.09.5_ (the last out-of-the-box working version in my environment) the definition for "MC206" already was "**GEN2**"...

Maybe you or somebody can assist a bit, problem solution can not be far away. :)

Best regards

dirk79, 2025-01-12:
Crazy thing Georg!
I did the same change and about 90% of the values are available now. 😀
Hopefully this helps Wills106 to fix this soon 👍

ggliniak, 2025-01-12:
@dirk79 
yes, checked it again but the entities are a bit wild sorted. Run mode and Temp is fine, some other seem to have wrong scale or something. Also my firmware and hardware version is shown again, but completely different to the values shown in last out-of-the-box running version (2024.05.9) in my environment. ;)

dirk79, 2025-01-12:
@wills106 
If it helps, here are my Inverter details:
Model: X3-MIC-5K-G2
Serial starts with: MC205T...

ggliniak, 2025-01-12:
@dirk79

ggliniak, 2025-01-13:
With some help and some dirty code changes I got it back running again with .
Maybe our problems depend on old firmware or any other limitations in MIC GEN2... 

I learned that it is not possible to read a very big register block on my inverter like the Integration is doing in the actual version, so I splitted them a bit and also there seems that "Export Power Limit" is not readable on my device, also not manually. That caused the unknown entities du to a modbus exeption.

diff Downloads/homeassistant-solax-modbus-2025.01.6/custom_components/solax_modbus/plugin_solax.py plugin_solax.py

```
6385a6386,6387
> 	newblock=True,
> 	ignore_readerror=True,
6392a6395,6396
> 	newblock=True,
> 	ignore_readerror=True,

```

dirk79, 2025-01-13:
Thank you for sharing! I'm on a business trip till the end of the week. I'll give it a try when I'm back.

yuso08, 2025-01-15:
> Hi @wills106 I would like to join in the thanks! :) Seems something is broken with GEN2 in plugin_solax.py
> 
> I searched a couple of time now with a diff between _2024.09.5/plugin_solax.py_ and _2025.01.6/plugin_solax.py_ but I am not able to find the affecting position in code. :(
> 
> Just did a small test and changed "**GEN2**" to "**GEN**" at this position in plugin_solax.py:
> 
> ```
>         elif seriesnumber.startswith('MC206T'):
>             invertertype = MIC | GEN | X3 # MIC X3
>             self.inverter_model = "X3-MIC"
> ```
> 
> Most of my entities (and the Hardware/Firmware Version) came back working again after restart. But I can not really understand why this works, because in _2024.09.5_ (the last out-of-the-box working version in my environment) the definition for "MC206" already was "**GEN2**"...
> 
> Maybe you or somebody can assist a bit, problem solution can not be far away. :)
> 
> Best regards

This worked for me thanks

dirk79, 2025-01-20:
In my case (serial starting with MC205T) it seems like the value mapping is mixed up. The values and their description doesn't make sense in most cases. At least when I change GEN2 to GEN values show up at all.
@wills106  Is there any chance, that this may be fixed in  a future update? Since this worked fine until 09/2024 I am hopefully looking forward :)

Best, 
Dirk

dirk79, 2025-01-27:
Hello together,

since there seems no progress, I invested some time and could implement the most necessary values for me and I could deinstall the Solax integration.

Value for current power is no thing since it is just reading  from register.
Todays yield and Total yield uses two registers each which made some try&error necessary. I ended up in reading all registers and using templates to calculate the correct values.

For who is interested in the solution, here is the yaml code:
```
modbus:
  - name: Solax
    type: serial
    port: /dev/ttyUSB0
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    stopbits: 1
    sensors:
      - name: Solax Momentanleistung
        address: 0x40e
        unit_of_measurement: W
        data_type: uint16
        input_type: input
      - name: totalenergy.lsb
        address: 0x423
        scale: 0.1
        precision: 1
        data_type: uint16
        input_type: input
      - name: totalenergy.msb
        address: 0x424
        scale: 0.1
        precision: 1
        data_type: uint16
        input_type: input
      - name: todayenergy.lsb
        address: 0x425
        scale: 0.1
        precision: 1
        data_type: uint16
        input_type: input
      - name: todayenergy.msb
        address: 0x426
        scale: 0.1
        precision: 1
        data_type: uint16
        input_type: input

template:
  - sensor:
      - name: "Solax Gesamtertrag"
        unit_of_measurement: "kWh"
        state: >
          {% set lsb = states('sensor.totalenergy_lsb') | float %}
          {% set msb = states('sensor.totalenergy_msb') | float %}
          {{ (msb * 65536 + lsb) | round(1) }}
  - sensor:
      - name: "Solax Tagesertrag"
        unit_of_measurement: "kWh"
        state: >
          {% set lsb = states('sensor.todayenergy_lsb') | float %}
          {% set msb = states('sensor.todayenergy_msb') | float %}
          {{ (msb * 65536 + lsb) | round(1) }}

``` 

If you need to alter the registers or expand them I have a good experience b just asking Solax for the register documentation .

Best,
Dirk

closed 2025-09-17 by github-actions[bot]