wills106/homeassistant-solax-modbus #1186: [Bug]: 'Sofar' - 'HYD10KTL-3P 2nd' EPS Mode

muzzy81, 2024-12-31:
### Describe the bug

Hello, i can't change the EPS mode. A few seconds after selecting another mode it changes back to the previous Mode. The only way is on the inverter display. I use latest ha with latest solax with lse-3 and also waveshare.... same issue.

### Integration Version

Latest

### Homeassistant core version

Latest 

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES11

### Connection Method

Lse-3 and waveshare 

### Detailed Error Log

Logger: pymodbus.logging
Quelle: runner.py:154
Erstmals aufgetreten: 22:29:55 (580 Vorkommnisse)
Zuletzt protokolliert: 23:04:47

General exception: unpack requires a buffer of 2 bytes
Failed to connect
Failed to connect [Errno 113] Connect call failed ('172.16.6.213', 502)

cschlipf, 2025-02-03:
See also #1197 and the updated [FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/)

We can no longer recommend the LSE-3.

maxirass, 2025-02-23:
@cschlipf I have the same problem while using the Waveshare RS485 to Eth.
I can change the usage mode but not deactivating the EPS Mode.
Another question can I change the values of Depth of Discharge for the sofar as well?

cschlipf, 2025-02-26:
OK, I might have been too fast with that. WIll check it, but I am currently down with fever, so please allow me some time.

As for the second question I will have to update the FAQ as I explained that so often.

No, we won't support it due to the awkward protocol of Sofar, where you have to write many registers in batch. In this case just to update the DoD, we need to write 23 registers containing critical battery parameters, just to change the DoD. This is way too dangerous. See #703 for more details.

maxirass, 2025-02-28:
@cschlipf Oh no, Get well soon!

I tried the last day using automations and the passive mode to out play the dod. But unfortunately the battery Output always stops at a soc of 40%, because of the dod of 60%. Do you know to fix this? Is it necessary to deactivate the EPS for a proper function?

Thanks a lot for this awesome integration!

cschlipf, 2025-03-02:
Hm... I had a different use case in mind, but I see your problem.

You would have to set the DoD to 80% and stop discharging via HA and passive mode at 60% for EPS. Then if power fails, you have to enable the battery again to discharge for EPS. But that last part is what breaks it: In case of a power failure you cannot update the passive mode fast enough to emulate an EPS mode and very likely the power failure will prevent you from controlling this entirely as your HA installation will likely depend on power.

Hm... I don't know how to solve it. Blame the Sofar engineers for their awkward protocol.

maxirass, 2025-03-04:
@cschlipf I got the point, changing the DoD via solax-modbus is too complicated and risky.
For me it will be helpfull enough to deactivate EPS in the winter via solax-modbus for providing the energy only to huge consumers

cschlipf, 2025-03-05:
I am a little bit surprised of this 'Bug'. I see that @fboundy has added a register for EPS Control, but obviously he ignored that this is one of the registers that needs to be written in batch:

[image]

@fboundy - did you ever test this after adding this register?

fboundy, 2025-03-06:
Sorry - nothing to do with me. All my edits were for Solis inverters

cschlipf, 2025-03-12:
OK, probably that always was wrongly implemented, but as I never activated EPS mode I didn't notice it.

cschlipf, 2025-03-16:
This is fixed now. Note that you have to press the button 'EPS: Update" after changing the value.

[image]

closed 2025-03-24 by wills106

wills106, 2025-04-02:
I have released this change as [2025.04.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.04.1)

PR #1290 (2025-03-24): Sofar: Fix setting EPS Mode
Fix for setting the EPS Mode. Requires a new commit button as this is just another bulk write.

[image]

Closes #1186
commit 75eb1c19 (2025-03-24): Sofar: Fix setting EPS Mode

Fix for setting the EPS Mode. Requires a new commit button as this is just another bulk write.

Closes #1186