wills106/homeassistant-solax-modbus #1188: [Bug]: 'Sofar Solar' - 'HYD 15KTL-3PH' incorrect sensor values

aerucu, 2025-01-02:
### Describe the bug

Incorrect sensor values. 
[image]
[image]
[image]
[image]

### Integration Version

2024.12.2

### Homeassistant core version

2024.12.5

### Inverter brand

Sofar Solar

### Plugin used

plugin_sofar.py

### Serial prefix

SP1ES1

### Connection Method

Solarman LSE-3

### Detailed Error Log

2024-12-29 19:23:00.343 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:03.415 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:03.417 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:06.487 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:09.559 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:12.631 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:15.703 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:15.704 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:18.775 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:21.847 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:24.919 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:24.920 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:27.991 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:31.063 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:34.135 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:34.137 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:37.207 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:40.279 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:43.351 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:43.352 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:46.423 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-29 19:23:49.495 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.17.29:8899
2024-12-31 08:54:12.530 WARNING (MainThread) [custom_components.solax_modbus] Sofar: read failed at 0x445: serial_number 
2024-12-31 08:54:12.530 WARNING (MainThread) [custom_components.solax_modbus] Sofar: read failed at 0x44d: hardware_version 
2024-12-31 08:54:12.530 WARNING (MainThread) [custom_components.solax_modbus] Sofar: read failed at 0x44f: software_version 
2024-12-31 14:50:10.226 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.sofar_solar_generation_today from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 3.29 (3.53) with last_updated set to 2024-12-31T13:47:38.920504+00:00. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2025-01-01 12:25:10.227 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.sofar_export_energy_total from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 7615.6 (7615.7) with last_updated set to 2025-01-01T11:24:32.958583+00:00. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

d4ystek, 2025-01-05:
Hello
did you fix the problem? Same for me - incorrect entity values. My inverter is sofar ktlx 5.5 and logger LSE-3.
Regards

aerucu, 2025-01-10:
Hello @d4ystek, unfortunately, the problem has not been solved. For the time being, I am using my LSW-3 with the Solarman integration again.

konwin, 2025-01-18:
Same here with HYD20KTL-3PH and LSE-3 on ME_0D_270A_1.09 and SolaX Inverter Modbus 2025.01.08.
I fist installed 2024.12.x on December 26th and with that Version it was running flawlessly.
Maybe some UTC/timezone conversion Problem?
Is there a safe way to downgrade the SolaX Inverter Modbus?

aerucu, 2025-01-18:
@konwin, I was recommended to use version 2025.01.6. I was also recommended to increase the polling intervals. I had set 10s everywhere.

You can downgrade by downloading Solax Modbus Integration again under HACS. There you can also select the version you want to downgrade to in the drop-down menu. 

According to @cschlimpf, version 2025.01.8 uses a newer version of pyModbus, which is probably causing this problem.

konwin, 2025-01-18:
Downgrading did not work. I could install 2014.12.02 of SolaX Inverer Modbus but when trying to configure it I get:
```
Error

Config flow could not be loaded: {"message":"Invalid handler specified"}
```
Maybe this is related to the pymodbus version?! I think the newer version is still active in the background and 2024.12.02 won't start with this newer version.
I couldn't find a how to downgrade pymodbus for now...

zollak, 2025-01-29:
Hi guys! Do you have any updates? Same here with my  HYD5KTL-3PH and LSE-3 on ME_0D_270A_1.11 and SolaX Inverter Modbus 2025.01.08. Recently, I have downgraded it to v2025.01.06.

cschlipf, 2025-01-29:
@zollak Version `ME_0D_270A_1.11` is known to have issues. See https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/#i-am-using-the-lse-3-logger-stick-but-requests-timeout-frequently

aerucu, 2025-01-29:
@zollak, fill out the service contact form at the bottom of the support page and ask them if they can downgrade the firmware to v1.09. In my case, the firmware was downgraded the next morning. 

However, you must provide the serial number of the logger.

[https://www.solarmanpv.com/supportservice/service-contact/](https://www.solarmanpv.com/supportservice/service-contact/)

zollak, 2025-02-01:
> [@zollak](https://github.com/zollak), fill out the service contact form at the bottom of the support page and ask them if they can downgrade the firmware to v1.09. In my case, the firmware was [...]

It worked. They have made the downgrade within 3 days. Interesting, when I have restarted the inverter it caused invalid value for battery discharge and I had to repair the statistics...

aerucu, 2025-02-02:
@cschlipf Have you already been able to test v2025.01.9? Has the problem with the pyModbus been fixed in this version?

wills106, 2025-02-02:
The problem is pyModbus / my ability to attempt to fix it as I don't use a Data Logger and I never have connection problems or spikes etc, so there is nothing for me to personally fix.

Have a read https://github.com/wills106/homeassistant-solax-modbus/issues/1238#issuecomment-2629374562

Also @cschlipf is starting to update the docs to move away from Data Loggers like the LSE-3 #1244

cschlipf, 2025-02-02:
Like @wills106 said. I was always a supporter of the LSE-3. However recently more and more problems are appearing. It started with the wrong return codes, then issues with how the IP address is assigned, then pyModbus became stricter (which is not a bad thing per se, as there is a standard that is badly implemented by the LSE-3), then SolarMan updates started to break the communication (which never was an officially supported feature). See the FAQ that is listing more and more issues with the LSE-3: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/

Yesterday I went back to RS485 and I must say, once properly terminated it's definitely worth to give it a try. Setting values is so much faster without errors.

aerucu, 2025-02-02:
Is this the USB to RS485 converter that is one of the best and most reliable USB adaptors?

https://eckstein-shop.de/WaveShare-Industrial-USB-to-RS485-Bidirectional-Converter?ws_oss_lieferland=DE&srsltid=AfmBOopoPOl-whJexPadTULqgZaowQ7jDVfkmTKY6SCEyM6DwIHPPkLgi60[https://eckstein-shop.de/WaveShare-Industrial-USB-to-RS485-Bidirectional-Converter?ws_oss_lieferland=DE&srsltid=AfmBOopoPOl-whJexPadTULqgZaowQ7jDVfkmTKY6SCEyM6DwIHPPkLgi60](url)

cschlipf, 2025-02-02:
At least this is the one I am using ;-)

aerucu, 2025-02-02:
@cschlipf, does this converter require a 120 Ohm resistor to be connected to pins 2 and 3 or does the 120R terminal resistor built into the WaveShare converter take care of this?

cschlipf, 2025-02-02:
No, the adaptor does not need a termination - it's built in. You just need one at the inverter side as you can see in the updated installation doc: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-installation/

aerucu, 2025-02-02:
My inverter was supplied with 109 GOhm resistors with 5% tolerance. Color coding brown, black, white, white, gold. Are these also suitable or should I buy a 120 Ohm resistor?

cschlipf, 2025-02-02:
109 GOhm? Definitely way too much.

aerucu, 2025-02-02:
So, I have now ordered the WaveShare USB to RS485 converter with FT232RNL chip and 50x 120 Ohm resistor. All together less than 18 Euro. That's much cheaper than the LSE-3 logger.

But I can still use the LSE-3 to send the data to Solarman, right?

Thanks for your support.

cschlipf, 2025-02-02:
Absolutely. I also keep the LSE-3 as I am more comfortable with a LAN connection than everything WIFi.

aerucu, 2025-02-02:
That means USB to RS485 converter and LSE-3 logger can run side by side on the inverter without interfering with each other?

cschlipf, 2025-02-02:
Yes, absolutely. You could also still access the LSE-3 modbus interface, while also using the RS485 interface for other purposes in parallel.

aerucu, 2025-02-09:
@cschlipf @wills106 
Am I doing something wrong or is the adapter broken?

cschlipf, 2025-02-09:
Your COM Port looks totally different than mine. Very likely you have a very different inverter and connector with totally different connections. Please compare your connector with the one in the doc: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-installation/#connect-an-rs485-usb-or-ethernet-adaptor

Have you checked whether pin 1-4 on your COM port connector have the same functions?

aerucu, 2025-02-09:
This is my inverter and according to page 62 the PIN assignment also fits.

[HYD 5-20KTL-3PH_User Manual_mit SP-14-stelligem SN_2022-11-01_V2.0_de-DE.pdf](https://github.com/user-attachments/files/18724659/HYD.5-20KTL-3PH_User.Manual_mit.SP-14-stelligem.SN_2022-11-01_V2.0_de-DE.pdf)

aerucu, 2025-02-09:
This is the page from the supplied user manual. It looks a little different there. The PIN assignment for RS485 is correct, but there is nothing about a 120 Ohm resistor.

[image]

aerucu, 2025-02-09:
@cschlipf @wills106
I have now also swapped the wiring, i.e. pin 1 to B- and pin 4 to A+, with resistor, without resistor but nothing has helped. The RS5485 symbol simply does not appear on the display.

cschlipf, 2025-02-09:
> I removed the resistor but that didn't help either.
> 
> How can I check whether the name of the serial port “/dev/ttyUSB0” matches my inverter?

The serial port has nothing to do with your inverter. It's the port, where the USB adaptor is plugged in to your computer (e.g. the raspberry PI). Run ls /dev/ttyUSB* to see, which inverter it is.

One question: You have the adaptor with the FT232RL chipset and NOT the CH343G chipset, right?

aerucu, 2025-02-09:
> As you can see in the wiring diagram you can serially connect multiple inverters after each other. So the last one definitely needs a termination. And yes - that's really badly described in the manual.

I agree with you. That's why I put the resistor back on. But it still doesn't work. I even removed the logger once, but that didn't help either.

aerucu, 2025-02-09:
> One question: You have the adaptor with the FT232RL chipset and NOT the CH343G chipset, right?

Yes, according to the article description it is the one with the FT232RL. The adapter with the CH343G chip is cheaper and also has a (B) printed on it.

https://eckstein-shop.de/WaveShare-Industrial-USB-to-RS485-Bidirectional-Converter-CH343G

cschlipf, 2025-02-09:
I assume the problem is on the computer side. How did you install HA? Where did you install HA? If HA was installed as a docker container, have you made /dev/ttyUSB0 available to the docker container?

aerucu, 2025-02-09:
My Home Assistant runs on an HP Thin Client T655 with HA OS.

But I have just solved the problem. I changed /dev/ttyUSB0 to /dev/ttyUSB1 and it worked fine. Presumably the address /dev/ttyUSB0 is already occupied by my Sonoff Zigbee stick.

aerucu, 2025-02-09:
Thanks for the crucial hint.

cschlipf, 2025-02-09:
Great to hear. Do you see correct sensor values now? Can this issue be closed?

wills106, 2025-02-09:
You might want to use [/dev/serial/by-id instead of /dev/ttyUSB*](https://homeassistant-solax-modbus.readthedocs.io/en/latest/modbus-adaptor-setup/#usb-to-rs485-adaptor)

aerucu, 2025-02-09:
> Great to hear. Do you see correct sensor values now? Can this issue be closed?

Yes, my LSE-3 logger and the RS485 adapter show the same number of entities. The issue can be closed.

@wills106 I'll keep that in mind.

aerucu, 2025-02-09:
One last question. I have now set up the inverter as RS485 in evcc.yaml. But I get the following errors. What am I doing wrong?

```   
  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv

    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8N1" 
```

evcc error log:
```
[site ] ERROR 2025/02/09 20:43:22 pv 1 power: read failed: unexpected EOF
[site ] ERROR 2025/02/09 20:43:58 pv 1 power: read failed: failed to read from serial port within deadline
[site ] ERROR 2025/02/09 20:44:25 pv 1 power: read failed: resource temporarily unavailable
[site ] ERROR 2025/02/09 20:44:55 pv 1 power: read failed: resource temporarily unavailable
[site ] ERROR 2025/02/09 20:45:25 pv 1 power: read failed: modbus: response data size '56' does not match request quantity '1'
[site ] ERROR 2025/02/09 20:45:55 pv 1 power: read failed: resource temporarily unavailable
[site ] ERROR 2025/02/09 20:46:25 pv 1 power: read failed: unexpected EOF
[site ] ERROR 2025/02/09 20:46:58 pv 1 power: read failed: failed to read from serial port within deadline
[site ] ERROR 2025/02/09 20:47:25 pv 1 power: read failed: modbus: response data size '56' does not match request quantity '1'
[site ] ERROR 2025/02/09 20:47:55 pv 1 power: read failed: resource temporarily unavailable
[site ] ERROR 2025/02/09 20:48:25 pv 1 power: read failed: resource temporarily unavailable
[site ] ERROR 2025/02/09 20:48:58 pv 1 power: read failed: failed to read from serial port within deadline
[site ] ERROR 2025/02/09 20:49:29 pv 1 power: read failed: failed to read from serial port within deadline
[site ] ERROR 2025/02/09 20:49:55 pv 1 power: read failed: modbus: response data size '56' does not match request quantity '1'
[site ] ERROR 2025/02/09 20:50:26 pv 1 power: read failed: modbus: response crc '495' does not match expected '5615'
[site ] ERROR 2025/02/09 20:50:55 pv 1 power: read failed: unexpected EOF
```

[image]

cschlipf, 2025-02-09:
You can connect only one of the services to the serial port. You can do the same I did: Connect EVCC to the serial port, and configure a ModBus proxy with EVCC.

Add this to the `evcc.yaml`.

```yaml
modbusproxy:
  - port: 5021
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: "8N1"
```

Then in HA configure a ModBus TCP connection to the host, where EVCC is running to the port 5021:

[image]

aerucu, 2025-02-09:
It works very well with evcc as a Modbus proxy. I was even able to update to 2025.01.9 without any dropouts.

But the data is fetched from the inverter every 30s. That's why I'd rather do it the other way around. But as far as I have seen, the HA Modbus proxy does not support RS485. Or how would you set that up?

cschlipf, 2025-02-10:
I can't tell you how it's done the other way round.

However you can decrease the EVCC interval. However I don't think that the interval controls how often HA receives data as HA can ask for way more registers than EVCC is querying. But just an assumption. I don't really know how the ModBus Proxy in EVCC really works. In a few tests I saw responses faster than my EVCC interval however. Could have been a coincidence though.

aerucu, 2025-02-10:
I learned from the evcc developers that the Modbus proxy in evcc does not cache. Instead, the requests from the slaves are simply forwarded to the master and the responses are sent to the slaves. This means that the query interval is determined by the slaves.

cschlipf, 2025-02-10:
Thanks! Good to know. So there is no reason not to use the EVCC modbus proxy.

aerucu, 2025-02-10:
You'r welcome and you're right.

closed 2025-02-10 by aerucu