wills106/homeassistant-solax-modbus #1189: [Bug]: HA 2025.01 breaks the integration

timotoots, 2025-01-03:
### Describe the bug

After update, the integration is not loaded on boot.

### Integration Version

2024.12.2

### Homeassistant core version

2025.1

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H34C15

### Inverter firmware versions

 X3-Hybrid  Gen4

### Connection Method

PocketWifi

### Dongle firmware

PocketWifi

### Detailed Error Log

Logger: homeassistant.setup
Source: setup.py:334
First occurred: 23:06:24 (1 occurrences)
Last logged: 23:06:24

Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

### Additional context

Logger: homeassistant.util.loop
Source: util/loop.py:77
First occurred: 23:06:24 (1 occurrences)
Last logged: 23:06:24
Detected blocking call to import_module with args ('custom_components.solax_modbus',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

matomatusov, 2025-01-03:
Hi, I have a similar problem. I had to return HA to 2024.12.5.

felixijc, 2025-01-03:
This release works: https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.0b2

funeve, 2025-01-03:
Same problem, installed via HACS  - is there an ETA for the update to go live?

felixijc, 2025-01-03:
Be aware this is still a beta version:
1.     Click on the integration card in HACS.
2.     Click on the three dots icon top right of the page.
3.     Select Redownload.
4.     Choose the version you want from the dropdown list.
5.     Click Download.

funeve, 2025-01-04:
Thanks felix, about 4 days in with HA.

Just incase anyone else with solis inverters is having an issue with this verision

```
2025-01-04 02:18:59.131 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Solis for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 177, in async_setup_entry
    hub = SolaXModbusHub(
        hass,
        plugin,
        entry,
    )
  File "/config/custom_components/solax_modbus/__init__.py", line 262, in __init__
    self._client = AsyncModbusTcpClient(host=host, port=port, timeout=5, framer=FramerType.Rtu, retries=6)
                                                                                ^^^^^^^^^^^^^^
AttributeError: type object 'FramerType' has no attribute 'Rtu'. Did you mean: 'RTU'?
```

Use HA File Editor  > open __init__.py in custom_components > solax modbus
self._client = AsyncModbusTcpClient(host=host, port=port, timeout=5, framer=FramerType.Rtu, retries=6)

Modify the FramerType.Rtu to FramerType.RTU.

self._client = AsyncModbusTcpClient(host=host, port=port, timeout=5, framer=FramerType.RTU, retries=6)

Restart, working perfectly again thank you dev

dougle03, 2025-01-04:
> self._client = AsyncModbusTcpClient(host=host, port=port, timeout=5, framer=FramerType.Rtu, retries=6)

I can't find this in my `__init__.py`
I don't have a `init.py` file...

darkrain-nl, 2025-01-04:
No need to change things in any file, just use the b2 release and be happy :)

wills106, 2025-01-05:
Closing down as related to #1172

closed 2025-01-05 by wills106