wills106/homeassistant-solax-modbus #1190: [Bug]: Integration is not loading

tonuj, 2025-01-04:
### Describe the bug

Integration is not loading after update to versions specified. pymodbus module is probably newer:
FramerType.Rtu has changed to FramerType.RTU

FIX: changing RTU to capital letters: /config/custom_components/solax_modbus/__init__.py:262

### Integration Version

2025.01.0b2

### Homeassistant core version

2025.1.0

### Inverter brand

SofarSolar 15KTL-3PH

### Plugin used

plugin_solax.py

### Serial prefix

123

### Inverter firmware versions

123

### Connection Method

Modbus RTU over TCP

### Dongle firmware

123

### Detailed Error Log

Logger: homeassistant.config_entries
Source: [config_entries.py:640](http://config_entries.py:640/)
First occurred: 4:59:38 AM (1 occurrences)
Last logged: 4:59:38 AM

Error setting up entry Sofar for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 177, in async_setup_entry
    hub = SolaXModbusHub(
        hass,
        plugin,
        entry,
    )
  File "/config/custom_components/solax_modbus/__init__.py", line 262, in __init__
    self._client = AsyncModbusTcpClient(host=host, port=port, timeout=5, framer=FramerType.Rtu, retries=6)
                                                                                ^^^^^^^^^^^^^^
AttributeError: type object 'FramerType' has no attribute 'Rtu'. Did you mean: 'RTU'?

darkrain-nl, 2025-01-04:
Maybe this helps? https://github.com/wills106/homeassistant-solax-modbus/issues/1189#issuecomment-2570015283

wills106, 2025-01-05:
Closing down as related to #1172

closed 2025-01-05 by wills106