wills106/homeassistant-solax-modbus #1192: [Bug]: Solax modbus not loaded after core update to Core 2025.1.0

Alfred99-HA, 2025-01-04:
### Describe the bug

After update to Cor 2025.1.0 solax was not loaded anymore with following error:
Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

A warning was issues as well:
Detected blocking call to import_module with args ('custom_components.solax_modbus',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. 

### Integration Version

2024.12.2

### Homeassistant core version

2025.1.0

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H3UE1

### Inverter firmware versions

can't read

### Connection Method

LAN

### Dongle firmware

no dongle used

### Detailed Error Log

[Setup failed.txt](https://github.com/user-attachments/files/18307354/Setup.failed.txt)
[Detected blocking call.txt](https://github.com/user-attachments/files/18307356/Detected.blocking.call.txt)

darkrain-nl, 2025-01-04:
Please try the b2 beta release, that works fine for me with HA 2025.1.

zefixx1, 2025-01-04:
Same error here
Thank you for your great work

theduca, 2025-01-04:
Can you please share where to locate the beta releases? I can only find the pre releases

wills106, 2025-01-05:
Closing down as related to #1172

closed 2025-01-05 by wills106

Alfred99-HA, 2025-01-05:
Perfect fix in no time during weekend!
Thanks a lot