wills106/homeassistant-solax-modbus #1194: Sofar / Pylontech Modbus setup - No individual pack voltages

HaraldRuppert, 2025-01-04:
### Describe the bug

I have a Sofar HYD20KTL-3P inverter with a Pylontech SC0500 / 10x H48050 battery stack using the LSE3 dongle

My Modbus connection is proxied via the HomeAssistant EVCC Addon. Everything is working great, except I'm not getting any per pack battery voltages.

[image]

### Integration Version

2025.01.0b2

### Homeassistant core version

2025.1.0

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP1ES1

### Connection Method

Sofar LSE-3

cschlipf, 2025-01-27:
These values are only supported for Sofar Solar BTS-5 power packs.

Pylontech has their own interfaces that you can use to monitor Pylontech Battery Packs. Maybe this helps: https://github.com/hidaba/PylontechMonitoring

@wills106 - you can close this as the submitter requests an unsupported configuration.

HaraldRuppert, 2025-01-27:
@cschlipf thank you for the clarification. Closing issue 

Have a great day everyone

closed 2025-01-27 by HaraldRuppert