wills106/homeassistant-solax-modbus #1195: Integrations/Entities Not Available

gemjack, 2025-01-05:
### Describe the bug

Since a reboot none of Solax entities have been available, if they are made available they go into not available,  even through I restored a known good version>  Not related to the lastest core version.  Has been working perfectly for over a year.  I can ping the inverter and the Solax App works fine (but is rubbish compared to HA

### Integration Version

2024.12.2

### Homeassistant core version

2024.12.2

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H4752a

### Inverter firmware versions

X1 Hybrid -G4

### Connection Method

TCP/IP Wifi

### Dongle firmware

Solax Pocket Wifi 3

### Detailed Error Log

[image]

### Additional context

[image]

[image]

masongary, 2025-01-05:
Same issue for me

wills106, 2025-01-05:
Have look at https://github.com/wills106/homeassistant-solax-modbus/issues/1168#issuecomment-2559653769

gemjack, 2025-01-05:
Reseated WiFi dongle, restarted HA, still the same. Solax App fine, Smart Plugs, Octopus Rates all fine.  This all started after a reboot.

gemjack, 2025-01-05:
OK, think I have figured most of it out.  For some reason the entities have been renamed to having 2 at the end of them.  I also have another Modbus device ?   Very strange, can you explain why this would happen.  I have one outstanding issue, not sure if it the Modbus related... I have mushroom card for mode (self use), Min Soc, Charge from Grid, and charge to....  The Charge from grid is not working.   I seem to have duplicate entities, one is unavailable...  

[image]
[image]

gemjack, 2025-01-05:
OK, fixed, sorry I list the wrong entity here, should have been select.solax_selfuse_night_charge_enable, so I changed that one to select.solax_selfuse_night_charge_enable_2  All good now.... but I now have two Modbus integrations, I guess something has changed along the way and X1-Hydrid-7.5kw replaced the original..... the latest version and update the HA core been has everything is now working.....
[image]  Should I mirgrate to

epicbenz, 2025-01-06:
> OK, fixed, sorry I list the wrong entity here, should have been select.solax_selfuse_night_charge_enable, so I changed that one to select.solax_selfuse_night_charge_enable_2 All good now.... but I now have two Modbus integrations, I guess something has changed along the way and X1-Hydrid-7.5kw replaced the original..... the latest version and update the HA core has been ignored, everything is now working..... 

I have the same issue after updating. Don't know why and if I can delete the first "device" of SolaX without problems. Are we the only ones that have that issue?

Did you deleted the first device with not available entities?

gemjack, 2025-01-06:
No, I just left it there... Just incase it messes something up....

Paul Bonehill

prehor, 2025-01-08:
> OK, think I have figured most of it out. For some reason the entities have been renamed to having 2 at the end of them. I also have another Modbus device ? Very strange, can you explain why this would happen. I have one outstanding issue, not sure if it the Modbus related...

It seems that the Solax integration generated new `unique_ids` for the entities, and since there were already entities with the same name, HA added _2 to their names.

If you're not afraid, you can check the file `config/.storage/core.entity_registry`, find both entities, and compare their `unique_ids`.

If you would like to rename entities, please be aware of the bug described in https://github.com/home-assistant/core/issues/115747#issuecomment-2560857963