wills106/homeassistant-solax-modbus #1198: [Bug]: Integration shows update pending, but is tagged to HA core 2025.1.1 which does not exist

jefft4, 2025-01-05:
### Describe the bug

Integration failed to load after update of HA core to 2025.1.0. On restart, HA found an update to the integration to 2025.01.2, but this update can't be downloaded in HACS as it has a requirement for HA core 2025.1.1, which does not appear to exist - docker hub images are 2025.1.0 latest & rc and 2025.2.0 in dev.
Is the dependency on core 2025.1.1 correct, or should that be 2025.1.0?

### Integration Version

2025.01.2

### Homeassistant core version

2025.1.0

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

n/a

### Inverter firmware versions

n/a

### Connection Method

n/a

### Dongle firmware

n/a

wills106, 2025-01-05:
I forgot HA starts at 0 and not 1.

Just released [2025.01.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.3) correcting the hacs.yml

Logirius, 2025-01-05:
Hi, thanks for your prompt help, but it seems that the promlem still exist. Same error
[image]

darkrain-nl, 2025-01-05:
I managed to upgrade, but now the integration fails to initialize, because of the following:
```
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 11:12:42 (1 occurrences)
Last logged: 11:12:42

Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
```
Which is strange cause the b2 release worked fine.

wills106, 2025-01-05:
@Logirius have you tried Ctrl + F5 to see if it's the browser cache that's got the 2025.1.1 stuck in it's memory?

wills106, 2025-01-05:
@darkrain-nl I think I merged the 2025.1.1 branch into main the wrong way round.

Logirius, 2025-01-05:
@wills106 I have deleted manually the browser cache. For your information, I use the Homeassistant App on Ipad IOS. Tried with safari and chrome on IOS too… also tried with InPrivate.

darkrain-nl, 2025-01-05:
I used the "update information" button in HACS to update to the 2025.1.3 release...

@wills106 so I just wait for a new release? (b2 again runs great :) )

wills106, 2025-01-05:
It's also in the manifest, not just hacs.yml. I'll release a new version soon.

wills106, 2025-01-05:
Can you try [2025.01.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.4) I don't have access to my Installation or normal computer to check at the moment.

Logirius, 2025-01-05:
Yes I will try

darkrain-nl, 2025-01-05:
Sure already on it, same issue:

```
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 11:39:27 (1 occurrences)
Last logged: 11:39:27

Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
```

wills106, 2025-01-05:
🤦‍♂️ missed a line out [2025.01.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.5) should work.

darkrain-nl, 2025-01-05:
Yep working fine, thanks again!

Logirius, 2025-01-05:
Many thanks @darkrain-nl for the tip with HACS - here it was possible to install the update. Otherwise via normal settings -> Update not. Not with IOS Safari, Chrome and Windows 11 Chrome (Normal and InPrivate). 

I have installaed 2025.01.5 now and the Sofarsolar HYD 20 is sending now his values again. Thanks a lot @wills106 for your great integration and help!

wills106, 2025-01-05:
Closed as corrected in [2025.01.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.5)

closed 2025-01-05 by wills106