wills106/homeassistant-solax-modbus #1199: [Bug]: Setup failed for custom integration 'solax_modbus' after updating to HA Core 2025.1.0

SpiritofGS, 2025-01-05:
### Describe the bug

Just updated HA Core to 2025.1.0 and after that to Solax Modbus 2025.01.4
Since then, Solax Modbus no longer starts with the above error message. Otherwise, no changes were made. Only the updates offered by HA were started. Has worked perfectly so far.

### Integration Version

2025.01.4

### Homeassistant core version

2025.1.0

### Inverter brand

SolaX X3-Hybrid-G4

### Plugin used

plugin_solax.py

### Serial prefix

H34B08

### Inverter firmware versions

ARM 1.35-1.09

### Connection Method

Waveshare

### Dongle firmware

3.003.01

### Detailed Error Log

Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 11:49:21 (1 Vorkommnisse)
Zuletzt protokolliert: 11:49:21

Setup failed for custom integration 'solax_modbus': Unable to import component: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 64, in <module>
    from pymodbus.transaction import ModbusAsciiFramer, ModbusRtuFramer
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py)

### Additional context

I´ll get a "warning" log entry also:

Logger: homeassistant.util.loop
Quelle: util/loop.py:77
Erstmals aufgetreten: 11:49:21 (1 Vorkommnisse)
Zuletzt protokolliert: 11:49:21

Detected blocking call to import_module with args ('custom_components.solax_modbus',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

wills106, 2025-01-05:
This was corrected in https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.5

BEMSman, 2025-01-05:
> This was corrected in https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.5

Since upgrading to HA Core to 2025.1.0 and then to Solax Modbus 2025.01.5, the units no longer appear:
[image]
[image]

Top Snip is an instance before upgrade, 2nd pic is after upgrades.

Is there something we would need to do to bring back the units?

PSMGoossens, 2025-01-05:
@wills106 Running 2025.01.5 is giving me also an error:

`2025-01-05 13:35:30.002 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry schuur for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 177, in async_setup_entry
    hub = SolaXModbusHub(
        hass,
        plugin,
        entry,
    )
  File "/config/custom_components/solax_modbus/__init__.py", line 262, in __init__
    self._client = AsyncModbusTcpClient(host=host, port=port, timeout=5, framer=FramerType.Rtu, retries=6)
                                                                                ^^^^^^^^^^^^^^
AttributeError: type object 'FramerType' has no attribute 'Rtu'. Did you mean: 'RTU'?`

SpiritofGS, 2025-01-05:
After I completely restarted the computer on which HA is running, an update for Solax Modbus was displayed again. Same version number. I installed it and everything works again. So it makes sense to restart the machine first (not just HA), and then see if the error is still there. This solved the problem for me. 
Thanks for your support!

wills106, 2025-01-05:
Thanks @PSMGoossens for the first PR of this year.

Released [2025.01.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.6) with the correction.

@BEMSman I can't update my normal HA at the moment and don't have access to my dev instance to update that either. Do you get any errors in the logs?

Anyone else noticed the units are missing?

PSMGoossens, 2025-01-05:
@wills106 No problem. Happy to help out with this amazing plugin :)

I can confirm that the units are working for at least the Growatt devices (MIC and MOD).

BEMSman, 2025-01-05:
@wills106 

I have just updated to 2025.01.6, just in case that fixed the issue, sadly not.
When updating anything Modbus related, I always do a reboot and not just a HA restart.

Complete debug log for your perusal 
[home-assistant_solax_modbus_2025-01-05T13-29-40.493Z.log](https://github.com/user-attachments/files/18310395/home-assistant_solax_modbus_2025-01-05T13-29-40.493Z.log)

Thanks for the great work you do.

MrRickwi, 2025-01-05:
Hi @wills106, can confirm units are working for my Solax G4 after updating to 2025.01.6. Thanks for your work :-)

BEMSman, 2025-01-05:
> Hi @wills106, can confirm units are working for my Solax G4 after updating to 2025.01.6. Thanks for your work :-)

Hi @MrRickwi 
I'm wondering on any difference in upgrade process?
I upgraded HA first, Then Solax to 2025.01.5 (loss of units) then upgraded to 2025.01.6 (still loss of units).
Would your upgrade have been any different?

MrRickwi, 2025-01-05:
Nearly no difference - HA 2025 first, then Solax to 2025.01.6. But 2025.01.5 and 2025.01.6 are nearly identical. What I also did: After upgrading HA a complete system restart including reboot of my PI4.

PSMGoossens, 2025-01-05:
@BEMSman Could it be that you've locked a file or something? Based on your logging
`(SyncWorker_0) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run? or didnt you modify any DATA_LOCAL entity?`

wills106, 2025-01-05:
The X1-AC and other Gen3 inverters don't use the DATA_LOCAL function, so that error can be ignored.

BEMSman, 2025-01-05:
@wills106 - FYI
@MrRickwi Thank you for the upgrade procedure you used.

Both, I upgraded HA last night and then the Solax to 2025.01.5 this morning, that was the start of the issue with no units.
After reading MrRickwi upgrade, I then rolled my HA back to yesterday, then upgraded HA to 2025.1.0, then upgraded Solax to 2025.01.6, all is now correct.

I also decided to upgrade an old Pi4 instance from HA 2024.9.0b3 / Solax 2024.03.6 straight up to HA 2025.1.0 / Solax 2025.01.6 and interestingly the units are missing, this one doesn't concern me as I seldom use that one. 

I'll leave it a couple of days and then upgrade my "production" instance.
Thanks for all that helped.

wills106, 2025-02-23:
Can this be closed now?

BEMSman, 2025-02-23:
This is working great for me.
currently on 2025.01.8

Thanks for all your great work

closed 2025-02-23 by wills106