wills106/homeassistant-solax-modbus #1201: [Bug]: 'Sofar' - '4.4 KTLX-G3' No response received after 6 retries, continue with next request

darkrain-nl, 2025-01-05:
### Describe the bug

To start, not sure if this is actually a bug, cause what happens is expected. I just wonder why it now shows up as an error as I don't recall it doing that before...

When the inverter can no longer be reached (because the sun is down) there is now an ERROR in the Home Assistant logs.
```
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:545
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:32:44 (14 occurrences)
Last logged: 16:46:59

No response received after 6 retries, continue with next request
```
I believe I have not seen this error before (searched for issues here as well), so I wonder if this could be handled more gently as this might give users the idea they have to do something... which is true, they just have to wait for the sun to rise again the next day.... But that might not be clear for everybody.

What are your thoughts here?

### Integration Version

2025.01.6

### Homeassistant core version

2025.1.0

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SS2ES1

### Connection Method

Waveshare adapter

### Detailed Error Log

```
2025-01-05 16:32:44.036 INFO (MainThread) [custom_components.solax_modbus] read_error : sofar cannot read holding registers at device 1 position 0x404
2025-01-05 16:47:34.322 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:47:34.323 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:48:09.333 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:48:09.335 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:48:44.343 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:48:44.345 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:49:19.355 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:49:19.358 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:49:54.367 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:49:54.369 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:50:29.375 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:50:29.377 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:51:04.384 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Input/Output] ERROR: No response received of the last 3 request, CLOSING CONNECTION. : sofar cannot read holding registers at device 1 position 0x488
2025-01-05 16:51:04.384 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:51:04.385 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:51:39.391 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:51:39.393 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:52:14.402 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:52:14.404 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:52:49.412 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:52:49.414 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:53:24.421 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:53:24.422 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:53:59.428 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:53:59.431 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:54:34.439 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:54:34.440 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:55:09.451 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:55:09.453 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:55:44.460 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:55:44.461 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:56:19.468 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:56:19.470 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:56:54.477 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:56:54.479 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:57:29.486 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:57:29.488 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:58:04.495 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:58:04.497 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:58:39.504 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:58:39.506 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:59:14.516 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:59:14.518 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 16:59:49.536 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 16:59:49.539 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 17:00:24.544 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 17:00:24.546 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 17:00:59.554 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-05 17:00:59.557 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.0.55.198:502
2025-01-05 17:01:34.566 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
```

honekr, 2025-01-24:
Same issue on SolaX X3-Hybrid ane Waveshare RS485 TO POE ETH:

```
Tato chyba pochází z vlastní integrace.

Logger: pymodbus.logging
Zdroj: custom_components/solax_modbus/__init__.py:553
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 10:03:32 (4 výskyty)
Naposledy logováno: 10:29:17

No response received after 6 retries, continue with next request
```

jone9618, 2025-02-12:
Same issue with a Solis inverter (serial prefix 160F522) and a USR IOT (type HB) RS485 to ethernet gateway, running HA v2025.2.2

cschlipf, 2025-04-29:
Does this problem still exist with the latest version?

jone9618, 2025-04-30:
> Does this problem still exist with the latest version?

It does for me

closed 2025-09-17 by github-actions[bot]