wills106/homeassistant-solax-modbus #1203: [Bug]: Unknown values for MIC X3

the-real-mcarn, 2025-01-06:
### Describe the bug

Plugin connects to the inverter correctly, though throwing a warning, and all values remain either 'Unknown' or just 0. Behaviour is since 2024.12.01 but I had not updated since the breaking changes HASS version 2025.1.0. Reverting to a previous version fixes the issue. 

### Integration Version

2025.01.6

### Homeassistant core version

2025.1.0

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

MC208T

### Inverter firmware versions

ARM 1.15 DSP1.18

### Connection Method

Waveshare Adapter

### Dongle firmware

V1.452

### Detailed Error Log

2025-01-06 21:03:00.497 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2025-01-06 21:03:00.497 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.2.236:502
2025-01-06 21:03:00.505 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.2.236:502
2025-01-06 21:03:00.505 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2025-01-06 21:03:00.533 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2025-01-06 21:03:00.533 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number: None
2025-01-06 21:03:00.576 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x300 serial number: MC208T********
2025-01-06 21:03:00.577 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None
2025-01-06 21:03:00.638 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX sensor groups: 1
2025-01-06 21:03:00.638 DEBUG (MainThread) [custom_components.solax_modbus.sensor] SolaX group: 15
2025-01-06 21:03:00.638 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x318 rtc to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x347 q-curve to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x351 active_power_limit to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x352 firmware_dsp to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x353 firmware_arm to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x367 lock_state to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x371 export_power_limit to block with start 0x318
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x400 pv_voltage_1 to block with start 0x400
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x401 pv_voltage_2 to block with start 0x400
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x402 pv_current_1 to block with start 0x400
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x403 pv_current_2 to block with start 0x400
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x404 inverter_voltage_l1 to block with start 0x400
2025-01-06 21:03:00.639 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x405 inverter_voltage_l2 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x406 inverter_voltage_l3 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x407 inverter_frequency_l1 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x408 inverter_frequency_l2 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x409 inverter_frequency_l3 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40a inverter_current_l1 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40b inverter_current_l2 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40c inverter_current_l3 to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40d inverter_temperature to block with start 0x400
2025-01-06 21:03:00.640 DEBUG (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40e inverter_power to block with start 0x400
[... 20 lines cut ...]

---Extremely long entries for holdingBlocks, inputBlocks and computedRegs ---

2025-01-06 21:03:15.582 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-01-06 21:03:15.582 DEBUG (MainThread) [custom_components.solax_modbus] SolaX modbus holding block start: 0x318 end: 0x372  len: 90 
regs: [792, 839, 849, 850, 851, 871, 881]
2025-01-06 21:03:15.610 INFO (MainThread) [custom_components.solax_modbus] read_error : SolaX cannot read holding registers at device 1 position 0x318
NoneType: None
2025-01-06 21:03:15.612 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2025-01-06 21:03:15.612 DEBUG (MainThread) [custom_components.solax_modbus] device group read done
2025-01-06 21:05:30.590 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-01-06 21:05:30.618 INFO (SyncWorker_3) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run? or didnt you modify any DATA_LOCAL entity?
2025-01-06 21:05:30.618 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10

### Additional context

The following solutions I tried without success:
- Restarted inverter & RS485 adapter
- Reconnected RS485 adapter
- Run without PocketLAN (v3) connected
- Restarted home assistant
- Reconfigured inverter in home assistant
- Removed and added the inverter back in home assistant

Reading the holding register 0x318 (0d792) returns 0x0000 in Open ModScan.

KestasG, 2025-01-19:
Have same issue with X3-MIC Gen2. Added some logging and enabled debug modbuspy, but not sure what the output means for now.

2025-01-19 17:29:46.935 DEBUG (MainThread) [custom_components.solax_modbus] device group inverter
2025-01-19 17:29:46.935 DEBUG (MainThread) [custom_components.solax_modbus] SolaX modbus holding block start: 0x351 end: 0x372  len: 33 
regs: [849, 850, 851, 871, 881]
2025-01-19 17:29:46.935 DEBUG (MainThread) [custom_components.solax_modbus] will invoke async_read_holding_registers
2025-01-19 17:29:46.935 WARNING (MainThread) [custom_components.solax_modbus] async_read_holding_registers2
2025-01-19 17:29:46.935 WARNING (MainThread) [custom_components.solax_modbus] async_read_holding_registers2 after chekc connection
2025-01-19 17:29:46.935 DEBUG (MainThread) [pymodbus.logging] Adding transaction 3
2025-01-19 17:29:46.936 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x3 0x51 0x0 0x21
2025-01-19 17:29:46.962 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0x3 0x1 0x83 0x2 old_data:  addr=None
2025-01-19 17:29:46.963 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0x3 0x1 0x83 0x2
2025-01-19 17:29:46.963 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-01-19 17:29:46.963 DEBUG (MainThread) [pymodbus.logging] Getting transaction 3
2025-01-19 17:29:46.963 WARNING (MainThread) [custom_components.solax_modbus] async_read_holding_registers2 after read
2025-01-19 17:29:46.963 DEBUG (MainThread) [custom_components.solax_modbus] after invoking async_read_holding_registers
2025-01-19 17:29:46.963 INFO (MainThread) [custom_components.solax_modbus] read_error : SolaX cannot read holding registers at device 1 position 0x351
NoneType: None
2025-01-19 17:29:46.964 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2025-01-19 17:29:46.965 DEBUG (MainThread) [custom_components.solax_modbus] device group read done

If I try same 0x351 with nodered modbus I get:

[image]

Which seems correct as old version was displaying this value as 11000W

KestasG, 2025-01-28:
Found the issue, at least for my inverter:
_Device info
X3-MIC
by SolaX Power
Firmware: DSP v1.117 ARM v1.113
Hardware: Gen2
Serial number: MC210TIXXXXXXX_

Registers 0x367 and 0x371 are not supported (anymore ???) by the inverter and return error. After updating _allowedTypes_ for the registers to GEN3, so it does not get picked plugin started working.

KestasG, 2025-01-28:
It was this commit which added unsupported registers:
https://github.com/wills106/homeassistant-solax-modbus/commit/cf216d70ac78d92a4f81a09fc06fb392d84d71b9#diff-07e2bac61d9098486da200bb7225ffe36eb1a8ad77dee911243a2c83a2c77145

the-real-mcarn, 2025-01-28:
Nice! I tested this on my instance as well and I'm finally getting values again! I'm not sure if I can implement this fix tho, I don't know the codebase well enough for that.

LnK-Holly, 2025-04-10:
Amazing! This worked for me, too :) Thank you @KestasG ! @the-real-mcarn , does it work for you? If so you could close as solution, and perhaps Will could implement changes in next update?

> Found the issue, at least for my inverter: _Device info X3-MIC by SolaX Power Firmware: DSP v1.117 ARM v1.113 Hardware: Gen2 Serial number: MC210TIXXXXXXX_
> 
> Registers 0x367 and 0x371 are not supported (anymore ???) by the inverter and return error. After updating _allowedTypes_ for the registers to GEN3, so it does not get picked for my inverter, the plugin started working.

rosenrot00, 2025-09-07:
@KestasG could you share the log file running the latest integration.

May I ask, is you HA core also on the latest version?

KestasG, 2025-09-07:
My HA is 2025.9.1 Now I've patched in according to  https://github.com/wills106/homeassistant-solax-modbus/issues/1203#issuecomment-2618860782 and it started working again, so  don't have  logs now.