wills106/homeassistant-solax-modbus #1206: [Bug]: Integration not working after HA core update to 2025.1.1

Solimatorko, 2025-01-07:
### Describe the bug

The integration stops working after the update to 2025.1.1 HA core. It just fails to parse anything from the inverter bus. 
I've tried every version since 2025.01.2 to the latest. The error messages from the Logs:

1. hci0 (2C:CF:67:07:6C:97): Successful fall-back to passive scanning mode after active scanning failed (4/4)
2. Error setting up entry SolaX for solax_modbus

 `Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 23:23:19 (2 occurrences)
Last logged: 23:25:41

Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 190, in async_setup_entry
    await hub.async_init()
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in async_init
    self._invertertype = await self.plugin.async_determineInverterType(self, self.config)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 7714, in async_determineInverterType
    seriesnumber = await async_read_serialnr(hub, 0x0)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 112, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
        req, timeout=self.ctx.comm_params.timeout_connect
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError`

### Integration Version

2025.01.6

### Homeassistant core version

2025.1.1

### Inverter brand

Solax X3-Hybrid G4 

### Plugin used

plugin_solax.py

### Serial prefix

H34A08

### Inverter firmware versions

ARM 1.38-1.09, DPS 1.39

### Connection Method

Solax Pocket wifi+LAN

### Dongle firmware

1.002.009

yvosound, 2025-01-09:
same here, no response..........................

Gazzyoung, 2025-01-10:
I have updated my HA and since then nothing from my Solax system. I have removed the integration and reinstalled (which normally fixes) and now its not detecting any entities...

Gazzyoung, 2025-01-10:
> I just fixed it by unpluging the Pocket Wifi dongle from the Solax
> Inverter and pluging it back. Removed the integration from HA and add it
> back and it starts working.
>
> It is weird because I was able to communicate with the dongle through
> local API by calling
> curl -d "optType=ReadRealTimeData&pwd=xxxxx" -X POST http://192.168.1.164
> it responded OK with the real-time data but it seems that HA was not able
> to comunicates with the dongle.
>
> Try to unplug the Wifi dongle from the Solax inverter

I am at work but could try a reboot of the Solax system.

wills106, 2025-01-11:
The PocketWiFi is known to close it’s [Modbus connection down](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)

Might be worth seeing if there is a Firmware update available or ideally use something like a [Waveshare](https://homeassistant-solax-modbus.readthedocs.io/en/latest/compatible-adaptors/#waveshare-rs485-to-rj45-ethernet-din-rail-mount-rs485-to-eth-b-rs485-to-poe-eth-b) instead.

closed 2025-01-11 by wills106