wills106/homeassistant-solax-modbus #1208: [Bug]: 'Growatt' - 'MOD 6000-TL2-XH' Integration not possible, gives error

kelddamsbo, 2025-01-10:
### Describe the bug

No problem adding to HACS.

When adding Integration :
Error
Configuration flow could not be read: {"message":"Invalid handler specified"}

### Integration Version

2025.01.6

### Homeassistant core version

2025.1.2

### Inverter brand

Growatt MOD6000-TL3-XH

### Plugin used

plugin_growatt.py

### Serial prefix

JCM0DA

### Connection Method

RS485 USB Waveshare

### Detailed Error Log

In logs:

Logger: homeassistant.util.loop
Kilde: util/loop.py:136
Første forekomst: 11.04.35 (1 forekomster)
Senest logget: 11.04.35

Detected blocking call to import_module with args ('custom_components.solax_modbus.config_flow',) inside the event loop by integration 'config' at homeassistant/components/config/config_entries.py, line 179: return await super()._post_impl(request, data) (offender: /usr/src/homeassistant/homeassistant/loader.py, line 1304: return importlib.import_module(f"{self.pkg_path}.{platform_name}")), please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+config%22 For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request resp = await request_handler(request) File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle return await handler(request) File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware response = await handler(request) File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle result = await handler(request, **request.match_info) File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin return await func(self, request, *args, **kwargs) File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper return await method(view, request, data, *args, **kwargs) File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 172, in post return await self._post_impl(request, data) File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 179, in _post_impl return await super()._post_impl(request, data)

And:

Logger: homeassistant.config_entries
Kilde: config_entries.py:3314
Første forekomst: 11.04.35 (2 forekomster)
Senest logget: 11.16.44

Error occurred loading flow for integration solax_modbus: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)

kelddamsbo, 2025-01-11:
It works in Integration Version 2025.01.2

wills106, 2025-01-11:
What version of pyModbus are you running.
Sounds like a 3.6.x version.

https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#how-to-check-a-version-of-a-python-module-eg-pymodbus

kelddamsbo, 2025-01-11:
Hmm, I'm running a Supervised installation.
If I use terminal in HA:
➜  pip show pymodbus 
WARNING: Package(s) not found: pymodbus

If I use terminal in Raspberry pi OS:
pip show pymodbus
WARNING: Package(s) not found: pymodbus

Should I install it ?

kelddamsbo, 2025-01-11:
Tried to install 
RPI OS: sudo apt install python3-pymodbus ver 3.0.0RC1
HA pip install pymodbus (not persistant after reboot) ver 3.8.3

But no change still same problem

kelddamsbo, 2025-01-12:
Sorry, saw now how to check version

docker exec -it homeassistant bash
pip show pymodbus

Shows:
homeassistant:/config# pip show pymodbus
Name: pymodbus
Version: 3.7.4
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: Galen Collins, Jan Iversen
Author-email: 
License: BSD-3-Clause
Location: /usr/local/lib/python3.13/site-packages
Requires: 
Required-by: pyiskra, pysmarty2, pystiebeleltron
homeassistant:/config# 

So I'm running 3.7.4

kelddamsbo, 2025-01-14:
After update of HA tonight:
homeassistant:/config# pip show pymodbus
Name: pymodbus
Version: 3.5.4
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: "Galen Collins, Jan Iversen"
Author-email: 
License: BSD-3-Clause
Location: /usr/local/lib/python3.13/site-packages
Requires: 
Required-by: pyiskra, pysmarty2, pystiebeleltron
homeassistant:/config#

PSMGoossens, 2025-01-16:
Hmmm are you sure that you are running the latest home assistant? https://github.com/home-assistant/core/blob/dev/homeassistant/components/modbus/manifest.json it seems that the standard is using version 3.8.3

wills106, 2025-01-16:
[2025.01.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.8) Should pin pyModbus to 3.8.3

kelddamsbo, 2025-01-16:
Yes all is updated, I run home assistant supervised.
But it does not use 3.8.3

kelddamsbo, 2025-01-25:
Are solax modbus pinning to a certent version of pymodbus ?

PSMGoossens, 2025-01-25:
> Are solax modbus pinning to a certent version of pymodbus ?

Yes, in the mainfest.json towards version 3.8.3

kelddamsbo, 2025-01-25:
Ahhh thanks for the hint, I found another integration installed https://github.com/Breina/PowerTagGateway 
This has pymodbus 3.5.4 as requirement.
I guess HA takes the lowest version and install.
I think my problem is there

kelddamsbo, 2025-01-28:
PowerTagGateway, updated also, so now everything works perfect with pymodbus 3.8.3

closed 2025-01-28 by kelddamsbo