wills106/homeassistant-solax-modbus #1215: [Bug]: 'Inverter Brand' - 'Inverter Model' Description of Error

TheEragon, 2025-01-14:
### Describe the bug

After upgrading to 2025.01.7 integration stopped working with Growatt SPH 10000, I get following error

`
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 64, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=5)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
`

### Integration Version

2025.01.7

### Homeassistant core version

2025.1.2

### Inverter brand

Growatt SPH 10000

### Plugin used

plugin_growatt.py

### Serial prefix

0

### Connection Method

RS485

wills106, 2025-01-16:
Try [2025.01.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.8) there was an issue with pyModbbus 3.8.3 in Integration 2025.01.7

TheEragon, 2025-01-16:
Works now thanks!

closed 2025-01-16 by TheEragon