wills106/homeassistant-solax-modbus #1217: [Bug]: Failed to read serial number

MrRickwi, 2025-01-14:
### Describe the bug

When starting up, the latest version reports the following error messages and does not read anything from the inverter:

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:70
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 17:09:04 (402 Vorkommnisse)
Zuletzt protokolliert: 17:31:16

SolaX: attempt to read serialnumber failed at 0x0 data: None
SolaX: attempt to read serialnumber failed at 0x300 data: None
SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:86
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 17:09:04 (402 Vorkommnisse)
Zuletzt protokolliert: 17:31:16

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed
SolaX: reading serial number from address 0x1a10 failed; other address may succeed

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:7720
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 17:09:04 (134 Vorkommnisse)
Zuletzt protokolliert: 17:31:16

SolaX: cannot find any serial number(s)

Logger: custom_components.solax_modbus.plugin_solax
Quelle: custom_components/solax_modbus/plugin_solax.py:8003
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 17:09:04 (134 Vorkommnisse)
Zuletzt protokolliert: 17:31:16

unrecognized inverter type - serial number : unknown

### Integration Version

2025.01.7

### Homeassistant core version

2025.1.2

### Inverter brand

SolaX (QCells)

### Plugin used

plugin_solax.py

### Serial prefix

?

### Inverter firmware versions

?

### Connection Method

MODBUS

### Dongle firmware

?

### Detailed Error Log

No data is read from inverter because integration crashes on startup.

chronvas, 2025-01-14:
I have the same, slightly different error messages

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:61
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 4:52:13 PM (7 occurrences)
Last logged: 4:53:13 PM

solisinverter: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 61, in async_read_serialnr
    inverter_data = await hub.async_read_input_registers(unit=hub._modbus_addr, address=address, count=8)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 553, in async_read_input_registers
    resp = await self._client.read_input_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
```

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:73
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 4:52:13 PM (7 occurrences)
Last logged: 4:53:13 PM

solisinverter: reading serial number from address 0x80ec failed; other address may succeed
```

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2523
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 4:52:13 PM (7 occurrences)
Last logged: 4:53:13 PM

solisinverter: cannot find serial number, even not for other Inverter
```

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2571
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 4:52:13 PM (7 occurrences)
Last logged: 4:53:13 PM

unrecognized solisinverter inverter type - serial number : unknown
```

Lake292, 2025-01-14:
Same issue on Solax 10kW

`Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

2025-01-14 18:47:19.078 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed

2025-01-14 18:47:19.079 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
2025-01-14 18:47:19.082 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2025-01-14 18:47:19.082 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):

2025-01-14 18:47:19.087 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed

2025-01-14 18:47:19.087 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)

2025-01-14 18:47:19.088 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown`

Reverting to .6 version fixes the problem.

luftdieb, 2025-01-14:
Hi, same on my side with Solis RHI-4,6K-48ES-5G and solax-modbus version 2025.1.7. 
Showstopper bug because no data can be fetched from inverter anymore :-(. 
Going back to previous version 2025.1.6 solved problem for now.

Logger: custom_components.solax_modbus.plugin_solis
Quelle: custom_components/solax_modbus/plugin_solis.py:2571
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 19:00:23 (133 Vorkommnisse)
Zuletzt protokolliert: 19:22:26

unrecognized solis_modbus_ inverter type - serial number : unknown

BrAiNeeBug, 2025-01-15:
Same... After Update reading from Inverter fail.switched one version back.

ertho22, 2025-01-15:
Same problem here (two SolaX Inverters, X1 Mini an X3 Hybrid G4).

KevinD987, 2025-01-15:
Same problem here.

timstephenson20, 2025-01-15:
Same issue here on 2025.01.7. Solis inverter

solis: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 61, in async_read_serialnr
    inverter_data = await hub.async_read_input_registers(unit=hub._modbus_addr, address=address, count=8)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 553, in async_read_input_registers
    resp = await self._client.read_input_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

martintothgit, 2025-01-15:
Same Issue here. How can i go back to 2025.01.6 version without loosing historical data? Should i just restore backup from few days ago?

timstephenson20, 2025-01-15:
> Same Issue here. Solax X3 Hybrid G4 10.0 D. How can i go back to 2025.01.6 version without loosing historical data? Should i just restore backup from few days ago?

I used HACS to redownload the previous release; use 'redownload' & select 2025.01.6 from the list of releases. Let it install & restart HA. 
[image]

michalkak, 2025-01-15:
The same issue with my **Solax X3-Hybrid-G4** after todays update
2025.01.6 - works 
2025.01.7 - does not work
I had to rollback to 2025.01.6 - it fixed issue

wills106, 2025-01-16:
Try [2025.01.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.8) there was an issue with pyModbbus 3.8.3 in Integration 2025.01.7

slate-1, 2025-01-16:
Resolved for me with Solis.py with 2025.01.8. Thank you.

ertho22, 2025-01-16:
Works perfect for me, thank you.

martintothgit, 2025-01-16:
2015.01.8  solved the issue also for me. Thanks

wills106, 2025-01-16:
Closing down as 2025.01.8 has resolved the issue.

closed 2025-01-16 by wills106