wills106/homeassistant-solax-modbus #1222: [Bug]: 'Growatt' - 'Growatt - MOD 6000TL3-XH' Modbus Connection over Modbus Proxy failed

cracyfloyd, 2025-01-15:
### Describe the bug

After the last update yesterday (14.01.2025) i can not connect over my modbus proxy anymore. Before all worked for month fine. the connetion direct (without proxy) to the inverter (over waveshare) worked. 

### Integration Version

2025.01.7

### Homeassistant core version

2025.1.2

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

xxxxxx

### Connection Method

Waveshare Adapter

Ocramius, 2025-01-15:
Similar for me as well: moving to `2025.01.7` seems to have made my ZCS inverter unavailable.

According to this integration, I'm running a `HYDxxxxKTL ZCS HP` `V003` @ Firmware `0V6000020V6000020V600002`, serial prefix `ZM2ES060P`

Ocramius, 2025-01-15:
Well, this one seems quite clear to me :D 

```
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
```

boredomwontgetus, 2025-01-15:
Same for me on the latest update. The connection to a Solax X3-Hybrid Gen3 over Modbus Proxy does not work anymore.

dOn-MoP, 2025-01-15:
similar problem for me:

**Integration Version**
2025.01.7
**Homeassistant core version**
2025.1.2
**Inverter brand**
Growatt
**Model**
SPH
**Connection Method**
Waveshare Adapter

Rollback to Integration Version 2025.01.6 fixed issue.

wills106, 2025-01-16:
Try [2025.01.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.8) there was an issue with pyModbbus 3.8.3 in Integration 2025.01.7

Ocramius, 2025-01-16:
Indeed, reporting started again, with this release. Thanks @wills106!

wills106, 2025-01-16:
Closing down as 2025.01.8 has resolved the issue.

closed 2025-01-16 by wills106