wills106/homeassistant-solax-modbus #1223: [Bug]: BinaryPayloadDecoder is deprecated and will be removed in v3.9.0

wills106, 2025-01-16:
### Describe the bug

Warning in log for BinaryPayloadDecoder is deprecated and will be removed in v3.9.0

### Integration Version

2025.01.8

### Homeassistant core version

2025.1.2

### Inverter brand

All

### Plugin used

plugin_solax.py

### Serial prefix

N/A

### Inverter firmware versions

N/A

### Connection Method

N/A

### Dongle firmware

N/A

### Detailed Error Log

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

### Additional context

I have raised this to try and reduce multiple Issues raised for the same warning.

Please don't spam this Issue saying you have the same error, as everyone will have this warning.

PSMGoossens, 2025-01-16:
It seems that that the deprecation is no longer the  case, but it's strange that it's popping up
https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.8.3
Anyway it's good that it is removed, because otherwise it means a bigger rewrite, since the new "solution" is less flexible and doesn't support some key features such as ordering

wills106, 2025-01-16:
I'll keep track of pyModbus in case we do need to make the change, although there seems to be little examples of the changes required other than the very basic ones.

PSMGoossens, 2025-01-16:
Yeah I agree,
But it seems that we can only convert one datatype at the  same and "skip" functionality is not there, and no support for ordering of int32.
See also discussion at pyModbus: https://github.com/pymodbus-dev/pymodbus/pull/2456

smartminjos, 2025-01-17:
same issue here ..... if you need any logs, let me know ...

MrRickwi, 2025-01-18:
Same here, log is flooded:

Logger: pymodbus.logging
Quelle: custom_components/solax_modbus/__init__.py:771
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 17. Januar 2025 um 06:27:19 (**48216** Vorkommnisse)
Zuletzt protokolliert: 11:09:20

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

flopp999, 2025-01-22:
I don't know but maybe this can help you.
I have an Growatt MID 15KTL3-XH.
I tried and this code is working, at least for Growatt.
I didn't try those lines in Solax integration but I ran those lines by themself in a .py file

```
from pymodbus.client.mixin import ModbusClientMixin
registers = [19071, 20016, 17462, 17457, 12874]
serial=ModbusClientMixin.convert_from_registers(registers, data_type=ModbusClientMixin.DATATYPE.STRING)
print(serial)
```
will give output
`N0D6D12J`

honekr, 2025-01-24:
Same issue:

```
Tato chyba pochází z vlastní integrace.

Logger: pymodbus.logging
Zdroj: custom_components/solax_modbus/__init__.py:771
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 08:45:11 (1764 výskyty)
Naposledy logováno: 09:48:02

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
```

PSMGoossens, 2025-01-25:
I've created a PR where payload.py is brought to the application and the deprecration message is removed #1232

steps56, 2025-01-27:
> I've created a PR where payload.py is brought to the application and the deprecration message is removed [#1232](https://github.com/wills106/homeassistant-solax-modbus/pull/1232)

Thanks.
Is there any download/update of the integration available already ?
Would like to get rid of the error messages flood in HA.

steps56, 2025-01-28:
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: pymodbus.logging
Quelle: custom_components/old_solax_modbus/__init__.py:771
Integration: old_solax_modbus
Erstmals aufgetreten: 11:28:42 (4 Vorkommnisse)
Zuletzt protokolliert: 11:28:43

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

Sadly its back again after HA Update today...😪

wills106, 2025-01-30:
Just released [2025.01.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.9) with the change.

Looks like reading groups of registers will return in the new version https://github.com/pymodbus-dev/pymodbus/discussions/2525#discussioncomment-11980900

smartminjos, 2025-01-30:
work like a charm ! thanks

wills106, 2025-01-30:
> ~Looks like reading groups of registers will return in the new version~ [pymodbus-dev/pymodbus#2525 (comment)](https://github.com/pymodbus-dev/pymodbus/discussions/2525#discussioncomment-11980900)
> 
> Edit: It's been abandoned.

Probably need to look at how many separate reads would be needed with the new method. Unless we can string together the convert calls?

PSMGoossens, 2025-02-04:
> Probably need to look at how many separate reads would be needed with the new method. Unless we can string together the convert calls?
> 
> Or just stick with payload decoder being in this Integration.

Calling the separate registers will be a massive overhead, so I think for now we should keep this solution

steps56, 2025-02-19:
> [@wills106](https://github.com/wills106) Unfortunately looks like problem appeared again. Or its some issue on my side?

... and an other one, those spikes at many values from time to time.
Mentionend at a separate thread, but never answered if been under investigation by wills106 😪.

wills106, 2025-02-20:
I have noticed you get the warning on restarting HA. But you only get it the once and not 10,000 times.

steps56, 2025-02-20:
> I have noticed you get the warning on restarting HA. But you only get it the once and not 10,000 times.

Yes, this is right.

hubertozog, 2025-02-20:
In my case i got 12k times till last restart. So maybe somthing on my side, but what?

2025-02-20 12:07:14.600 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:14.981 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:15.047 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:15.451 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:15.834 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:15.901 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:29.602 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:29.985 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:30.063 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:30.471 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:30.853 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:30.919 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:44.614 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:44.996 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:45.062 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-20 12:07:45.464 WARNING (MainThread) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !

wills106, 2025-02-23:
> In my case i got 12k times till last restart. So maybe somthing on my side, but what?

What version of the Integration are you using?

hubertozog, 2025-02-23:
Right now 2025.02.1.

wills106, 2025-02-23:
Reopening it for now, although it's not a major issue.

It only write to the log once per HA restart.

hubertozog, 2025-02-24:
@wills106 In my case its spaming errors all the time, right now 9k times

[image]

PSMGoossens, 2025-03-09:
Removed the left overs in all plugins via PR #1274

steps56, 2025-03-09:
Is there a new Beta or ? out now containing your changes?
Can't find any now ...
Thanks for explaining please now.

wills106, 2025-03-12:
Just released [2025.03.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.03.1) with the changes.

wills106, 2025-03-13:
Closing down as fixed. Thanks @PSMGoossens

closed 2025-03-13 by wills106