wills106/homeassistant-solax-modbus #1226: SofarModbusSensorEntityDescription scale wrong (plugin_sofar.py)

erichsch, 2025-01-18:
3719
SofarModbusSensorEntityDescription(
        name="Realtime Capacity",
        key="realtime_capacity",
        native_unit_of_measurement=PERCENTAGE,
        device_class=SensorDeviceClass.BATTERY,
        register=0x900E,
        **scale=0.1,** (Must bee 1)
        allowedtypes=BAT_BTS,

cschlipf, 2025-03-13:
This scale has been removed already. But now the capacity is diplayed wrong: 

[image]

Why do you think it must be 1? Because of the modbus protocol description, where it seems to be wrong?

erichsch, 2025-03-13:
> Weird, for other users it definitely seems to be 1. Looks like there are
> two versions.
>
> @erichsch <https://github.com/erichsch> , what are the first digits of
> your serial and which HW version? In my case:
>
> Hardware: V001
> Seriennummer: SP2ES110...

Seriennummer: SP2ES110...
wo finde ich die Hardware: ? PCU = V2.00

cschlipf, 2025-03-13:
Ich habe das von der Geräteseite der Solax Integration:

[image]

Anyway: Ich habe Realtime Capacity erstmal rausgenommen und sorge dafür, dass stattdessen SoC angezeigt wird.

cschlipf, 2025-03-13:
Welche Firmware Version hast Du denn bei Dir installiert?

Wechselrichter: 110262
BTS habe ich 10026

wills106, 2025-03-13:
Shall I leave your PR #1278 open for now?
Or is this unrelated?

cschlipf, 2025-03-13:
@wills106 No, you can close that. The general fix is that Realtime Capacity should not be used for displaying the SoC.

wills106, 2025-03-13:
As the PR closed this down, do you want me to reopen? Or to move the issue else where?

closed 2025-03-13 by wills106

cschlipf, 2025-03-13:
Please keep it closed for now. I think the primary issue here was the display of the SoC, not Realtime Capacity. Am I right @erichsch ?

erichsch, 2025-03-14:
> Welche Firmware Version hast Du denn bei Dir installiert?
> 
> Wechselrichter: 110262 BTS habe ich 10026

V113562
10026

erichsch, 2025-03-14:
> Please keep it closed for now. I think the primary issue here was the display of the SoC, not Realtime Capacity. Am I right [@erichsch](https://github.com/erichsch) ?

you are right. ;-)

cschlipf, 2025-03-14:
> > Welche Firmware Version hast Du denn bei Dir installiert?
> > Wechselrichter: 110262 BTS habe ich 10026
> 
> V113562 10026

Yesterday I upgraded the inverter to 115162 and BMS to 011526.

And now my Realtime Capacity accuracy is back to 1 as well. Seems like this was a bug in my older Firmware. So I will re-add dee Realtime Capacity, however without the Battery Device Class so that we stay on display the SoC entity.

cschlipf, 2025-03-14:
See PR #1278

PR #1278 (2025-03-13): Sofar: Fix display of battery SoC for Battery Modules
For sofar battery modules three entities had the device class `SensorDeviceClass.Battery` set: Realtime Capacity, SoC and SoH. So one of these entities was chosen as SoC display.

SoH is totally wrong for that. I have removed the device class `SensorDeviceClass.Battery` here.

Realtime Capacity has the problem that different inverters have a different scale. But it does not provide additional value besides SoC. Therefore I have removed that entity entirely.

Now only SoC is left with the proper device class, which should fix the display val [... cut]