wills106/homeassistant-solax-modbus #1229: [Bug]: "ZCS Azzurro inverter reporting spikes in consumption/battery/production values that revolve around the magic `65535` integer

Ocramius, 2025-01-20:
### Describe the bug

Since 2025-01-14, which corresponds to the #1222 fix, I have numerous spikes in reported power consumption values.

Screenshots below mostly show the symptom in the HA dashboard.

[image]

[image]

[image]

[image]

[image]

I've since adjusted these spikes via manual statistics adjustments in HA, but I noticed that these spikes seem to suspiciously revolve around the magic `65536` (16 bit max integer) value.

I've observed such spikes also before 2025.01.8, but I'm wondering what may have changed, that might have caused an increase in these outliers.

I'm wondering how I may help further, with debugging such issues 🤔 

### Integration Version

2025.01.8

### Homeassistant core version

2025.1.2

### Inverter brand

ZCS Azzurro (sofar)

### Plugin used

plugin_growatt.py

### Serial prefix

ZM2ES060

### Connection Method

TCP/Ethernet via LSE-3 LAN stick Logger

manuelmrmorgado, 2025-01-20:
Same issue with Solax. Since 2025.1.8. Downgraded and its fine again.

Ocramius, 2025-01-22:
@manuelmrmorgado yep, had to downgrade to `2025.01.6` myself too: spikes are effectively the norm, right now, so something broke completely here.

Looking at the diffs: https://github.com/wills106/homeassistant-solax-modbus/compare/2025.01.6...2025.01.8

I think the biggest offender may be `pymodbus` itself.

The diff of PyModbus is massive though: https://github.com/pymodbus-dev/pymodbus/compare/v3.7.4...v3.8.3

Zwer2k, 2025-01-24:
Same Issue with Sofar

[image]

matejkaf, 2025-01-25:
SolaX X3 G4 here. I have spikes all over the place. Spikes are short (around 10 Seconds), one polling interval maybe.
House Load peaked to 3,932,190.0 W today 2 times. Energy total values drop to zero several times a day. Energy statistic is garbage. Will downgrade to 2025.01.6.

mbeetle, 2025-01-25:
Same here for me with Solax X3 G4. Downgrade back to 2025.1.6 solves the Issue

jelmeronline, 2025-01-26:
No problems with the magic number but sometimes the total generated power, lets say 2000kWh in the inverters lifetime, is read as 0 and then back to normal. So Homeassistant has these values
2000
2001
2002
0
2003
2004
And will calculate a total generated power of 4006kWh for this period.

Ocramius, 2025-01-26:
Yeah, in cases where a `0` would be read, "unavailable" would be the correct metric.

Does anyone know the correct SQL manipulation to reset these metrics? I saw that metrics store a `max`, `min`, `avg` and `sum`, but I couldn't get proper logic written for it. I'm thinking `WHERE (current.sum - previous.sum) > :impossible_change_threshold` or such 🤔

cschlipf, 2025-02-03:
> Same here for me with Solax X3 G4. Downgrade back to 2025.1.6 solves the Issue

That is the solution. In the longer term it's no longer recommended to use the LSE-3 due to the rising number of issues. See also the [Sofar FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/). Please consider switching to an RS485 Adaptor.

Ocramius, 2025-02-03:
> I switched to RS485 

@cschlipf do you mean that you switched to a different hardware? If so, which?

cschlipf, 2025-02-03:
I am using the [Waveshare USB Adaptor](https://homeassistant-solax-modbus.readthedocs.io/en/latest/compatible-adaptors/#waveshare-usb-to-rs485-ft232rl). In my case this is directly connected to EVCC and EVCC is providing a ModBus proxy for Home Assistant to connect to via ModBus TCP.

wills106, 2025-02-23:
I don't know if this https://github.com/wills106/homeassistant-solax-modbus/issues/1238#issuecomment-2670890675 will help users with the LSE-3 or not?

But I have released [2025.02.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.1) which changes the timeout to 10. In case anyone is unsure on how to edit the files locally.

Ocramius, 2025-02-23:
@wills106 I can try jumping between `2025.1.6` (working -ish) and ``2025.02.1` for some time, to observe the behaviour .

I don't see any code that renders a timeout into `unavailable` or `unknown` though

Ocramius, 2025-02-24:
Nope, we're back at having spikes: 

[image]
[image]

```
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:716
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: February 23, 2025 at 15:38:41 (1473 occurrences)
Last logged: 07:51:06

zcs_inverter: read failed at 0x692: export_energy_total
zcs_inverter: read failed at 0x694: battery_input_energy_today
zcs_inverter: read failed at 0x696: battery_input_energy_total
zcs_inverter: read failed at 0x698: battery_output_energy_today
zcs_inverter: read failed at 0x69a: battery_output_energy_total
```

Reverting to `2025.01.6`, meanwhile.

I really believe a failed read should not emit a numeric value (current behaviour)

wills106, 2025-02-24:
Could you share the equivalent of this:
(Need to select "Show Extended Logs" under the 3dots on the Logs page)
```
2025-02-23 09:13:52.280 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX sensor groups: 1
2025-02-23 09:13:52.280 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX returning holding block: 0x7d 0xbc [125, 131, 133, 139, 140, 141, 142, 143, 144, 145, 146, 148, 154, 156, 170, 175, 180, 181, 182, 187]
2025-02-23 09:13:52.280 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX returning holding block: 0xe8 0x118 [232, 233, 253, 254, 256, 258, 259, 260, 264, 265, 266, 267, 271, 272, 273, 274, 275, 277, 278, 279]
2025-02-23 09:13:52.280 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX returning input block: 0x0 0x55 [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 19, 20, 21, 22, 23, 24, 28, 29, 32, 33, 35, 36, 37, 70, 72, 74, 80, 82, 84]
2025-02-23 09:13:52.280 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX returning input block: 0x66 0xba [102, 103, 104, 105, 140, 145, 146, 148, 150, 152, 154, 156, 168, 170, 172, 174, 176, 184, 185]
```

On the SolaX X3 Mic inverters we are using ignore read error on some problematic registers.
https://github.com/wills106/homeassistant-solax-modbus/blob/6779c25f4dd2764beb732ff995c7261043d821bf/custom_components/solax_modbus/plugin_solax.py#L6815-L6829

I want to try this on Sofar, but I need to see how the blocks are formed.

Ocramius, 2025-02-24:
<details>
<summary>I found these relevant-ish logs from the 07:00~08:00 local time, where the readings where all broken</summary>

```log
[33m2025-02-24 07:46:37.192 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x1023: feedin_limitation_mode [0m
[33m2025-02-24 07:46:37.192 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x1024: feedin_max_power [0m
[33m2025-02-24 07:46:37.192 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x102b: battery_active_control [0m

[33m2025-02-24 07:46:44.007 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x626: battery_charge_cycle_5 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x48d: voltage_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x48e: current_output_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x48f: active_power_output_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x490: reactive Power_output_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x491: power_factor_output_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x492: current_pcc_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x493: active_power_pcc_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x494: reactive Power_pcc_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x495: power_factor_pcc_l1 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x498: voltage_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x499: current_output_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x49a: active_power_output_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x49b: reactive Power_output_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x49c: power_factor_output_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x49d: current_pcc_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x49e: active_power_pcc_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x49f: reactive Power_pcc_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a0: power_factor_pcc_l2 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a3: voltage_l3 [0m
[33m2025-02-24 07:46:47.521 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a4: current_output_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a5: active_power_output_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a6: reactive Power_output_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a7: power_factor_output_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a8: current_pcc_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4a9: active_power_pcc_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4aa: reactive Power_pcc_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4ab: power_factor_pcc_l3 [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4ae: active_power_pv_ext [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4af: active_power_load_sys [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4b0: voltage_phase_l1n [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4b1: current_output_l1n [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4b2: active_power_output_l1n [0m
[33m2025-02-24 07:46:47.522 WARNING (MainThread) [custom_components.solax_modbus] zcs_inverter: read failed at 0x4b3: current_pcc_l1n [0m
[... cut ...]  4176, 4177, 4178, 4180, 4181, 4182, 4183, 4184, 4186][0m
[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x1104 0x1111 [4356, 4368][0m
[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x1184 0x118d [4484, 4485, 4487, 4489, 4491][0m
[32m2025-02-24 09:37:25.762 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}[0m
```

Ocramius, 2025-02-24:
It looks like there are:

1. wrong reads between 07:11:48 and 07:15:19 today
2. timeouts between 07:47:34 and 07:49:55 (they don't seem to affect any stats)

Here's the list of what's affected/when:

* 07:47:34~07:49:55
  * `Desired Grid Power`
  * `Minimum Battery Power`
  * `Maximum Battery Power`
* 07:11:48 (single second blip?)
    * `Solar Generation Today`
    * `Load Consumption Total`
    * `Import Energy Today`
    * `Import Energy Total`
    * `Export Energy Today`
    * `Export Energy Total`
    * `Battery Input Energy Today`
    * `Battery Input Energy Total`
    * `Battery Output Energy Today`
    * `Battery Output Energy Total`
* 07:12:06~07:15:19
    * `Solar Generation Today`
    * `Solar Generation Total`
    * `Load Consumption Today`
    * `Load Consumption Total`
    * `Import Energy Today`
    * `Import Energy Total`
    * `Export Energy Today`
    * `Export Energy Total`
    * `Battery Input Energy Today`
    * `Battery Input Energy Total`
    * `Battery Output Energy Today`
    * `Battery Output Energy Total`

This is what it looks like, graphed:

[image]

Ocramius, 2025-02-24:
Another data point: I just checked, and I still had the Azzurro ZCS integration loaded.

The [Azzurro ZCS integration](https://github.com/aturri/ha-zcsazzurro) shows these value changes:

```
[32m2025-02-24 07:12:24.179 INFO (MainThread) [custom_components.azzurrozcs.sensor] ZCS Rest response[0m
[32m2025-02-24 07:12:24.179 INFO (MainThread) [custom_components.azzurrozcs.sensor] 0.0[0m
[32m2025-02-24 07:12:24.179 INFO (MainThread) [custom_components.azzurrozcs.sensor] 0.0[0m
[32m2025-02-24 07:12:24.180 INFO (MainThread) [custom_components.azzurrozcs.sensor] 0.0[0m
[32m2025-02-24 07:12:24.180 INFO (MainThread) [custom_components.azzurrozcs.sensor] Return the state: True[0m
[32m2025-02-24 07:12:24.180 INFO (MainThread) [custom_components.azzurrozcs.sensor] Return the state attributes[0m
[32m2025-02-24 07:12:24.180 INFO (MainThread) [custom_components.azzurrozcs.sensor] Power_now = 0.0[0m
```

I'm now wondering:

1. is the LSE-3 LAN stick Logger crashing due to two clients? I disabled AzzurroZCS but this wouldn't explain why this component's version affects the readings
2. if not, is the concurrent reading perhaps affecting Modbus registers?

That said, it now at least seems to be that "unavailable" != "wrong reading": different problems, according to this timeline.

wills106, 2025-02-24:
> Here's the list of what's affected/when:
> 
> * 07:11:48 (single second blip?) no corresponding log entries to be found
>   
>   * `Solar Generation Today`
>   * `Load Consumption Total`
>   * `Import Energy Today`
>   * `Import Energy Total`
>   * `Export Energy Today`
>   * `Export Energy Total`
>   * `Battery Input Energy Today`
>   * `Battery Input Energy Total`
>   * `Battery Output Energy Today`
>   * `Battery Output Energy Total`

The read errors are coming from the same block of registers "Electric Power (0x0680-0x06BF)"

Could you try [2025.02.2b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.2b1)

Ocramius, 2025-02-24:
Giving it a spin, thanks in advance!

Ocramius, 2025-02-25:
I sadly have to report that `ignore_readerror=True,` didn't help there :S 

Today's situation, after letting `2025.02.2b1` run:

[image]

There was a spike that lasted few seconds around `04:57:19~04:57:25`:

[image]

These values all jumped together: there is some sort of correlation.
That time range has no relevant associated logs.

If I "zoom out" over the entire day, I can see how many other stats tend to have these momentary jumps:

[image]

It should suffice to say that my inverter certainly didn't reach 13872°C 😅

Ocramius, 2025-02-25:
Meanwhile, back to `2025.01.6` with my changes too.

Looking at the diff since then ( https://github.com/wills106/homeassistant-solax-modbus/compare/2025.01.6...2025.02.2b1 ) I see a couple of relevant details:

```diff
-from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
+from .payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
```

Is that a typo, or some Pythonism that I am not familiar with?

`custom_components/solax_modbus/payload.py` - I don't see any tests for this, so I can't evaluate what this does / whether it does what it's supposed to 😅 

## PyModbus errors in upstream?

I've also skimmed https://github.com/pymodbus-dev/pymodbus/compare/v3.7.4...v3.8.3 , and the diff is too big for me to understand without domain knowledge: I'm wondering if there's a sensible way to check intermediate versions on my setup?

For example, trying `pymodbus:v3.8.0` before skipping all the way to `v3.8.3`.

Alternatively, `v3.8.6` is also out, and could be tried: upstream is doing tons of simulation additions, so perhaps there's more fixed?

I'm pinging @janiversen just to see if the graphs / blips in the comment above say anything to them (they might already have seen issues like this, so it's worth asking, before running blindly)

janiversen, 2025-02-25:
Why are you pinging me ? I have nothing to do with that component, nor an opinion on what HA does or not does.

Ocramius, 2025-02-25:
Hey @janiversen: sorry, it was really just for an opinion on "spiky values": did pymodbus become more reactive to quick reactive/spiky changes? Is that something you've observed in pymodbus in general? Is that normal / something that should be handled downstream? That's really all I wanted to ask, sorry: feel free to unsubscribe/ignore.

janiversen, 2025-02-25:
Pymodbus works, just run our test suite, but to me it seems like a number of integrations (or maybe only modbus) was upgraded without adapting the code. Pymodbus 3.7 -> 3.8 have at least one change that should have been marked as a breaking change in HA.

janiversen, 2025-02-25:
If there is a problem in pymodbus, please open an issue there, with a pymodbus debug log, showing the problem, otherwise it is impossible to help....but please remember pymodbus is not the only item in the chain if e.g. the component using pymodbus do not do proper error handling you might have random values.

cschlipf, 2025-02-25:
What are you current read interval settings?

Please list all services that access the LSE-3 at the same time

Ocramius, 2025-02-25:
> If there is a problem in pymodbus, please open an issue there

Yeah, I poked to get some potential wisdom on known/seen behaviors: not assuming there's an issue there, but rather something that people have already "seen", and being a maintainer of very popular OSS projects myself, that sometimes leads to "yeah, this is common case XYZ" outcomes :-)

> What are you current read interval settings?

The component is configured as such (via GUI):

* "Modbus address": 1
* "Inverter type": sofar
* "The polling interval of the modbus registers in seconds": 15
* "Medium polling interval": 15
* "Fast polling interval": 15

[image]

> Please list all services that access the LSE-3 at the same time

No other services: everything else was shut down.

cschlipf, 2025-02-25:
Now that's already a lot of load for the LSE-3. There is a reason, why the polling intervals are split up like this. Unless you have a specific reason, why you need this, reduce the polling interval to 15/30/60. If that does not help, try again at 30/60/120. Should be good enough for most typical automations.

See also: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/#i-am-using-the-lse-3-logger-stick-but-requests-timeout-frequently

Ocramius, 2025-02-25:
Giving it a shot, thanks, but beware that timeouts don't seem to be in any logs 🤔

wills106, 2025-02-25:
> component using pymodbus do not do proper error handling you might have random values

I have added in `ModbusException` to `read_holding_registers` & `read_input_registers` (hopefully correctly) [2025.02.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.2b2)
Hopefully that will return a more detailed reason why it's failing. Then we can figure out what's causing the issue.

I have also increased the timeout form 10 to 15.

> Looking at the diff since then ( [2025.01.6...2025.02.2b1](https://github.com/wills106/homeassistant-solax-modbus/compare/2025.01.6...2025.02.2b1) ) I see a couple of relevant details:
> 
> -from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
> +from .payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
> Is that a typo, or some Pythonism that I am not familiar with?
> 
> `custom_components/solax_modbus/payload.py` - I don't see any tests for this, so I can't evaluate what this does / whether it does what it's supposed to 😅

payload decoder has been temporary moved into this custom component while we look at implementing `client.convert_from_registers()` #1223

Ocramius, 2025-02-25:
@wills106 deploying your changes and giving it a spin: likely having more info in 12~16h. Thanks in advance for the quick iterations you are cranking out here!

andbad, 2025-02-26:
I report an additional problem. I have some automations that manipulate the values:
passive_mode_grid_power
passive_mode_battery_power_max
passive_mode_grid_power

It happened twice that the passive_mode_grid_power value is set to 65535 instead of 0, probably in conjunction with a spike like those reported (the automation was triggered when the storage SOC shows a spike at 0).

I'm trying to install [2025.02.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.2b2).

By(t)e

Ocramius, 2025-02-26:
@wills106 reporting from my installation: zero spikes of any kind since my last comment 20h ago.

Everything is running smooth as butter 💪 

Below are just some details that can help further, but I think `2025.02.2b2` is usable.

## Log findings since `2025.02.2b2`

Looking at logs pertinent to the [last introduced exceptions](https://github.com/wills106/homeassistant-solax-modbus/compare/2025.02.2b1...2025.02.2b2).

### Crash due to undefined `_log_error`

Here is a crash indicating that your patch, while working, still leads to further errors (due to missing logger):

```
[31m2025-02-26 10:59:25.452 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request[0m
[32m2025-02-26 10:59:25.456 INFO (MainThread) [custom_components.solax_modbus] exception 'SolaXModbusHub' object has no attribute '_log_error' : zcs_inverter cannot read holding registers at device 1 position 0x488
```

I wonder if a type-checker could help here, in catching these? They aren't a massive issue, and they currently land in my Sentry.

### Crash due to Pymodbus corrupt read

This one looks like incorrect `pymodbus` data that got caught by `MainThread`:

```
[31m2025-02-26 10:56:23.524 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request[0m
[31m2025-02-26 10:56:24.773 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state[0m
```

wills106, 2025-02-26:
Thanks @Ocramius for the feedback.

Even though you have these errors in your log, does the Integration still carry on functioning?

wills106, 2025-02-26:
> Here is a crash indicating that your patch, while working, still leads to further errors (due to missing logger):
>
> ```
> �[31m2025-02-26 10:59:25.452 ERROR (MainThread) [pymodbus.logging] No response [...]

Just released [2025.02.2b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.2b3) which should correct that error.

cschlipf, 2025-02-26:
Note that

```
[31m2025-02-26 10:59:25.452 ERROR (MainThread) [pymodbus.logging] No response received after 6 retries, continue with next request�[0m
```

ist normal with the LSE-3. That is because of the wrong return codes that it returns.

wills106, 2025-02-26:
> ist normal with the LSE-3. That is because of the wrong return codes that it returns.

Hopefully now I have some form of `ModbusException` working for `read_holding_registers` & `read_input_registers` we can at least better handle the wrong codes it responds with.

> Here's the block you asked for instead, [@wills106](https://github.com/wills106) - this probably leaks my serial number, but not that worried :D
> 
> ```
> �[32m2025-02-24 09:37:25.697 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter sensor groups: 1�[0m
> �[32m2025-02-24 09:37:25.697 INFO (MainThread) [custom_components.solax_modbus.sensor] newblock declaration found for empty block�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x404 0x432 [1028, 1029, 1030, 1031, 1032, 1033, 1034, 1035, 1036, 1037, 1038, 1039, 1040, 1047, 1048, 1049, 1050, 1051, 1056, 1057, 1068]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x445 0x45b [1093, 1101, 1103]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x484 0x4bd [1156, 1157, 1158, 1159, 1160, 1161, 1162, 1165, 1166, 1167, 1168, 1169, 1170, 1171, 1172, 1173, 1176, 1177, 1178, 1179, 1180, 1181, 1182, 1183, 1184, 1187, 1188, 1189, 1190, 1191, 1192, 1193, 1194, 1195, 1198, 1199, 1200, 1201, 1202, 1203, 1204, 1205, 1206, 1207, 1208, 1209, 1210, 1211, 1212]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x584 0x58a [1412, 1413, 1414, 1415, 1416, 1417]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x5c4 0x5c5 [1476]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x604 0x60b [1540, 1541, 1542, 1543, 1544, 1545, 1546]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x60b 0x612 [1547, 1548, 1549, 1550, 1551, 1552, 1553]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x612 0x619 [1554, 1555, 1556, 1557, 1558, 1559, 1560]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x619 0x620 [1561, 1562, 1563, 1564, 1565, 1566, 1567]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x620 0x627 [1568, 1569, 1570, 1571, 1572, 1573, 1574]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x627 0x62e [1575, 1576, 1577, 1578, 1579, 1580, 1581]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x62e 0x635 [1582, 1583, 1584, 1585, 1586, 1587, 1588]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x635 0x63c [1589, 1590, 1591, 1592, 1593, 1594, 1595]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x667 0x66a [1639, 1640, 1641]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x684 0x69c [1668, 1670, 1672, 1674, 1676, 1678, 1680, 1682, 1684, 1686, 1688, 1690]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x100a 0x102c [4106, 4131, 4132, 4139]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x1044 0x105b [4164, 4165, 4166, 4167, 4168, 4169, 4170, 4171, 4172, 4173, 4174, 4175, 4176, 4177, 4178, 4180, 4181, 4182, 4183, 4184, 4186]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x1104 0x1111 [4356, 4368]�[0m
> �[32m2025-02-24 09:37:25.713 INFO (MainThread) [custom_components.solax_modbus.sensor] zcs_inverter returning holding block: 0x1184 0x118d [4484, 4485, 4487, 4489, 4491]�[0m
> ```

I also want to relook into the amount of `newblock=True,` we are using for Sofar Inverters. Having so many read requests wont be helping. I know there are some "Black Holes" in the registers on the Sofar mapping, but I am sure we can consolidate them better.
A lot of the `newblock=True,` got introduced in early development, this was before the understanding of the importance of the terminating resistors played in communications especially as the user had two Inverters on the same device.

cschlipf, 2025-02-26:
One more thing: the message only comes as response to write requests.

Note that if you have enabled battery read out, then this is involving a write request to send the battery ID to the inverter.

@Ocramius - did you enable battery red out? Note that because of this problem with the LSE-3 battery read out does not work with the LSE-3.

Ocramius, 2025-02-26:
Nope, battery readout is disabled 😁

Ocramius, 2025-02-27:
Current status: everything operating normally. 

No spikes in any statistics.

Logs do show a lot of small crashes, roughly every 2 minutes (current long polling interval):

```
[31m2025-02-27 01:25:30.402 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 1160 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request[0m
[31m2025-02-27 01:25:30.406 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 681, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 848, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "holding")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 779, in async_read_modbus_block
    if realtime_data.isError():
       ^^^^^^^^^^^^^^^^^^^^^
```

```
[31m2025-02-27 01:29:00.436 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 1028 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request[0m
[31m2025-02-27 01:29:00.440 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 681, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 848, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "holding")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 779, in async_read_modbus_block
    if realtime_data.isError():
       ^^^^^^^^^^^^^^^^^^^^^
```

```
[31m2025-02-27 02:04:55.976 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 1547 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request[0m
[31m2025-02-27 02:04:55.980 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 681, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 848, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "holding")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 779, in async_read_modbus_block
    if realtime_data.isError():
       ^^^^^^^^^^^^^^^^^^^^^
```

```
[31m2025-02-27 02:08:06.020 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 1668 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request[0m
[31m2025-02-27 02:08:06.025 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 681, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 848, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "holding")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 779, in async_read_modbus_block
    if realtime_data.isError():
       ^^^^^^^^^^^^^^^^^^^^^
```

```
[31m2025-02-27 02:11:18.840 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 4106 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request[0m
[31m2025-02-27 02:11:18.844 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
[... 53 lines cut ...]
       ^^^^^^^^^^^^^^^^^^^^^
```

```
[31m2025-02-27 02:31:29.899 ERROR (MainThread) [custom_components.solax_modbus] Error: device: 1 address: 4484 -> Modbus Error: [Input/Output] No response received after 6 retries, continue with next request[0m
[31m2025-02-27 02:31:29.903 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 681, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 848, in async_read_modbus_registers_all
    res = res and await self.async_read_modbus_block(data, block, "holding")
[... cut ...]
The above are just examples taken from various addresses read failures.

I'd say (and it's just intuition, not proof) that the LSE-3 is really struggling, whenever a batch of reads comes in.

I've set the read offsets to `31`, `61`, `121`, so that they tend to avoid polling at the same time, but perhaps we're really asking too much of this stick, at the same time: I wonder if some stats could be disabled, perhaps?

That said, IMO we can close here, and we can move such a discussion to a new issue.

wills106, 2025-02-27:
Hopefully when we look at reducing the amount of `newblock=True,` we are using for Sofar Inverters, then less demand will be placed on the LSE-3. As we will hopefully be requesting fewer groups of registers.

Ocramius, 2025-02-27:
Closing here meanwhile: let's call this "done" with ` 2025.02.2b2` meanwhile 🚀

closed 2025-02-27 by Ocramius