wills106/homeassistant-solax-modbus #1233: [Bug]: X3 Ultra - Inverter power always 0

styriaman, 2025-01-25:
### Describe the bug

HA displays always 0 as inverter power  although the sum of inverter power1 to inverter power3 is not equal to 0
I haven't tried it yet, but it will probably also have negative effects on Remote Control, where ‘houseload_nett’ is also calculated from it

Perhaps it can be calculated in the 1st step from the sum of inverter power 1 to 3 

[image]

### Integration Version

2024.12.2

### Homeassistant core version

2024.12.5

### Inverter brand

Solax X3 Ultra 

### Plugin used

plugin_solax.py

### Serial prefix

H3BC30

### Inverter firmware versions

ARM 018.02-000.04 DSP 017.05

### Connection Method

SolaX Pocket Wifi 3.0

### Dongle firmware

not relevant

styriaman, 2025-01-26:
Possible working changes in plugin_solay.py:

Add

def value_function_inverter_power_g5(initval, descr, datadict):
    return (
        datadict.get("inverter_power_l1", 0)
        + datadict.get("inverter_power_l2", 0)
        + datadict.get("inverter_power_l3", 0)
    )

and

   SolaXModbusSensorEntityDescription(
        name="Inverter Power",
        key="inverter_power",
        value_function=value_function_inverter_power_g5,
        native_unit_of_measurement=UnitOfPower.WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        allowedtypes=AC | HYBRID | GEN5,
    ),

and remove  | GEN5 from the register based "Inverter Power" EntityDescription 

Enjoy

wills106, 2025-01-30:
Just released [2025.01.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.01.9) with the suggested change.

closed 2025-02-03 by styriaman