wills106/homeassistant-solax-modbus #1234: [Bug]: 'Growatt' - 'SPF 5000 ES' unrecognized Growatt inverter type-firmware version : unknown

Theodor0001, 2025-01-26:
### Describe the bug

Tried to connect to my inverter but nothing happened except that my web browser disconnected from HA

### Integration Version

2025.01.8

### Homeassistant core version

2025.1.4

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

113.01/114.01

### Connection Method

Industrial USB to RS485 Bidirectional Converter, waveshare

### Detailed Error Log

Following is 10 errors starting from the top of the list marked 1) to 10)

1)
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_growatt
Source: custom_components/solax_modbus/plugin_growatt.py:6011
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:59:08 PM (209 occurrences)
Last logged: 7:37:16 PM

unrecognized Growatt inverter type - firmware version : unknown

2)
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_growatt
Source: custom_components/solax_modbus/plugin_growatt.py:5974
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:59:08 PM (209 occurrences)
Last logged: 7:37:16 PM

Growatt: cannot find firmware version, even not for other Inverter

3)
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_growatt
Source: custom_components/solax_modbus/plugin_growatt.py:70
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:59:07 PM (418 occurrences)
Last logged: 7:37:16 PM

Growatt: reading firmware number from address 0x9 failed; other address may succeed
Growatt: reading firmware number from address 0xbb9 failed; other address may succeed

4)
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_growatt
Source: custom_components/solax_modbus/plugin_growatt.py:64
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:59:07 PM (418 occurrences)
Last logged: 7:37:16 PM

Growatt: attempt to read firmware failed at 0x9
Growatt: attempt to read firmware failed at 0xbb9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 64, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=5)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address=address, count=count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 103, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg_read.ReadHoldingRegistersRequest(address=address, count=count, slave=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 92, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusSerialClient /dev/ttyUSB0:0]

5)
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:533
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:59:07 PM (627 occurrences)
Last logged: 7:37:16 PM

Unable to connect to Inverter at /dev/ttyUSB0:0

6)
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:525
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 6:59:07 PM (627 occurrences)
Last logged: 7:37:16 PM

Failed to connect [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

7)
Logger: homeassistant.components.go2rtc.server
Source: components/go2rtc/server.py:156
integration: go2rtc (documentation, issues)
First occurred: 7:21:56 PM (6 occurrences)
Last logged: 7:21:56 PM

18:59:03.110 INF config path=/tmp/go2rtc_s0ga0fzt.yaml
18:59:03.110 INF [rtsp] listen addr=127.0.0.1:18554
18:59:03.111 INF [webrtc] listen addr=:18555/tcp
18:59:03.111 INF [api] listen addr=127.0.0.1:11984
exit with signal: terminated

8)
Logger: homeassistant.components.go2rtc.server
Source: components/go2rtc/server.py:186
integration: go2rtc (documentation, issues)
First occurred: 7:21:56 PM (1 occurrences)
Last logged: 7:21:56 PM

Go2rtc unexpectedly stopped, server log:

9)
Logger: homeassistant.components.go2rtc.server
Source: components/go2rtc/server.py:149
integration: go2rtc (documentation, issues)
First occurred: 7:21:56 PM (1 occurrences)
Last logged: 7:21:56 PM

exit with signal: terminated

10)
Logger: homeassistant.config_entries
Source: config_entries.py:855
First occurred: 7:16:51 PM (2 occurrences)
Last logged: 7:16:51 PM

Error unloading entry Growatt for sensor
Error unloading entry Growatt for switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 855, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 101, in async_unload_entry
    return await hass.data[DATA_COMPONENT].async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!

### Additional context

thank you for this contribution to HA, if it can get to work, it will really help me out and a lot of people that have the same issue with trying to get away from the official GW data logging.

I'm using HA on a Oracle VirtualBox virtual machine, I'm doing this to test out HA before purchasing a dedicated device. I have installed a new instant of HA to try to test how i can connect to my Growatt inverter with proper real-time stable updating eg 1 sec intervals to actively switch switches on and off.
I'm having problems with this ne instance that when loading it keeps giving a supervisor failed to load error, but i don't think it is related

HellixFix, 2025-01-26:
same problem with MOD 10KTL3

PSMGoossens, 2025-03-09:
@Theodor0001  Could you enable full debug logging for solax and paste the log file here?

flopp999, 2025-04-03:
Still problem?
Maybe the problem is the USB dongle that is not working or your setup in Virtual that is wrong.
@HellixFix what connection do you have from HA to inverter?

HellixFix, 2025-04-03:
problem gone after some update

flopp999, 2025-04-03:
> problem gone after some update

Great. You can close this issue by clicking on Close at the bottom

HellixFix, 2025-04-03:
> > problem gone after some update
> 
> Great. You can close this issue by clicking on Close at the bottom

Not my issue :-) No permissions.

flopp999, 2025-04-03:
> > > problem gone after some update
> > 
> > 
> > Great. You can close this issue by clicking on Close at the bottom
> 
> Not my issue :-) No permissions.

Sorry, I thought you started this issue.
@Theodor0001 still problem for you?

wills106, 2025-08-28:
Closing down as no response.

closed 2025-08-28 by wills106