wills106/homeassistant-solax-modbus #1237: [Bug]: stellt selbst den Modus wieder zurück

xLex8x, 2025-01-27:
### Describe the bug

Wenn ich über Ha den Modus zb. Von manuell in Backup ändern möchte, dann springt er selbst wieder zurück.
Wenn ich in der Solax App dann auf Backup stelle und dann im Ha auf Manuelll möchte, springt er ebenfalls wieder zurück.

Automationen sind alle deaktiviert.

Hat dieses Problem schon jemand lösen können?

### Integration Version

Neueste

### Homeassistant core version

Neueste

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

.

### Inverter firmware versions

.

### Connection Method

Wlan

### Dongle firmware

.

### Additional context

[image]

steps56, 2025-01-28:
War bei mir nach neusten Updates HA core usw. und 2 Neustarts weg, also Mode Änderungen wurden wieder fest gespeichert.

cschlipf, 2025-02-03:
The maintainer does not speak German.

steps56, 2025-02-03:
Everything okay, restart twice solved the problem, as told in german above already.

closed 2025-02-04 by wills106