wills106/homeassistant-solax-modbus #1238: [Bug]: Sudden peaks appearing in the measured data on Solax X3 G4

shaarkys, 2025-01-28:
### Describe the bug

Hi, 

sudden anomalous peaks are appearing in the Solax integration data, most probably when inverted stops responding for a while. These spikes are not observed in other platforms like Homey (same sampling rate) or the Solax Cloud. The setup uses Modbus TCP via a Multiplexer and had been stable for months prior to the upgrades.
I thought it was caused by V3 Wifi+LAN dongle upgrade that was done remotely, which caused inverter to lock several times during the day, but after reverting it, the Lock issues are gone but the peaks remains.

**While I have to clarify why modbus disconnects (visible on Homey logs) suddenly for a moment, is there some possibility to avoid such peaks ?** Apologies if this would be some kind of HW failure on my side but trying to rule out everything. 

[image]

[image]

[image]

### Integration Version

2025.01.8 (testing now 2025.01.6 to see any difference)

### Homeassistant core version

2025.1.4

### Inverter brand

Solax X3 G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP v1.46 ARM v1.44

### Connection Method

Solax Wifi+LAN adapter V3 with fw via LAN 

### Dongle firmware

1.003.11 (reverted as troubleshooting steps from 1.005.04 )

### Detailed Error Log

[image]

```
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:771
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: January 27, 2025 at 20:04:19 (21864 occurrences)
Last logged: 11:25:39

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
```

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:670
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: January 27, 2025 at 21:51:40 (6 occurrences)
Last logged: 10:59:20

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 670, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 848, in async_read_modbus_registers_all
    data[descr.key] = descr.value_function(0, descr, data)
                      ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 300, in value_function_battery_voltage_cell_difference
    return datadict.get("cell_voltage_high", 0) - datadict.get("cell_voltage_low", 0)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
```

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:716
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: January 27, 2025 at 21:51:39 (248 occurrences)
Last logged: 10:59:20

Solax X3 G4: read failed at 0xbd: cell_voltage_low
Solax X3 G4: read failed at 0xbf: battery_soh
Solax X3 G4: read failed at 0xc8: grid_frequency
Solax X3 G4: read failed at 0xc9: grid_voltage
Solax X3 G4: read failed at 0xca: grid_voltage_l1
```

```
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:670
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 08:50:12 (4 occurrences)
Last logged: 08:56:03

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 658, in async_write_registers_multi
    resp = await self._client.write_registers(address=address, values=payload, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 155, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
```

```
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 670, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all(group)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 874, in async_read_modbus_registers_all
    await self.async_write_registers_multi(
    ...<3 lines>...
    )
  File "/config/custom_components/solax_modbus/__init__.py", line 661, in async_write_registers_multi
    raise HomeAssistantError(f"Error writing multiple Modbus registers: {original_message}") from e
homeassistant.exceptions.HomeAssistantError: Error writing multiple Modbus registers: Modbus Error: [Input/Output] No response received after 6 retries, continue with next request
```

### Additional context

Note - around 9AM I upgraded firmware to DSP v1.46 ARM v1.44 as another troubleshooting step.

shaarkys, 2025-01-29:
Seems that  2025.01.6 is not showing such sudden peaks, despite of short/random disconnections (at least based on this short test).

Just wonder, is it due to the recent Python related changes and will this change maybe address it ? 
https://github.com/wills106/homeassistant-solax-modbus/commit/048ac759f9059ce267d6a9a5d4336b361cade70b

Shortykiller, 2025-01-30:
Hi

Just want to state that I to have the same problem with random spikes. Im new to the integration, so have no history to cross check with.

My Solar System is Growatt, running Modbus/TCP custom esp32, directly on the modbus com port.

thkn1973, 2025-01-30:
I should check github more often as soon as errors occur.
I have also had these spikes since then.

[image]

matejkaf, 2025-01-30:
Anyone tried latest release 2025.01.9? Spike Problem solved? At the moment I am stuck at 2025.01.6

Shortykiller, 2025-01-30:
> Anyone tried latest release 2025.01.9? Spike Problem solved? At the moment I am stuck at 2025.01.6

No, not yet.. I downgraded to 2025.01.6 yesterdays to check if that solved it temporary..

shaarkys, 2025-01-30:
> Anyone tried latest release 2025.01.9

I did, running it almost full day, it looks much better... but still some smaller peaks appears

[image]

But finally no crazy reading in energy ... !

[image]

shaarkys, 2025-01-30:
Hmm, so reverting back ;-(

Zwer2k, 2025-01-31:
> Anyone tried latest release 2025.01.9? Spike Problem solved? At the moment I am stuck at 2025.01.6

No. Complety destroys my Statistics. Back to 2025.01.6

[image]

shaarkys, 2025-01-31:
Just use /developer-tools/statistics and remove faulty values - it's better to do it the same day it happens, it's faster then. ;-)

shaarkys, 2025-02-02:
So far 2025.01.6 is really working fine for me.

@wills106 , any thoughts on that please ?

wills106, 2025-02-02:
Integration 2025.01.6 is using pyModbus 3.7.4
Integration 2025.01.8 and above is using pyModbus 3.8.3 to match HA.

The developer of pyModbus is striving to make pyModbus more Modbus compliant, which many Data Loggers such as the SolaX PocketWiFi / PocketWiFi + LAN and the Sofar LSE-3 aren't.

With the Sofar LSE-3 the latest firmware ME_0D_270A_1.11 made the Data Logger even more uncompliant, people are recommended to downgrade to ME_0D_270A_1.09. But even then they are still getting issues.

I will be updating the documentation to recommend people to use a USB-RS485 or an Ethernet-RS485 device such as the Waveshare as these are less problematic, instead of these Data Loggers.

It is possible that improvements could be made to the Integration using pyModbus 3.8.3 to improve things, but I own a single Inverter with built in Modbus over TCP. So I can't actively improve support for data loggers that aren't fully Modbus compliant, as I don't own one nor can I use one on my Inverter.

shaarkys, 2025-02-02:
Thank you @wills106 , I thought it was related to pyModbus ...however in my case, I use Wifi V3 LAN+Wifi dongle, accessing it via TCP Multiplexer on Solax X3 G4, worked very much fine last year.  So not sure what do you mean with Data Loggers ? 
Also I'm quite sure those peaks just occurs when dongle temporarily refuse to provide data (yet still being accessible).

_In my case, I can't use WaveShare RS485, haven been this route (even created some tickets some time ago) and never worked for me reliably, not sure why, but also was getting peaks. One of the reason could be that Wifi + RS485 basically creates two devices accessing the Modbus, not sure_

Zwer2k, 2025-02-02:
I use WaveShare RS485 and have the peaks that did not exist until version 2025.01.6. In my case, the data is read with two end devices (Solax Modbus + EVCC) via Modbus proxy. I have seen that there have been no changes to the integration that could cause these spikes, it must actually be due to the new pyModbus version.
This is annoying, especially since I've been struggling with pyModbus compatibility issues for two months now. Solax forces new versions of pyModbus all the time, which in turn leads to problems with other integrations that also use pyModbus. In my case Huawei-Solar.  I keep having to switch to older Ha-Core, Huawei and Solax versions.  
I don't think it's a good idea to use versions of libraries that are not yet included in HA-Core.

Translated with DeepL.com (free version)

wills106, 2025-02-03:
pyModbus 3.8.3 is in HA core, it's been in there for about a month now.

https://github.com/home-assistant/core/pull/134809

I try and keep the Integration matched with HA, as some users also use the HA Modbus client directly for Air source Heat Pumps etc.

shaarkys, 2025-02-03:
> ...however in my case, I use Wifi V3 LAN+Wifi dongle, accessing it via TCP Multiplexer on Solax X3 G4, worked very much fine last year. So not sure what do you mean with Data Loggers ?
> Also I'm [...]

@wills106 and please your response to peaks on Solax / with Wifi Dongle V3 LAN+Wifi ? Those peaks were not there so it must be something with the library itself. I checked changelog, haven't seen anything obvious and also no "bug" planned to be fixed in the next version, but I'm not expert ;-(  I hope it will not result into "dongle unsupported / not recommended" anymore ;-)

Zwer2k, 2025-02-04:
Yesterday I connected the integration directly to the WaveShare RS485 adapter (without Modbus proxy) and since then I have had no more spikes. EVCC continues to run via Modbus proxy.
The problem only seems to occur in association with Modbus proxy. Version of Modbus proxy is still the same as before the problem, I have not changed anything

matomatusov, 2025-02-06:
Hi,
It happens to me that some registries fall out. In the past, everything was OK.

[image]

darkrain-nl, 2025-02-06:
Just want to share I see peaks as well with plugin_sofar...

steps56, 2025-02-07:
Me too a few rare times at SolaX.

HA-Supi, 2025-02-09:
I can also confirm this behavior with Qcell's Hyp-G3 (Solax X3 G4). 
Data retrieval via Modbus TCP / Stick 3.0 FW 19.01 / Inverter FW 48 08 45

 Please take a look at  the following information on troubleshooting >  Wills106: 
I switched from the 2024.05.3 integration to 2025.01.2 in January 2025. Since I use a tile with FW info, the omission of the entity bothered me. 
Previously, the integration ran for 6 months without any problems. 

I suspect that there are data errors during retrieval that are not recognized. (no data checksum?) 

Here are a couple of screenshots: 
They do not show all entities with errors... only a selection.

[image]

[image]

[image]

[image]

[image]

steps56, 2025-02-09:
Yes, I got same Errors at my system.
Newest Modbus Integration, newesst HA core and supervisor and Qcells/SomaX brandnew Firmware 1.50/1.47 from this week.

After each newstart of HA, the depricates error shows up once too !

shaarkys, 2025-02-11:
@wills106 , you probably know already but seems that there are some changes in PyModbus 3.8.4, 3.8.5 and 3.8.6, which could help

https://pymodbus.readthedocs.io/en/latest/source/changelog.html

godfreym29, 2025-02-13:
This also issue is also causing issues withe inverter charge and discharge time on Solis.
Most of the times the entity values are to high to do anything, but sometimes they are within the 0-23 for hours or 0-59 for minutes and can cause the inverter to discharge the battery, on occasions emptying the battery.

steps56, 2025-02-15:
I went back to 2025.01.6
No spikes... and hopefully wills106 may find the reason why newest Version pruduces those spikes.... and may find a solution.
Hope for solution update soon by 2025.02....

godfreym29, 2025-02-15:
Just to confirm
Downgrading to 2025.01.6 has resolved the issue. 
6 hours without any peaks or issues with the inverter charge and discharge time on Solis.

steps56, 2025-02-16:
@wills106
Any news from your side about working at the spikes problem and to find the reason and a solution ?
Downgrading and to stuck at 2025.01.6 don't look like a permanent solution to me.
I love your work/integration, but communikation at github generally lacks ... its always guessing about "work in progress"... lr sadly ben ignored at all 🤷🏼‍♂️.
Bye for now and please answer about this, thanks in advance.

shaarkys, 2025-02-16:
@steps56 I'm sure wills106 will provide some solution, when / if it will be ready, Probably currently busy with different priorities but asking repeatedly for updates is not helping anyway. Let's be patient here.

steps56, 2025-02-16:
> [@steps56](https://github.com/steps56) I'm sure wills106 will provide some solution, when / if it will be ready, Probably currently busy with different priorities but asking repeatedly for updates is not helping anyway. Let's be patient here.

Asking just for any Info by communication, just to know about... and not always only asume, guess... to know just nothing about "work in progress".
(Thats what I hate most concerning Github)
Communication is always not any miracle, but AVOIDING any (un)necessary asking.
Not that hard to understand, isn't it ?

shaarkys, 2025-02-16:
Yep, not also hard to understand nobody was asking for comment nor reply ;-) As all those who commented here are also automatically receiving notifications, it's something which is not welcome either. 

But don't worry, just avoiding developer to be, unintentionally, mad. It's quite hard for any developer to read any "demands" when the developer is working on it during his free time or almost for free.  

That's last reply from my side, we have enough proof it's caused by the library used, also proved temporary workaround, so currently there is nothing to be hurried about.

wills106, 2025-02-20:
I have noticed in this [comment](https://github.com/home-assistant/core/issues/138077#issuecomment-2648132614)

There is mention about increasing the timeout to 9.
Could someone who's having this issue try and see if 9 or 10 makes an improvement?

https://github.com/wills106/homeassistant-solax-modbus/blob/c889aa77137fd181252676c34b1a29a24efde1b4/custom_components/solax_modbus/__init__.py#L261-L262

I can't find any reference to what `message_wait_milliseconds:` does in the HA Integration as it doesn't seem to link to anything in the pyModbus Library.

darkrain-nl, 2025-02-20:
I have set it  to 9 and restarted HA, let's wait and see :)

steps56, 2025-02-20:
I now installed last Integration version .9 again and set the timout to 10 at _init_.py line 262 as mentioned by wills106 before.
Hope it will avoid the spikes... I watch it and will tell here anyway.
TANKS A LOT for investigating this problem to @wills106 and all others involved. 👌

steps56, 2025-02-20:
Update/Edit:
As I'm using Modbus Tcpip (SolaX newest Lan/Wlan Dongle using Wlan connection), I set the 2 other timeout settings also at now all, to 15 Sec. timeout.

I assume, @wills106 hint war "just" for the "waveshare" users and not Tcpip users as myself.
Tell me please, if I'm wrong about this and/or about the 15 Sec. setting for safety.

See screen about my edited lines concerning timeout:

[image]

wills106, 2025-02-20:
I thought it was mostly the PocketWiFi that was playing up?
But it's worth trying a change or 10 or whatever to see if it improves it or not for anyone getting these spikes.

I'm not getting any of these spikes, so there isn't much I can do locally.

steps56, 2025-02-20:
> I thought it was mostly the PocketWiFi that was playing up? But it's worth trying a change or 10 or whatever to see if it improves it or not for anyone getting these spikes.
> 
> I'm not getting any of these spikes, so there isn't much I can do locally.

Yes, its the PocketWifi about the spikes.

Your change about timeout line 262 looks as "just" concerning the rtu / Waveshare connection only.

Modbus via tcpio = Pocketwifi Dongle may be connected to the next lines 264 and 266, if I'm not wrong.
So I assumed to change timeout there too.
Wrong or right 🤷🏼‍♂️.
Thanks about any infos and again for your really important work 👍.

Thrasher2020, 2025-02-20:
Upgraded and modified to timeout=10

Will feed back after a few days (unless it spikes soon!)

steps56, 2025-02-20:
Just my "2 Cents" additional:
If anybody is using the "Modbus Proxy Addon" (and I assume, many will necessarily do like me), I set timeout to 15 Sec at -init_.py to avoid collisions (?) to Proxy's 10 Sek. intervall (screen shows my settings):

[image]

darkrain-nl, 2025-02-21:
> I have set it to 9 and restarted HA, let's wait and see :)

And I have a peak, will now try 10...

HA-Supi, 2025-02-21:
I downgraded to 2025.01.06... Because of These Spikes.. But now no Spike, but "lost Data".. 
[image]
[image]

steps56, 2025-02-21:
> I downgraded to 2025.01.06... Because of These Spikes.. But now no Spike, but "lost Data".. [image] [image]
> 
> I will try the bugfix mentioned above.

2x kpl. HA Neustart und es läufg meistens wieder.

wills106, 2025-02-23:
> Your change about timeout line 262 looks as "just" concerning the rtu / Waveshare connection only.
> 
> Modbus via tcpip = Pocketwifi Dongle may be connected to the next lines 265 and 266, if I'm not wrong. So I assumed to change timeout there too. Wrong or right ???🤷🏼‍♂️.

Sorry wrong line, if using the default Modbus TCP it should be this one set to 9 or 10:
https://github.com/wills106/homeassistant-solax-modbus/blob/c889aa77137fd181252676c34b1a29a24efde1b4/custom_components/solax_modbus/__init__.py#L268

But no harm in setting them all.

wills106, 2025-02-23:
Released [2025.02.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.1) which changes the timeout to 10. In case anyone is unsure on how to edit the files locally.

Thrasher2020, 2025-02-24:
Had two spikes today with the latest update. Reverting again.

steps56, 2025-02-24:
> Had two spikes today with the latest update. Reverting again.

Try setting all 4 timeout settings from 10 to 15 Sec. at the _init_.py
I get no spikes since.

SpiritofGS, 2025-02-24:
> Had two spikes today with the latest update. Reverting again.

Unfortunately, I have to confirm this. Will change the settings now to 15. I'll keep watching.

shaarkys, 2025-02-24:
No peaks for me on Solax X3 G4 with TCP Multiplexer, using [2025.02.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.1)

[image]

steps56, 2025-02-24:
> No peaks for me on Solax X3 G4 with TCP Multiplexer, using [2025.02.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.02.1)
> 
> [image]

As assumed by wills106 before (and me too), its mainly a problem with the Wlan Dongle tcpip connection.
I assume too, its a collision of setting of polling intervalls set at the Integrationen config.
(Look for details about at elder post here I made)

steps56, 2025-02-24:
@shaarkys
I assume, lower settings at timeout just 10 Sec. will / may collide with the 15 Sec. at Integration config (I use 25, 20, 15 from low to fast).

@SpiritofGS
Glad to read, its not just a Dongle problem.
Was a few time told before by other users, that waveshare a. s. o. connection doesn't produce any spikes, just the Dongle tcpip connected setups.
Glad about its not true, as I don't want to use other connection as the easy Dongle anyway.
So no need to switch to waveshare to me, also because not want to loose any settings only giving an access by App way.

HA-Supi, 2025-02-24:
Just to tell my experience: 
Using the Addon since Summer 2023 via wifi Dongle. A long Time without this Bug. 

Spikes started at the end of 2024.. at least then i noticed this bug.
Before it also worked with request Intervall every 5 sec.. 

Between 06/24-12/24 i was using Addon Version 2024.05.03 .. 
Switched to current Release in 2025. 

Maybe it also depends on the inverter FW?

steps56, 2025-02-24:
May be... I use newest SolaX Inverter Firmware 1.50/1.47 and T30 Battery 4.0, since last days of 2024.

romankysely, 2025-02-24:
> [@SpiritofGS](https://github.com/SpiritofGS) Glad to read, its not just a Dongle problem. Was a few time told before by other users, that waveshare a. s. o. connection doesn't produce any spikes, just the Dongle tcpip connected setups. Glad about its not true, as I don't want to use other connection as the easy Dongle anyway. So no need to switch to waveshare to me, also because not want to loose any settings only given by an access the SolaX/Qcells App way.

I have Waveshare and no spikes at all. OS 14.2, Core 25.2.5, Supervisor 25.02.1, Solax Modbus 25.01.9
BTW you can replace the dongle with another one and you will not lose historical data. Data are bounded to the dongle. Keep the old key in the configuration and add the new one and continue... This is message form Solax support.

Shortykiller, 2025-02-25:
I am now on day 5 with the increased timeout to 10, and have had no peaks on 2025.01.9

shaarkys, 2025-02-25:
Just got peak consumption on 2025.02.1, reverting to  2025.01.6

[image]

[image]

Yet, it was first "peak" after 24 hours or so.

@wills106 , so new pyModbus is not worth to try ? I understand that the frequency of those issues with higher timeouts will be less and less, but why they start appearing now, after you changed pymodbus ? Sorry, not expert here.

 https://pymodbus.readthedocs.io/en/latest/source/changelog.html

Version 3.8.6[](https://pymodbus.readthedocs.io/en/latest/source/changelog.html#version-3-8-6)

    Allow id=0 and check if response.id == request.id. (#2572)

Version 3.8.5[](https://pymodbus.readthedocs.io/en/latest/source/changelog.html#version-3-8-5)

    New simulator is WIP, not to be used. (#2568)

    dev_id=0 no response expected (returns ExceptionResponse(0xff)). (#2567)

    New simulator datastore. (#2535)

Version 3.8.4[](https://pymodbus.readthedocs.io/en/latest/source/changelog.html#version-3-8-4)

    Parameterize string encoding in convert_to_registers and convert_from_registers (#2558)

    Fix client modbus function calls in remote by adding count as keyword argument (#2563)

    Fix exception text in ModbusPDU.validateAddress (#2551)

    Typo arround no_response_expected (#2550)

    Trace new connection in server. (#2549)

    Add trace to server.

    Update misleading DATATYPE text. (#2547)

    Fix pylint.

    Clarify server usage.

    Solve instable transaction testing. (#2538)

wills106, 2025-02-25:
You can try a newer version locally by altering
https://github.com/wills106/homeassistant-solax-modbus/blob/b523f783bfe2ee5a6c7a3918057a319968cdb950/custom_components/solax_modbus/manifest.json#L12

I try and stick to the same version as HA, as other users use this Integration and the built in Modbus for other uses such as Heatpumps etc. You can only have one version of pyModbus installed at a time.

I don't really understand what's changed between 3.7.4 and 3.8.3 to cause the spikes.

Thrasher2020, 2025-02-26:
I had 3 spikes in 3 hours with the latest build. Gone back to .6 again. Will await further developments.

s-fordham, 2025-02-26:
I’ve tried using the pre-release version **2025.02.1b2** that increases the timeout, and so far so good for the last 7 hours.

For anyone that may find it useful, for the sensors that cannot be modified in Statistics in HA, I managed to clear the spikes using this article, was straight forward, but use at your own risk as modifying the DB tables.

/[https://adriaanrol.com/posts/2024/editing_historical_data_home_assistant/](url)

shaarkys, 2025-02-26:
Not having exactly spikes, but those sudden decreases and strange readings about consumption from 1AM morning...trying now with pymodbus 3.8.6

[image]

shaarkys, 2025-02-26:
> /[https://adriaanrol.com/posts/2024/editing_historical_data_home_assistant/](url)

https://adriaanrol.com/posts/2024/editing_historical_data_home_assistant

Thanks for sharing.

steps56, 2025-02-26:
Modbus 3.8.6 ???
I read here about 3.8.3 as newest, isn't it?
Please tell (who knows !) about a few points:
1.
What's newest Modbus Version?
2.
Ist just changing the number at manifest.json all to do for changing version used by integration?
3.
Does it make sense to change Modbus versions by just guessing about its changelog/changes ?

Please bring any light to all this.

By the way, I've got no spikes the last few days, since altering to 15 seconds timeout, using latest official Integration with its Modbus 3.8.3 version and the 25, 20, 15 sec polling times at config.

wills106, 2025-02-26:
> I had 3 spikes in 3 hours with the latest build. Gone back to .6 again. Will await further developments.

@Thrasher2020 Was this on 2025.02.2b2?

Did you get any ModbusException's in the error log?
Should look like:
https://github.com/wills106/homeassistant-solax-modbus/blob/ad1292330ba8c171cd23e14f5190d97769be7d2e/custom_components/solax_modbus/__init__.py#L549

[158 more replies]

closed 2025-04-30 by shaarkys