wills106/homeassistant-solax-modbus #1240: [Bug]: Sensor Home

ChristophPeterseil, 2025-01-30:
### Describe the bug

Since the Last Update of this Integration (to Version 2025.1.4) the sensor "House Load" brings neagtive values and the values have nothing to do with my house load!
Bevor the Update it worked perfectly.

[image]

### Integration Version

2025.01.9

### Homeassistant core version

2025.1.4

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H35A15

### Inverter firmware versions

DSP 4 ARM 12.5

### Connection Method

Pocket Wifi 3.0

### Dongle firmware

1.005.04

wills106, 2025-02-23:
Possibly linked to random spikes?

https://github.com/wills106/homeassistant-solax-modbus/issues/1238#issuecomment-2670890675

https://github.com/wills106/homeassistant-solax-modbus/issues/1238#issuecomment-2676736589

ChristophPeterseil, 2025-02-23:
no there are no spikes,
I have the same situation, since the last update of the integration.

markushue, 2025-02-27:
same problem here!

closed 2025-09-16 by github-actions[bot]