wills106/homeassistant-solax-modbus #1241: [Bug]: Meter2 Today values do not reset daily

rudibarani, 2025-01-31:
### Describe the bug

We have connected a second inverter to Meter2. The Values of "SolaX Inverter Meter 2 Export Today" and "SolaX Inverter Meter 2 Import Today" should reset every day at midnight. Instead, they resets about every week, but not consistently:

[image]

[image]

### Integration Version

2025.01.9

### Homeassistant core version

2025.1.4

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34B15

### Inverter firmware versions

ARM 1.45 and DSP 1.48

### Connection Method

Waveshare 2-CH RS485 to RJ45 Ethernet Converter Module with POE Function

### Dongle firmware

-

wills106, 2025-02-03:
Does it help if you change `state_class=SensorStateClass.TOTAL_INCREASING,` to `state_class=SensorStateClass.TOTAL,`

https://github.com/wills106/homeassistant-solax-modbus/blob/f8caa1fd35268e13414a6703f7d905f975c518e6/custom_components/solax_modbus/plugin_solax.py#L5087-L5116

Just not sure if [`last_reset`](https://developers.home-assistant.io/docs/core/entity/sensor/#entities-representing-a-total-amount) also needs to be set?

rudibarani, 2025-02-03:
Thanks for looking into this. I have changed the state_class for both sensors. I will keep an eye on them for next days and will get back to you in a few days.

rudibarani, 2025-02-07:
I have changed the state_class for both daily meter2 sensors, but the reset behavior did not change. Any further ideas, what I could try?

wills106, 2025-02-23:
Possibly raise this as an issue with SolaX. There might be a firmware update correcting this?
I don't use the 2nd Meter.

knogler, 2025-03-13:
I've had the same problem since October 30. I received an update from Solax here.

[image]

closed 2025-09-16 by github-actions[bot]