wills106/homeassistant-solax-modbus #1250: Question about Modbus Support for Solax X3 MIC Gen1

Johan219, 2025-02-05:
Hello,

I’ve been trying to integrate my Solax X3 MIC Gen1 inverter with Home Assistant using Modbus (via RS485 to WiFi/Ethernet module), but I haven’t been able to get any data from it after two full days of troubleshooting.

Setup Details:

-     Inverter Model: Solax X3 MIC Gen1
-     Firmware Version: Arm 1.00
-     Connection Method: RS485 → Waveshare RS485 to WiFi/Ethernet module

Current Status:
-     MQTT: Device successfully connects, but no data is received.
-     Modbus RTU (QModMaster): Connection fails (no response).
-     Telnet Check: Port 8899 is open, but no Modbus data is received.
-     Tried Addressing: Unit ID 1 & 11, multiple baud rates.
-     Firmware Issue? Documentation states that Modbus support starts from firmware Arm 1.38 – I am still on Arm 1.00 and can’t find an update.

Questions:

-     Does anybody know if the X3 MIC Gen1 (firmware Arm 1.00) even supports Modbus via RS485?
-     If not, is there a way to get an official firmware update to enable it?
-     Any suggestions for alternative debugging methods?

Any help or pointers would be greatly appreciated! Thanks in advance for your time and for maintaining this integration.

Best regards,

David

Vaszago, 2025-02-06:
The best thing to do is to chat with support via the Solax homepage. And simply ask your Solax to install the latest version. It's always worked really well and quickly for me.

Johan219, 2025-02-06:
I also asked them to do this and then they updated the DSP. Since they don't provide further support, I don't know whether it's because of ARM's firmware or whether the problem of not being able to read out has something else to do with it.