wills106/homeassistant-solax-modbus #1251: [Bug]: SolaX: attempt to read serialnumber failed

KIvosak, 2025-02-08:
### Describe the bug

After upgrading HA to 2025.2.1, the integration stopped working because it was unable to read the serial number.

Root cause: TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given.

HA 2025.2.0 has the problem described here: [GitHub Issue #1238](https://github.com/wills106/homeassistant-solax-modbus/issues/1238).
Currently, I have reverted to HA 2025.1.4 and integration 2025.1.6.

### Integration Version

2025.1.9

### Homeassistant core version

2025.2.1

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP v1.47 ARM v1.44

### Connection Method

SolaX Pocket WIFI+LAN

### Dongle firmware

1.005.04

### Detailed Error Log

2025-02-07 21:45:37.571 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-02-07 21:45:37.655 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2025-02-07 21:45:37.655 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-02-07 21:45:37.656 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2025-02-07 21:45:37.656 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-02-07 21:45:37.657 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x1a10 failed; other address may succeed
2025-02-07 21:45:37.657 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find any serial number(s)
2025-02-07 21:45:37.657 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2025-02-07 21:45:47.659 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-02-07 21:45:47.661 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2025-02-07 21:45:47.661 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 70, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-02-07 21:45:47.664 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2025-02-07 21:45:47.664 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x1a10 data: None

btdkomplex, 2025-02-09:
Woke up this morning with this exact issue. Rolling back HA and the Integration has not fixed it. No changes were made prior to it breaking.

btdkomplex, 2025-02-09:
Manually updated the firmware on the Pocket Wifi to version 3.010.16 and that seems to have fixed it. Not sure if that was the underlying issue or not.

MaximVB86, 2025-02-09:
> Manually updated the firmware on the Pocket Wifi to version 3.010.16 and that seems to have fixed it. Not sure if that was the underlying issue or not.

Care to share the firmware?

Thanks in advance!

btdkomplex, 2025-02-09:
In case anyone else needs it I found a link to a box.com account with them all on here: https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-firmware-update/

Direct Link to Pocket Wifi dir: https://app.box.com/s/3nvo7ic523fhojf8uuto105q9a8dgk9n/folder/145866811000

pavel-B, 2025-02-11:
what fw version did you upgrade to? I run WIFI3.0+ stick with 3.037.04 and what I understand this one is not upgradable manually.

sumitkm, 2025-03-04:
@klv

> Rhetorical question: Why downgrade the firmware of the dongle when everything worked fine with the original firmware? The issue arose after upgrading HA or possibly the integration.
> 
> And one small note: The unofficial firmware source on box.com has been dead and unmaintained for about 3/4 of a year.

I have the same dongle which i guess doesn’t have a random binary blob that can be installed, any luck with the issue? There seems to be an update of pymodbus version in the build that broke it, but i think HA may have installed a newer version? How isolated are python dependencies in HA/HACS? Noobi question!

aen42, 2025-03-04:
I have had the same problem when using a Serial RS-485 adapter. Restarts and going back to a known good backup still failed. Replaced the serial adapter (and it seemed the old one had indeed broken when I checked with mbpoll) but solax modbus could still not read out the serial number. As a last resort (and for some more flexibility) I just bought a waveshare eth to RS-485 for TCP to modbus and that helped for me. Now working well again. Thanks for the great work on this custom component!

sumitkm, 2025-03-06:
@KIvosak I was asked by SolaX support to unplug the wifi dongle and replug in after a minute, for a different issue. I didn't think of checking the plugin again after I did that, but lo and behold, it is working again. I can only imagine the Wifi dongle had crashed/run out of memory and the "reboot" fixed it.

sumitkm, 2025-03-14:
@mupper did you get a chance to unplug the wifi stick, give it a minute and plug it back? That’s what fixed it for me.

mupper, 2025-03-14:
> [@mupper](https://github.com/mupper) did you get a chance to unplug the wifi stick, give it a minute and plug it back? That’s what fixed it for me.

THX alot, it helps. Now it runs again. 

What I don't understand, however, is that the other Solax integrations were still able to access the adapter and the web interface also received the data. So, it must still be the integration's fault. Or am I misunderstanding something?

wills106, 2025-03-14:
The cloud connects over MQTT, so completely different protocol and set of code within the PocketWiFi.
There is probably a memory leak or connection restarts too quick or something like that on the Modbus side which forces the connection to close within the PocketWiFi.

It's only an ESP32 so not a lot of RAM.

It's a known issue and it's in the [SolaX FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)

The issue will get resolved when either SolaX fix something in the Firmware or someone with a PocketWiFi x.x can improve the integration. Unless someone is willing to buy and pay for a newer Gen4 or Gen5 Inverter to be installed for me 😉 as there is nothing I can work off. As I can't use a PocketWiFi on my Gen3.

romankysely, 2025-03-14:
I heard from Solax distributor support that they are having problems with communication load when they switched to 10s interval and are now trying to separate traffic between LAN for Cloud and WiFi only for local Modbus TCP. Maybe it will help to use LAN PocketWiFi for connection to cloud/MQTT and WiFi path only for local Modbus TCP communication.

### **Seriously I am willing to contribute 100GBP for Wills106 for a new G4 as a reward for the development work on the integration. Will enough contributors join?**

wills106, 2025-03-14:
> Seriously I am willing to contribute...

My comment was meant to be a bit tongue in check (Not sure if that saying translates into other languages)
I was just trying to highlight there are some issues I can't replicate, so I can't personally fix them.

Why on earth SolaX dropped the built in ethernet connection on later models, I'll never know?

wills106, 2025-03-14:
> > Seriously I am willing to contribute...
> 
> My comment was meant to be a bit tongue in cheek (Not sure if that saying translates into other languages) I was just trying to highlight there are some issues I can't replicate, so I can't personally fix them.
> 
> Why on earth SolaX dropped the built in ethernet connection on later models, I'll never know?

romankysely, 2025-03-14:
I understood your sarcasm well, even though I'm not a native English speaker. Still, I mean my offer seriously. It would be great if you had the same lab as those of us with the G4. I don't know what we would do without your help.
😉

mupper, 2025-03-14:
> I heard from Solax distributor support that they are having problems with communication load when they switched to 10s interval and are now trying to separate traffic between LAN for Cloud and WiFi [...]

I will spend some money too. wills did make a perfect job and i used his integration every day. 

@wills106: Sorry, my friend, I just wanted to understand. I thought that with the LAN/WIFI dongle, the LAN connection was used internally. So I connected my WallBox and my LAN network together with the Solax.

Thanks anyway for the explanation, it really helped. If I had known this beforehand, it would have been less stressful. The question for the future is how we can perhaps share this information better and faster with the entire community.

PS: Thank you for your great work!

steps56, 2025-03-14:
If there will be some fund rising about wills106 to buy a SolaX G4, I will support too. (but not by those buy me a coffee or odd similar thing.)

About the Modbus via Dongle thing, its internally transformed to modbus and not at any healthy way, I read.
Its mainly for the China cloud aka Portal and App using http or whatever.
I switched yesterday to waveshare via COM Port.
This connection is the one to be made better for data stream.
The Dongle is often pain.
Sidebenefit using Waveshare is to block Web access of the System to China or whatever.
This could be done also using the Dongle, but just to separate Dongle from lokal datastream to HA, is my perferred way..
I try to find the SolaX Knowledgebase pdf about it to link it here later.

ruisilva93, 2025-04-15:
Same problem here, with 3.039.05

apospa, 2025-04-26:
> [@KIvosak](https://github.com/KIvosak) I was asked by SolaX support to unplug the wifi dongle and replug in after a minute, for a different issue. I didn't think of checking the plugin again after I [...]

Thanks, it helped! After rollback integration, restore backup of whole system I was desperate and tried this easy step. And it worked! Last time I'll try to read more first 🤣

closed 2025-08-28 by wills106