wills106/homeassistant-solax-modbus #1253: [Bug]: Homeassistant vypina na 1 vteřinu přetoky

VikyTT, 2025-02-10:
### Describe the bug

Dobrý den, od poslední aktualizace se mi v HA objevuje krátkodobé vypnutí přetoku. Nemáte s tím zkušenost nebo stejný problém? Viz foto. 

[image]

### Integration Version

2025.01.9

### Homeassistant core version

2025.2.1

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

X85vcm

### Inverter firmware versions

Arm 47 09 44

### Connection Method

Wifi3

### Dongle firmware

Originál

romankysely, 2025-02-10:
Look there, it could be the same problem: https://github.com/wills106/homeassistant-solax-modbus/issues/1238

shaarkys, 2025-02-11:
@VikyTT ... only English + it's indeed duplicated, please close this report

wills106, 2025-02-23:
Closing down as it's a duplicate of #1238 

Have a look at this comment https://github.com/wills106/homeassistant-solax-modbus/issues/1238#issuecomment-2670890675

> only English

I actually prefer people to post in their native language over using a translator. Another user on here may natively speak that language as well as English and better understand the question.

shaarkys, 2025-02-23:
Ah, ok, tak to jsi první, dobře tedy ;-)