wills106/homeassistant-solax-modbus #1254: Upon upgrade of the HA core to latest release 2025_2_1 brakes integration with Solax

pavel-B, 2025-02-11:
Upon upgrade of the HA core to latest release 2025_2_1 brakes integration with Solax. Shows entities as unavailable and upon deleting/adding Solax, like in past it does not come up. Any hints why? Thank you

MaximVB86, 2025-02-11:
Try updating the firmware of your dongle: https://github.com/wills106/homeassistant-solax-modbus/issues/1251

pavel-B, 2025-02-11:
I do run 3.037.04 on the short interval USB Wifi stick. That means this one is not upgradable by firmware. Only the old ones.

Elvisak, 2025-02-11:
I have the same problem. I am using an older dongle with a five minute update. I have 3.10.16 installed and that is the latest version. Home Assistant version is 2025.2.2.
I have also removed and added again.

pavel-B, 2025-02-11:
[home-assistant_solax_modbus_2025-02-11T07-24-12.041Z.log](https://github.com/user-attachments/files/18751147/home-assistant_solax_modbus_2025-02-11T07-24-12.041Z.log)

Pearse500, 2025-02-11:
Same issue here

Elvisak, 2025-02-12:
I removed and set everything up, including restarting the dongle.
Now everything works.

luismsousa, 2025-02-14:
Leaving this here for posterity, every time I get this kind of error I have to bounce my pocket WiFi from the WiFi network by either power cycling it or going into my ubiquity controller and forcing it to reconnect to the access point.
After that, I restart HA and it magically connects.

steps56, 2025-02-15:
All postet problems seems to not be regarded to the Modbus Integration.
I'm back to 2025.01.6 as latest 2 Versions are producing often unlogic and extreme Spikes at some entities. (crosschecked with App and so on, its just the Integration wich produces Spikes +/-)
That's the only problem so far now.
The mentioned stuff here is Wlan, Lan, Dongle and so on personal system problems.
Tip/Hint:
If your system accepts (check for it at thd web) the latest Lan/Wlan combination Dongle, buy one low at about 40 Euro and use it.
Its by far best Dongle ever regarding stability and connectability.

02curls, 2025-02-18:
I too saw nonsensical spikes in battery SOC, solar yields during night etc. with HA core 2025.2.1
But now the latest HA core upgrade left me with all Solax entities lost

steps56, 2025-02-18:
> I too saw nonsensical spikes in battery SOC, solar yields during night etc. with HA core 2025.2.1 But now the latest HA core upgrade left me with all Solax entities lost

Did you restart HA TWICE after updating ?
SolaX modbus integration needs very often 2 or mord new HA starts, to start working again... since a long time from time to time at some updates 🤷🏼‍♂️.

02curls, 2025-02-18:
> Did you restart HA TWICE after updating ?
> SolaX modbus integration needs very often 2 or mord new HA starts, to start working again... since a long time from time to time at some updates 🤷🏼‍♂️.

I downgraded to 2025.1.4
no luck after the initial restart
second restart and all entities are back :)

02curls, 2025-02-18:
> Why back to 2025.01.4 ... as 2025.01.6 is also working nice WITHOUT any spikes.... may be .7 too 🤷🏼‍♂️.

I do not see any later HA CORE 2025.1.x versions:
2025.1.1 - January 7
2025.1.2 - January 9
2025.1.3 - January 20
2025.1.4 - January 24

steps56, 2025-02-18:
> > Why back to 2025.01.4 ... as 2025.01.6 is also working nice WITHOUT any spikes.... may be .7 too 🤷🏼‍♂️.
> 
> I do not see any later HA CORE 2025.1.x versions: 2025.1.1 - January 7 2025.1.2 - January 9 2025.1.3 - January 20 2025.1.4 - January 24

Latest HA core is 2025.02.4 for sure already and working.
I was always talking about SolaX Modbus versions !!!!
HA core no problem, integration is the spikes problem root !
And as this is zl avoid by going back to SolaX Modbus Integration. 2025.01.6

wills106, 2025-02-23:
Closing this down, the PocketWiFi's are known to be a problem when restarting [HA](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)

I don't own and I can't use a PocketWiFi 3.x so I can't work on this.

closed 2025-02-23 by wills106