wills106/homeassistant-solax-modbus #1255: [Bug]: SolaX Today's Yield shows only Master Inverter Yield when using two Inverters in parallel mode

intellicasa, 2025-02-16:
### Describe the bug

I have two Solax Inverters (X3-Hybrid-G4) working in parallel mode. In Home Assistant I'm configuring the Master inverter and check the Option **Parallel Mode (Master-Slave)**

The entity **SolaX Inverter Today's Yield** shows only the yield of the master inverter:

[image]

I added a second hub to get the values for the slave hub separately and here I get the slaves value:

[image]

My expectation would be to get either both values separately within the first device (hub).

Best regards
Daniel

### Integration Version

2025.01.9

### Homeassistant core version

2025.2.4

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM 1.35-1.09 and DSP 1.36

### Connection Method

SolaX Pocket WIFI3.0

### Dongle firmware

3.001.02

romankysely, 2025-02-19:
Yes you are right. But there is missing not only this entity unfortunately. 

(is for Master only, for Slave is missing)
sensor.solax_battery_input_energy_today 
sensor.solax_battery_output_energy_today
sensor.solax_remaining_battery_capacity
sensor.solax_house_load
sensor.solax_today_s_solar_energy

Be carefuly when you add second HUB. Both instance of the HUB at Integration Solax Invertor Modbus is running at the same time and there could be collisions and corrupt data in database. My own experience. 
I am afraid thet problem is not at integration because modbus registers of these entities is missing in document so than can not be involved at intgration. 
Suppose of solution: 
1. use Proxy Multiplexor which sort TCP request to queue and protect collision on one WiFi dongle and try to address via one WiFi dongle two Modbus address (Invertor Master and Invertor Slave). Then SUM entities M+S via templates.
2. use separate HW (WiFi Dongle for each invertor) and setup HUB for each dongle directly. Then SUM entities via templates.
I suppouse same procedure for Modbus RTU with Waveshare convertor when you have LAN dongles only.

Or any other idea??

wills106, 2025-02-23:
I have marked this as Wont Fix, as there is a limitation to what registers can be read through Parallel Mode.

closed 2025-02-23 by wills106