wills106/homeassistant-solax-modbus #1256: sensor.solax_battery_output_energy_today shows one wrong data point early in the morning

gpi2602, 2025-02-19:
I'm using the HACS Solax Power Modbus Integration.  It is working really well for me. A big thank you to the author(s)!!!

I'm not sure if I'm here in the right discussion, but let me start here.

I see nearly every day a wrong data point in sensor.solax_battery_output_energy_today. It is just one datapoint early in the morning. It looks like the sensor is showing the total of the Solax PV energy all together. I then have to go in Home Assistant Developer Tools Statistics and have to correct this wrong data point to zero.  Then everything is looking fine again. 

Does anybody else have this issue?

Is that an issue with the Solax Power Modbus integration or the energy dashboard or with solax providing a wrong data point?

 Kind regards
Gerd

SpiritofGS, 2025-02-19:
I've been having the same problems for days. The sensor usually stores 100 or 200 KWh too much and sometimes too little. I've currently given up trying to delete the wrong values several times a day using the developer tools. Now I can forget all my statistics, but it's just too time-consuming for me. I'll have a look at the databases at some point and clean up the illogical values using a script. I don't have the patience at the moment. Thanks to the developers of the integration! But it's no fun at the moment

[image]

shaarkys, 2025-02-20:
Indeed it's the same as https://github.com/wills106/homeassistant-solax-modbus/issues/1238, you can still downgrade.

@gpi2602 please close this one, thank you.

gpi2602, 2025-02-28:
@shaarkys I'm not sure if that is the same as #1238. I got the Issue today again after installing the update yesterday which should fix the issue #1238. 
The value of sensor.solax_battery_energy_today is showing exactly at 7am (and only at 7am) the value of the Total Yield.  I have been correcting the value easily with the Home Assistant development tools and statistics, of course - however, I'm wondering if anybody else has this issue and if it can be fixed.

Thanks     Gerd

[image]

lazy-pete, 2025-03-05:
Hi - been having a similar issue for a while. Seemed to have got better but back this week. 

This evening I apparently had a house load of 81,921 kW. 

Not quite sure how to tweak the unbelievable values, but is making the energy display look a bit ridiculous. 

[image]

lazy-pete, 2025-03-10:
I’m wondering whether it’s down to blips in network connectivity - I used to have quite long gaps which were problematic for home assistant running optimisers, but since sorting that it’s been stable. All the spikes I see are associated with what I think is an intermittent drop in connection, and it can be various values that are ludicrous.

gpi2602, 2025-04-09:
since the last update I had no wrong data points any more. Thank you for fixing this. Issue  to be closed.

closed 2025-04-09 by gpi2602